MarlinFirmware/Marlin #1448: Menues on gaphical displays inconsistent

AnHardt, 2015-02-01:
On DOGM-displays in all menues including SD-menu reversed text (white on black) is used to index the current line. But for editable menu-items '>' is used to indicate the current position and no highlighting.

Possible fix in _drawmenu_setting_edit_generic() in dogm_lcd_implementation.h:

step a: use the highligting code from lcd_implementation_drawmenu_generic() and _drawmenu_sd().
step b: introduce a function maybe 'highlight_current_menuitem()' and use it in the relevant functions.

Or maybe the other way around. Mark current menuitem with a '>' like on the character displays.

Any opinions?
Will fix it by myself.

alexborro, 2015-02-01:
I think eu should be consistent. Highlight or > for all.

I prefer highlights because it saves space.

Cheers

Alex.

AnHardt, 2015-02-01:
Now:
[image]
[image]

AnHardt, 2015-02-01:
Fix with highlighting:
[image]

AnHardt, 2015-02-01:
@alexborro This will not save space on the display as implemented now. The space at the beginning of the ling is used for the up-arrow used in the back-menu-item.
[image]
But this is redundant repeated at the end of the line.

If asked for i can remove this.

thinkyhead, 2015-02-02:
I like more obvious highlighting, but we're used to __>__ on character LCDs. Perhaps we should make it (yet another) configuration option. Users and vendors may have different preferences, eyesight issues, or whatever.

AnHardt, 2015-02-02:
@thinkyhead It's absolutely no problem to use highlighting and '>' at the same time. I don't like the idea of making it configurable.  The length of the strings (max 17) is optimized for the 20xX character displays. We should not use the additional space without making a second string set. What I also don't want.

thinkyhead, 2015-02-02:
@AnHardt Yes, exactly my point. Strings can't be longer without many additions. The language files actually did have LONG_ versions at one time, but they were unwieldy I guess and were removed.

alexborro, 2015-02-02:
Sometimes we need to decide and go ahead. I vote for highlight and more
space.

Highlighting is a feature only available for graphic LCDs; that's the
reason we use > for other character display.

Almost all regular users are used to graphical user interface that uses
highlighting.

And I think highlighting an option make it more visible than > in all
cases.

Cheers

Alex.

AnHardt, 2015-02-04:
Fixed.

closed 2015-02-04 by AnHardt