MarlinFirmware/Marlin #1454: LCD info replacement for X and Y position.

alexborro, 2015-02-02:
Am I the only one who don't give a damn for X and Y position information DURING print??

I think those information during print is completely useless.. The informed position is always late..
They are nice only in calibration process..

I'm thinking about replacing them during print by any other more useful information like:
a) Filament consumed,
b) Estimated remaining time,
c) maybe print cost computing filament costs and power consumed. 

Or whatever else you guys think useful... because X and Y are worthless.

Cheers.

Alex.

thinkyhead, 2015-02-02:
Yes, you have a valid point!

nicolas-rambaud, 2015-02-02:
+1

*De :* alexborro [mailto:notifications@github.com]
*Envoyé :* lundi 2 février 2015 16:48
*À :* MarlinFirmware/Marlin
*Objet :* [Marlin] LCD info replacement for X and Y position. (#1454)

Am I the only one who don't give a damn for X and Y position information
DURING print??

I think those information during print is completely useless.. The informed
position is always late..
They are nice only in calibration process..

I'm thinking about replacing them during print by any other more useful
information like:
a) Filament consumed,
b) Estimated remaining time,
c) maybe print cost computing filament costs and power consumed.

Or whatever else you guys think useful... because X and Y are worthless.

Cheers.

Alex.

—
<https://github.com/MarlinFirmware/Marlin/issues/1454>.[image: Image
supprimée par l'expéditeur.]

NarimaanV, 2015-02-02:
+1  Couldn't agree more! Like the idea of an estimated time remaining the most :D

bjmckenz, 2015-02-02:
For me:

d) z-height

This is because it gives me a different sense of progress than %complete.

NarimaanV, 2015-02-02:
@bjmckenz Isn't that already on the screen?

bjmckenz, 2015-02-02:
Er, yes. Point was that I want more than just a, b, c. Apologies for being unclear.

CONSULitAS, 2015-02-02:
+1

Layer count would be nice, but would be needed to be calculated. If you know your print has 70 layers and you should care about layer 45...

Hmmm... Not so easy - simply count any z-movement?

daid, 2015-02-03:
First re-implementation of the LCD menus I did did not have the X/Y values. Because, like most of you, I didn't care about that at all. But some people really wanted it.

Anyhow, looking at Z movements won't work for people that use Z-Hop.

thinkyhead, 2015-02-03:
On a similar note, it would be good if XYZ updated more often, at least outside of active printing. (I'd +1 that as a general enhancement too: the ability to update just a few LCD cels – not always the whole screen – to reduce bus chatter. Less traffic might mitigate stuttering issues with Delta+Graphical LCD too.)

Natealus, 2015-02-05:
+1 to Thinkyhead. Thinking on the right path there. :) I agree with others here but from a performance stand point during printing operation...it can only cause more work/cpu delay to have it report the X and Y position for every LCD refresh. And probably any printer moves way too fast for that slow refresh to be reliable.

alexborro, 2015-02-05:
I will change this as soon as have some time to play with the LCD...

ntoff, 2015-02-06:
Yes! +1

Also more digits after the decimal, at least on the full graphics, it only shows 1 digit after the decimal place, so if you're at 0.05mm, it'll show you're at 000.0.

bkubicek, 2015-02-06:
first of, you printer is hardly more accurate than 0.1mm in xy. Second:
Primary goal of a printer is to print. Anything that decreases print
quality is bad, and should not be there.
Fast LCD updates are bad for print quality. This is why there should not be
anything displayed that changes rapidly. One can display x y, but only if
the printer is at rest. Otherwise, its first not useful, second not
readable, as most lcds have a huge blurring if you change it too fast.
the z-value is relevant, e.g. for manual filament changes. the layer is not
existing, because for many prints, there are no layers at all, e.g. due to
jorrising. Also, if you tower (print two layers at island 1, then goto
island 2 and print two layers there) layers can decrease.
So, since its not universal, its not included.
My original lcd implementation was made in a way, that only things that
actually change on the display change. As anything else leads to flickering
and bad readability. And for those who believe its a non-issue, there were
people with graphical displays, who wondered why they get warts on the
print when they use the menu. Also I had to dump my primary LCD menusystem
due to decreased print quality.

Bernhard

thinkyhead, 2015-02-06:
@bkubicek Is your optimized LCD code (_made in a way, that only things that
actually change on the display change_) the code that is still in Marlin today? I did notice that if you rotate the dial during a print, to change feed rate or navigate menus, that the LCD updates very frequently, so maybe there should be some threshold to how frequently it updates, if there isn't already – no more than 5 times per second. I don't know what makes sense as a threshold for graphical displays. But are there any effective optimizations left that can reduce bus traffic, I wonder?

ntoff, 2015-02-06:
@bkubicek I was talking about the Z axis and I meant while manually moving it around to home it or adjust the Z offset, not while printing. Sorry for the confusion. My Z axis is well and truly capable of moving in 0.05mm increments, I have printed objects at that layer height.

daid, 2015-02-06:
The LCD is forced updated when you rotate the dial.

But, I think the main problem is the Arduino LiquidCrystal library. I'm using a custom LCD driver on the Ultimaker2, where I update the screen 10 times per seconds, full display update of 128x64 pixels. Without any issues. And this takes less them then the current mainline UltiLCD code.

alexborro, 2015-02-06:
@daid , what about using this custom LCD driver in official Marlin??

daid, 2015-02-06:
@alexborro well, would be kinda expensive to visit all the printers to replace their displays, as it's using a different display with a different protocol. I'm just saying that I think the display low level driver that is being used right now could use improvements.

alexborro, 2015-02-06:
Sorry @daid , I misunderstood you. I though you had a better driver for current/common LCDs.

thinkyhead, 2015-02-06:
If there's something we can do to optimize the most common displays with their chatty protocol, I'd be interested to hear some ideas. Will the trade-off of extra computation to avoid extra bus traffic pay off enough to be worth the effort?

clefranc, 2015-02-14:
I would like to see the total mm extruded since the print started. And Z height of course.

thinkyhead, 2015-02-16:
:+1: @clefranc

Natealus, 2015-02-17:
Total filament extruded for the print actually would be pretty easy wouldn't it? I've noticed watching g-code terminal that its already counting the millimeters as they go up.

thinkyhead, 2015-02-19:
I like all the suggestions, as the XY is quite useless during print. Perhaps this could be done, not as a feature that's either on or off, but as a menu option, so that you can switch to a different data display on the fly, during printing or otherwise. I can see live coordinates still being useful in some calibration exercises. Some other modes to consider:

- Destination coordinates display instead of current coordinates.
- Total filament used since EEPROM reset
- Total volume printed (this print or since EEPROM reset).
- etc.

ntoff, 2015-02-21:
@thinkyhead Is there a way to store total print time to the SD card if one is installed? Maybe display that as an option so you can see how many hours your printer has done and maybe give it some lube every 100 hours or whatever.

thinkyhead, 2015-02-22:
@ntoff The EEPROM offers one possibility. Currently the EEPROM routines are set up to write the whole settings blob at once, as in M500. This is overkill for saving these kinds of statistics, so what we need is a routine that can write to specific parts of the EEPROM, or into a section above the saved settings.

daid, 2015-02-22:
@thinkyhead https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/lifetime_stats.cpp

thinkyhead, 2015-02-22:
:cool: :sunglasses: !

boelle, 2015-05-31:
i dont care much for XYZ at all during print.... its not updated fast enough etc...

but should we keep this one open? or close it and let someone submit a PR ?

ntoff, 2015-06-01:
Z I care about because if the print fails I can use the Z height to re-print from there up.

thinkyhead, 2015-06-10:
I notice lately that my XY coordinates update a lot more often than they used to. But I still rarely care about them. However, now the display will show `---` for an unknown X/Y so that in itself is kind of useful to know that you need to do homing.

closed 2016-04-23 by jbrazio