MarlinFirmware/Marlin #1460: extruder_multiply[active_extruder] not used?

thinkyhead, 2015-02-03:
I notice that the planner is using the current value of `extrudemultiply` but I can't find any location where `extruder_multiply[active_extruder]` (as set in M221) is copied into this variable, as I would expect with multiple extruders. Wherever it is that extruders are switched, that seems like the place where this value should be set. Perhaps in `planner.cpp` near `block->active_extruder = extruder;` ?

Wurstnase, 2015-02-03:
Maybe there? https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L604

MagoKimbra, 2015-02-03:
in my marlin i deleted extrudemultiply and use only extruder_multiply[active_extruder]...

thinkyhead, 2015-02-04:
@MagoKimbra Great! That's a change that needs to come to the main branch, asap. It's definitely a bug.

KiteLab, 2015-04-04:
Should work now.
Has to be tested.

closed 2015-04-26 by thinkyhead