MarlinFirmware/Marlin #1468: [Feature Request] Allow for CoreXZ printers

CptanPanic, 2015-02-05:
There is some new development in the reprap forums for something called CoreXZ, which is the the standard CoreXY mechanism, turned 90 degrees to be used on a prusa printer.  See http://www.thingiverse.com/thing:526673 , https://www.youtube.com/watch?v=vp-ECU6ZiiU&feature=youtu.be and  http://forums.reprap.org/read.php?2,427389,466327#msg-466327

alexborro, 2015-02-05:
Nice design, the change is quite simple, just replace anything about CoreXY
Y_Axis by Z_Axis..

Unfortunately I'm quite busy with other CoreXY stuffs, so not enough time
right now.

Cheers.

Alex.

boelle, 2015-05-31:
closing... no activity and we have agreed that hardware support should come from makers and not us

fmalpartida, 2015-05-31:
This one is more of a kinematics comment, not an actual HW electronics one.

thinkyhead, 2015-06-09:
`COREXY` affects mostly planner and stepper code. When `COREXZ` is implemented it will do all the same stuff, just with different axes. I will try to roll up a PR within the next few days if it's not too tricky.

thinkyhead, 2015-06-10:
If you have a CoreXZ machine, try #2272 to see if it works properly.

closed 2015-09-25 by AnHardt