MarlinFirmware/Marlin #1471: Full graphics controller shows fan at 1% when off

ntoff, 2015-02-07:
With a full graphics controller LCD, the fan speed indicator always shows 1% even when the fan is turned off with M106.

Line 228 in dogm_lcd_implementation.h is to blame I believe with u8g.print(itostr3(int((fanSpeed*100)/256 + 1))); being the culprit. No matter what the speed is, "1" is always added to it so if the speed is 0%, it still gets 1 added to it, showing up as 1% on the LCD panel.

thinkyhead, 2015-02-07:
Hmm, weird. It should really just be `(fanSpeed * 100 / 255)` because the full range is 0-255. However, then you only get 100% at exactly 255. So perhaps it should be shifted so that only 0 shows 0%, while 253-255 show as 100%. The calculation for that is `((fanSpeed+2) * 100 / 255)`.

ntoff, 2015-02-07:
It goes from 0 to 255 so there are 256 possible positions, 0 being the first. I "fixed" it in my own version by just putting the 1 in a different spot. It's still not completely accurate but at least while off the fan does show 0%.

thinkyhead, 2015-02-07:
#1472

closed 2015-02-07 by thinkyhead

nophead, 2015-02-07:
The correct round to nearest expression is (fanSpeed * 100 + 127) / 255.
Yours is periodically high by one. E.g. 253 is closer to 99% than 100.

ntoff, 2015-02-07:
yeah I know, but constantly having that 1% there bugged me to no end, so I was happy it was gone even if the other readings were slightly off :p

nophead, 2015-02-07:
Yes but my formula will also be 0 for PWM values less than 2. I.e. correct maths is always correct to the nearest percent.

thinkyhead, 2015-02-07:
@nophead I'll update with your fixed point rounding asap.

ShrikantPai123456, 2017-04-21:
Hello ...

I am new to this..

I have sanguino clone controller and FAN EXTRUDER working fine, Also controller FAN is working fine.

I have issue related to graphical LCD... When FAN is ON, FAN symbol is rotating on LCD. But FAN percentage is not displaying.

I went to CONTROL and FAN kept for 128 (50%) but only message "---" coming instead of message "50%"...

Pls help ASAP....

This is photo when FAN is ON....

[image]

ShrikantPai123456, 2017-04-21:
What shall I add or comment or uncomment in MARLIN ?????

ShrikantPai123456, 2017-04-21:
[image]

At this 1% FAN is rotating FULL.. What shall I do sir????

ShrikantPai123456, 2017-04-21:
Hi all...

I am using 128*64 GLCD. On the LCD its 1% FAN PERCENTAGE but my controller FAN is FULL rotating (up to 255)... Pls tell me what can I do???

Here is pins.h

#define LED_PIN -1

#define FAN_PIN -1
#if FAN_PIN == 12 || FAN_PIN ==13
#define FAN_SOFT_PWM
#endif

#ifdef MELZI
#define LED_PIN 27 /* On some broken versions of the Sanguino libraries the pin definitions are wrong, which then needs LED_PIN as pin 28. But you better upgrade your Sanguino libraries! See #368. */
#define FAN_PIN 4 // Works for Panelolu2 too
#endif

#ifdef STB
#define FAN_PIN 4
// Uncomment this if you have the first generation (V1.10) of STBs board
#define LCD_PIN_BL 17 // LCD backlight LED
#endif

Using sanguinololu clone...

Here is config_adv.h

#define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
#define CONTROLLERFAN_SECS 60 //How many seconds, after all motors were disabled, the fan should run
#define CONTROLLERFAN_SPEED 255 // == full speed

// When first starting the main fan, run it at full speed for the
// given number of milliseconds. This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100

// Extruder cooling fans
// Configure fan pin outputs to automatically turn on/off when the associated
// extruder temperature is above/below EXTRUDER_AUTO_FAN_TEMPERATURE.
// Multiple extruders can be assigned to the same pin in which case
// the fan will turn on when any selected extruder is above the threshold.
#define EXTRUDER_0_AUTO_FAN_PIN -1
#define EXTRUDER_1_AUTO_FAN_PIN -1
#define EXTRUDER_2_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255 // == full speed

help asap....

thinkyhead, 2017-04-22:
What version of Marlin are you running @ShrikantPai123456 ?