MarlinFirmware/Marlin #1473: Compilation instructions don't mention libraries for LCDs

filipmu, 2015-02-07:
The compilation instructions don't mention how to compile with the add-on libraries required for the LCD or GLCD.  There are instructions for loading those libraries, but they are in:

Marlin/ArduinoAddons/Arduino_1.0.x/libraries/U8glib/INSTALL.TXT

Its a little buried for the average user.  Would be worth a mention on the Compile page.

thinkyhead, 2015-02-07:
We should copy this info from the Configuration file where it's explained that extra libs are needed, point out that INSTALL.TXT file, and include a link to the Arduino documentation on how to install libraries generally.

NarimaanV, 2015-02-07:
This is the same as https://github.com/MarlinFirmware/Marlin/issues/1387. To summarize, after some discussion, the main issue became wanting to have users use the copies of libraries in the ArduinoAddons folder since if we said use the links, the libraries might update to the point of breaking compatibility in the future. Problem with this is, only half (2/4) libraries are in the ArduinoAddons folder so then we thought we should add the rest. THEN we realized 3 of the 4 libraries should share some functionality in all being Liquid Crystal libraries. Then there was talk of trying to merge them and that's pretty much where it ended. So that's the main issue with this issue as well, as what to do with the libraries. Personally I think for now we might as well include all 4 and maybe attempt to create a custom merged library later, but that seems to deserve its own discussion. Just thought I'd point this issue to an earlier one where almost the exact same thing was brought up.

thinkyhead, 2015-02-20:
With recent work on Russian / Cyrillic / Unicode / Kana support we are looking at patching up the missing functionality in the included library, but if possible entirely within the Marlin code. As a second option, including a second LiquidCrystal library, already patched, for Russian. Follow the recent threads on these topics for latest code updates.

closed 2015-02-20 by thinkyhead