MarlinFirmware/Marlin #1475: The g29 command only runs the z-probe 10 mm torwards the bed.

hawkan-rules, 2015-02-07:
The g29 command only runs the z-probe 10 mm torwards the bed. If the probe dont detect the bed after running for 10 mm it will just continue as it succeded instead of failing and halting the print.

I've changed the distance to 200mm and if you guys think its a good solution I can make a pull request.

The code I use when starting a new print.
g28 x0 y0 
g29

Wurstnase, 2015-02-07:
Before you start with g29 make a g28 z0 in the middle of your bed.

thinkyhead, 2015-02-07:
G29 checks to see if it has a _known X and Y position_ and aborts if it doesn't. But it doesn't check to see that the Z is within a certain range of the bed. Perhaps an extra check should be done. Or… perhaps G29 should invoke G28 when it needs to, instead of just printing an error to the serial out.

hawkan-rules, 2015-02-07:
Why should it need to do a g28 z0 first? It just takes more time and the probe point wont be used in the calculations.

thinkyhead, 2015-02-08:
I see no reason why G29 should not move as far as it needs to in order to probe the bed. But perhaps someone can shed light on why it limits the Z movement to 1cm in the first place.

brainscan, 2015-02-08:
It does if you home z first, seriously don't see the problem here. It takes 5 seconds to home z before G29 so I don't understand the argument that it was takes too long.

galexander1, 2015-02-08:
For Marlin, this gcode sequence is incorrect for initialization:
   G28 X0 Y0
   G29
It leaves the Z axis uninitialized.

The correct one is:
   G28
   G29

I am not sure but I think @boelle deleted the last conversation about
this, so I will just re-explain.

G29 only attempts to probe 10mm below the bed, which relies on knowing
where the head is when it starts.  G28 X0 Y0 doesn't home Z, so Marlin
has no idea where the bed is -- generally, it assumes it is at Z=0
(already touching the bed) when it starts.

Marlin's G29 behavior is a safety feature to prevent Marlin from
endlessly pounding the head into the bed in the case where the Z probe
fails to trigger.  That seems like a good idea to me, because most Z probe
designs I've seen are, frankly, awful.  Though arguably pounding the bed
for 10mm is as bad as pounding it for 1000mm -- even 10mm might damage
the robot.  *sigh*

The alternative is that G29 could be willing to probe a large distance
(such as Z_MAX_POS+10) below the bed, which is presumably what some other
firmwares do.  That's a little more dangerous but will work with this
common G-code initialization process.

I don't use G29, so I don't have a horse in this race.  I hope a
conversation follows that is a little deeper than the one I remember last
time -- "G28 X0 Y0 is correct" "no it isn't."  If we settle on it being
incorrect, then I hope this conversation does not get deleted, so we can
point people at it in the future when they report this issue again and
again.

thinkyhead, 2015-02-09:
Is that a Documentation issue, an issue with hot software, or more of a widespread notion that we need to dispel? I mentioned that the sanity check in G29 isn't checking whether Z is in a known location. Should that be added?

galexander1, 2015-02-09:
Is there a flag somewhere to indicate whether an axis has already been
homed?

Wurstnase, 2015-02-09:
Something like this? https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L241

thinkyhead, 2015-02-09:
Yep, and the `home_axis()` function sets the flag for each axis when it is homed. These flags persist until ... the motors shut off, I believe.

Wurstnase, 2015-02-09:
Yes, in Marlin.h

```
#define disable_x() do { WRITE(X_ENABLE_PIN,!X_ENABLE_ON); WRITE(X2_ENABLE_PIN,!X_ENABLE_ON); axis_known_position[X_AXIS] = false; } while (0)
```

galexander1, 2015-02-09:
Awesome!  I missed that in my skim.  We could make the 10mm dependent on
the axis known flag.

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L1066

Could become
   float zPosition = (axis_known_position[Z_AXIS] ? 0.0 : -Z_MAX_POS) - 10.0;

But there are a few more details...  G29 isn't meant to determine the
Z-axis position.  So it doesn't set axis_known_position[Z_AXIS]=true; and
also it will wind up with large values for the correction factors,
instead of the small values it is expected to come up with.  That may or
may not really matter, though for floating point numbers big+small is the
worst case for rounding errors.

Just some things to keep in mind if G29 is going to replace G28 Z0.

thinkyhead, 2015-02-20:
Move this conversation to #1499 :8ball:

closed 2015-02-20 by thinkyhead