MarlinFirmware/Marlin #1480: Ugly format of values on displays

AnHardt, 2015-02-08:
Fact - unnecessary leading 0s.
PID-s - unnecessary leading 0s - unnecessary sign.
[image]

Vz-jerk  - unnecessary leading 0s - unnecessary sign.
[image]

steps/mm all  - unnecessary leading 0s - unnecessary sign.
[image]

Fact - unnecessary leading 0s.
PID-s - unnecessary leading 0s - unnecessary sign.
[image]

Vz-jerk - unnecessary leading 0s - unnecessary sign.
[image]

steps/mm all  - unnecessary leading 0s - unnecessary sign.
[image]

Will try to fix.

thinkyhead, 2015-02-09:
Great! I recently did this with the Z display on the status screen. More pleeze!

thinkyhead, 2015-02-20:
We can move this conversation to the PR when that's ready. Closing issues to make room for a code freeze, perhaps!

closed 2015-02-20 by thinkyhead