MarlinFirmware/Marlin #1481: Float conversion utilities: Redundancys, naming conventions, comments, typedefs and errors in ultealcd.cpp

AnHardt, 2015-02-08:
Some remarks on float and int conversion utilities:
- I see a bunch of functions doing basicly the same, Just the amount of digits differs.
- I see different naming conventions. ftostr52 and ftostr32 both deal with numbers of the type 123.45 .
- I see comments not fitting to the right function.  Line 1541++ 
- I see "char *itostr2(const uint8_t &x)" what is the only on wanting a  uint8_t.

A possible fix could include:
a Change "char *itostr2(const uint8_t &x)" to "char *itostr2(const int &x)" and make the calls fit.
b Introduce a generic xtostr() function.
c Use the generic function in the functions used now (with #ifdef) to see what the impact will be.
d Implement a unified self explaining naming convention based on the parameters  of xtostr().

rfc
        AnHardt

AnHardt, 2015-02-09:
Reached step c.
Preview availabe at: https://github.com/AnHardt/Marlin/blob/xtostr/Marlin/ultralcd.cpp#L1522

thinkyhead, 2015-02-09:
I've cleaned up some of those in my play branch, but haven't bothered to reduce redundancy too much. I did adjust some of the logic. See how it compares with your adjustments. https://gist.github.com/thinkyhead/e590c6c2e7408afba21c

thinkyhead, 2015-02-09:
Ah, yours is much more comprehensive. I will have to examine it more closely later on!

odewdney, 2015-02-10:
There is always dtostrf, and then fixup the output if you want leading zeros, or is that the slippery slope to adding xxprintf...

AnHardt, 2015-02-10:
@odewdney 
Using  dtostrf() will use about 1522 byte of progmem and would need a additional wrapper to reach the wanted features. I think this is to expensive. And i have no clue what the run time will be.
Until now i saved a few bytes with my converter. We'll see where i will end up when this damn 'spacey' function is integrated. At least i have an idea now, how to make if. Not locking forward but finishing normal and replacing '0's and the '.' with ' ' from the end could be the key.

AnHardt, 2015-02-10:
So. Reached step d)
Now saving about 446 bytes. All is locking the same as before - at lest all I'v tested. .Even the special formatting for Z on the info screen on hitachi displays is working again.
All in all I may have lost a tiny little speed (some more comperations) compared to the single functions but won some bytes.

At least I can now predict what the display will show when locking at a functions name.
Naming of the functions - formats - parameters is a point I would like do discuss.

have a preview at:
https://github.com/AnHardt/Marlin/blob/xtostr/Marlin/ultralcd.cpp#L1522
or follow the development at:
https://github.com/AnHardt/Marlin/compare/xtostr?expand=1

thinkyhead, 2015-02-10:
This is quite lovely looking code. I'm concerned about the computational cost more than the program memory cost. It's just that these functions are used particularly during the LCD drawing, and I'm not alone in wanting to reduce time spent in that codepath as much as we can.

Is there any way to profile this to see how much time it takes versus the existing code? If it costs significantly more then it might be possible to re-do the generic functions as macros so that they generate concise functions. The compiler would strip out unreachable code that the invariant constant arguments would produce. (`if (DEFINE_EQUAL_TO_3 == 6) { … }` will be stripped, for example)

thinkyhead, 2015-02-10:
Along the same lines, does the compiler optimize separate instances of `inline` functions in that way? If so, it would be cleaner to `FORCE_INLINE` the generic functions.

AnHardt, 2015-02-11:
Will fork here to make some benchmarks.
Serial output. Triggerd by a abused gcode.

galexander1, 2015-02-11:
Inlining generally precedes constant propagation. So yes, the compiler
will generally optimize each instance of an inline function to match its
context.  But it might be difficult to convince gcc to inline it with -Os
(optimize for size).  You can try __attribute__((always_inline))...but
size is important too.

I haven't benchmarked it myself but my guess is that when it comes to
these comparisons you're chasing a mythical 1% improvement.  Remember
Amdahl's law.  You have to find the expensive mistake, and fix it -- the
1%s here and there don't matter.  Even inside a loop, 1% * 500 still
becomes only 1%, because the other 99% also multiplies by 500.

If there's not something in the low-level I/O that's literally 5x as slow
as it should be, then the solution to this problem is probably to update
only X% of the graphical display at a time.  Somewhere there's a
   for (y = 0; y < height; y++)
that loops 128 times or whatever, make it only do 16 at a time and then
return to the main marlin loop.

If you are willing to look at the datasheet or reference manual for the
atmel you have, you can find out how many timers it has.  Odds are good
that it has an extra timer Marlin is not already using that you can set
up as a cyclical counter that you can use for benchmarking.  Failing
that, you can probably read the counter register associated with a timer
that is already in use.

thinkyhead, 2015-02-12:
Yeah, this probably isn't the most major piece of code that I will worry about as far as cycle shaving. The compiler will do some smart things. The best way I know of "profiling" this kind of situation is to just look at what the compiler generates.

ntoff, 2015-02-16:
>It's just that these functions are used particularly during the LCD drawing, and I'm not alone in wanting to reduce time spent in that codepath as much as we can.

I'd be quite happy if the LCD only updated say every 10 seconds while the print was running  and nothing is being touched, but once you press a button to access the menu (to change filament or tweak a setting) the refresh happens faster. I'd also be happy to lose the stupid X Y Z positions during a print as they're useless at the speed they update anyway. I'd rather it just be blank, or display some static text like "printer in use" and have the extra computation power go to running the print functions.

alexborro, 2015-02-16:
I think moving the LCD routine to a ISR will help a lot.
And @ntoff are right, just make the LCD responsive when moving through menus. 

Cheers.

Alex.

bkubicek, 2015-02-16:
four years ago I started a challenge to have a interupt driven LCD library.
However, you still only want to transmit the letters that change, and you
do not want 20x4 bytes as buffer in memory.  A third interrupt is also a
bad idea, IMHO.
Thats why it was never included.

Bernhard

AnHardt, 2015-02-16:
Benchmarking is working now.
But over the weekend, at the dutch shore, not staring at the code, i got several new ideas to test. This will last for a while.

thinkyhead, 2015-02-20:
Let's move this conversation over to the PR when that is submitted. It's been slow to get feedback on PRs. Clearing open issues at the moment to make room for a … dare I say? … code freeze.

closed 2015-02-20 by thinkyhead

thinkyhead, 2015-02-27:
@AnHardt Do you still have that gist? I want to try messing with those conversion functions and see how they perform, code size, etc.

AnHardt, 2015-02-27:
@thinkyhead 
Uploaded two of my branches again.
https://github.com/AnHardt/Marlin/tree/xtostr is pretty much the original idea. Was reasonable faster then most of the original functions but was a bit limited when dealing with 5 and more digits - when int overflowed.
In  https://github.com/AnHardt/Marlin/tree/xtostr-bench i build an infrastructure around it, to test for speed and experimented a lot but got stuck somehow. Wanted to play with pow() and isolating the part behind the dot with modf(), but ... . Ah - Misused M42 to init the benchmark. Output is on the serial console.