MarlinFirmware/Marlin #1485: How to keep the stepper motors from disabling after print is finished/cancelled?

Natealus, 2015-02-09:
This may be an odd question as I'm not sure it applies to every design. But my Rostock delta used to hold up its own weight even with the steppers disabled. But with my Kraken and all the wiring and water associated with it...the effector is quite a bit heavier than it used to be. The disabled tension can't hold it up and it can drop one of the nozzles into a newly finished print. 

I wish I could find a spot in the firmware/slicer to make it so that the stepper disable event after a print is completed/cancelled doesn't happen. So it would complete, the XYZ towers would home, and the XYZ steppers remain enabled for the timeout defined or I manually disable them. The extruder steppers could disable there's no reason for them to stay enabled.

I noticed in the firmware there is a possibility of it working on an SD print but I'm having stuttering issues with the Viki 2 enabled so I haven't even tested that functionality. I use MatterControl to print via USB all the time. Is it happening in the firmware or the slicer? Cause I took the disable code out of the slicers custom G-code spot...and I don't see it if I watch the g-code terminal. So it makes me think it's a default routine in the firmware after g-code finishes.

shaggythesheep, 2015-02-09:
Hi Natealus,

There's two ways of achieving what you're after:

1. Edit the firmware - in Configuration_adv.h there's a define called DEFAULT_STEPPER_DEACTIVE_TIME; it is set by default to 60 (seconds) but you should be able to set it to whatever you want (within reason)

2. via G-Code - if you'd rather not edit the firmware, you can send the M84 command with an S parameter to set the timeout. For example; M84 S300 would set the motors to disable 5 minutes after the last move.

Hopefully I've understood you properly and this helps.

Richard

thinkyhead, 2015-02-09:
Another trick, 3 magnets at the top strong enough to park your carriages, but easily overcome by the motors.

AnHardt, 2015-02-09:
@Natealus 
Set DEFAULT_STEPPER_DEACTIVE_TIME to 0!
Not documented but in the code https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L4411

Natealus, 2015-02-09:
Yeah right now I have it set to about an hour but it disables as soon as the homing event is done after finishing a print. I'll try the deactivate time to 0 and see if that works...if not Shaggy's idea of putting the timed variable at the ending g-code section code is another good idea. And @thinkyhead the magnet idea does seem like a good solution too I might prototype an attachment to do that.

Natealus, 2015-02-10:
I tried taking the stepper deactivate time to 0...holds if I manually initiate it there. But still the same issue with the print finishing and it disabling the stepper motors. I originally thought it was the custom g-code section in MatterControl but I took the M84 command out of the end of the line and it still does it.

thinkyhead, 2015-02-10:
@Natealus Thanks for taking a look at the behavior. Surely it must be something obvious… I'll take a look at the code.

Natealus, 2015-02-10:
Yeah cause for sure we would like the extruder steppers to be disabled but if there was a way we could have a define option to leave the movement steppers on after a print is finished/cancelled...that would be a tremendous help for me so I can let prints finish unsupervised. I have to catch it at the very end and manually send a homing command after the sequence is complete before the hot nozzle falls in my print lol.

On a side note...first print with wood. :) Heart shaped box for my sweetie.

thinkyhead, 2015-02-10:
That's why they call it "Lay Woo."

shaggythesheep, 2015-02-10:
Printing wood? Sounds awesome!

I don't suppose you have an M81 lurking at the end of your G-Code? That's the only other place I can see that would cause the steppers to be disabled.

shaggythesheep, 2015-02-10:
Also, I'm sure you're already aware of this but you can send M84 E to disable just the extruder steppers (once we've worked out why all of your steppers are being disabled)

nophead, 2015-02-10:
I think you will find the problem is caused here:
https://github.com/MarlinFirmware/Marlin/blob/32d4dd073fefcbd22c3299f0add6830efb0b71d7/Marlin/Configuration_adv.h#L291

galexander1, 2015-02-10:
@Natealus It sounds a bunch to me like there is something in your G-code
epilogue that is causing you troubles.  Can you provide us with the epilogue?

galexander1, 2015-02-10:
@nophead Oh my god, I bet you're right.  What idiot decided that SD card
users are incapable of providing their own epilogues?

Natealus, 2015-02-10:
@nophead Thats a good find unfortunately I have yet to print from SD card. I've been solely printing from my computer. And I've already tried changing that define to false to see if it magically affects this issue but it doesn't seem to.

I did look through last night's print log. Even with M84 absent from Every part of the program and custom g-codes...

->N34024 G1 F1800 X9.557 Y8.199 E2027.51591*53

<-ok
->N34025 G0 F18000 X9.488 Y8.152*116

<-ok
->N34026 G1 F1800 X8.504 Y9.857 E2027.58136*58

<-ok
->N34027 M107*39

<-ok
->N34028 G1 F2400 E2023.08136*49

<-ok
->N34029 G1 Z28.7*77

<-ok
->N34030 G0 F18000 X8.505 Y9.857 Z33.499*24

<-ok
->N34031 M104 S0*96

<-ok
->N34032 M104 T0 S0*39

<-ok
->N34033 M140 S0*98

<-ok
->N34034 G91*17

<-ok
->N34035 G1 Z10 E-5 F12000*122

<-ok
->N34036 G90*18

<-ok
->N34037 G28*16

<-ok
->N34038 M114*43

<-X:0.00 Y:0.00 Z:366.87 E:2018.08 Count X: 601.49 Y:601.49 Z:601.49
<-ok
->N34039 M105*42

<-ok T:235.1 /0.0 B:70.0 /0.0 T0:235.1 /0.0 T1:29.2 /0.0 T2:37.4 /0.0 T3:42.4 /0.0 @:0 B@:0
->M84

<-ok
->M105

<-ok T:234.5 /0.0 B:70.0 /0.0 T0:234.5 /0.0 T1:29.2 /0.0 T2:37.4 /0.0 T3:42.3 /0.0 @:0 B@:0
->G28 X0 Y0 Z0

The last event is me manually homing the towers again. There's that bugger of an M84. I certainly didn't give it permission to be there lol.

galexander1, 2015-02-10:
@Natealus So the M84 is in the G-code file?  It sounds like your slicer is putting it there...

Natealus, 2015-02-10:
@galexander1 Thats just last nights print log. The settings where I tell it what g-code I want to happen after the slicer is done or cancelling or starting, I've taken the M84 out of there but for some reason its still getting put in.

galexander1, 2015-02-10:
@Natealus The G-code file you send to the printer is a regular text file.
You can look at it in Notepad or Firefox or anything.  Scroll to the
bottom.  Does it have M84?  If so, the problem is the slicer.  If not,
the problem is probably whatever host software you are using to send
commands to the printer.  Most likely, the problem has nothing to do with
Marlin.  Good luck!

Natealus, 2015-02-10:
What're other people's setup? Do other people use MatterControl or other slicers over USB? Or is it most common for people to export to sdcard for a standalone print? I can see after you export the gcode it could be looked over and modified there so this doesn't happen. But at the moment I can manage better delta segments smoothly by not enabling the LCD subsystem...hence why I'm still preferring computer control. Yeah thats what we were looking at was the log and the end does stick the M84 there. I think mattercontrol and matterslice are both integrated...so its one of them doing it since it's in the log I believe.

Natealus, 2015-02-10:
I hesitate to try Kissslicer again last time it dove the nozzle into the bed without homing first so I need to study it better. But I've heard it slices really well.

thinkyhead, 2015-02-10:
I'm partial to Cura, but _shhh_ don't tell Braam. I seem to remember being annoyed that Slic3r (used to?) insert things into the epilogue that I couldn't disable. So, I have a crazy idea. How about an option to _actually ignore these_ at the end of the print?

Natealus, 2015-02-11:
Well so I tried Shaggy's idea with putting M84 S300 at the ending g-code section of the slicer...and it still does the exact same thing. -.-

The M84 rolls by itself after that command and kills them just the same. Sounds exactly like that Slic3r issue.

shaggythesheep, 2015-02-11:
Looking at the MatterControl source code (https://github.com/MatterHackers/MatterControl/search?utf8=%E2%9C%93&q=m84) you may be able to edit the config file for your particular printer and remove the M84 command.

I've not installed it, so can't verify this but it looks like it could work

galexander1, 2015-02-11:
@thinkyhead No!  Slicer bugs are not our domain.  Do not break Marlin in
response to broken slicers.

I can't believe I'm saying this.

thinkyhead, 2015-02-12:
@galexander1 Never shy away from at least throwing out the naïve idea, I say.

So, a better idea would be to add `M17` to your GCode epilogue, then turn off the extruder motor separately, which could be accomplished by extending M18 to take axis codes as extra arguments.

Natealus, 2015-02-12:
He wasn't talking about modifying the way Marlin works at it's core...just an option to define. It's a small problem that's oddly a large issue. But I'm sure there's a more elegant fix.

We know now that it's not a Marlin issue, but it doesn't hurt to ask other people what they use or how they suggest fixing the issue.

shaggythesheep, 2015-02-12:
@Natealus Did you get a chance to try editing the config file for your printer in MatterControl? 

Also, if you're going to have to edit the G-Code to add in the M17 command, why not just remove the M84 that's causing the issue?

@thinkyhead With respect to extending M18, I'm pretty sure M84 has all the necessary code ready to be copied across to save you having to write it from scratch.

galexander1, 2015-02-12:
@Natealus It's a large issue *in your slicer*.  It is not an issue in
Marlin at all.  Your slicer is asking Marlin to turn off the motors, then
Marlin turns off the motors.  Marlin is working correctly.  We do not
provide support for broken slicers here.

Natealus, 2015-02-12:
@shaggythesheep  when I was looking at the gcodes document last night it looked like M17 and 18 work slightly differently than M84 but its still pretty much the same thing. Its something coded into the program outside of the GUI cause I've reworked all that stuff. So yeah I was going to take a look at the MatterControl github.

@galexander1 Yes dude we realize it is not a marlin issue. If you have nothing else constructive to add, I've already heard your complaint in several posts.

galexander1, 2015-02-12:
On Thu, Feb 12, 2015 at 12:09:33PM -0800, Nate Stokes wrote:
> @galexander1 Yes dude we realize it is not a marlin issue. If you have
> nothing else constructive to add, I've already heard your complaint in
> several posts.

If you realize it's not a Marlin issue, shut up already.  Not being nice
because you're not respecting this forum.

closed 2015-02-12 by galexander1

shaggythesheep, 2015-02-12:
@Natealus I've just installed MatterControl on my PC to try to work out this printer config isssue. I'm assuming you're on Windows but I guess the steps will be quite similar on Linux or Mac.
When you added your printer, you will have selected a type from the dropdown. Make a note of this type and then navigate to `C:\Program Files (x86)\MatterControl\StaticData\PrinterSettings\` and then choose the folder that matches the printer you selected from the dropdown.

Within that folder you'll find a file called `config.ini` - inside there's a line that says `end_gcode`. Edit that line to remove the reference to M84.

For example; for a Blue Eagle Labs Kossel Clear the end_code line is this
`end_gcode = G28; home\nM104 S140; \nM190 S80;\nM84; disable motors`

This should be changed to:
`end_gcode = G28; home\nM104 S140; \nM190 S80;`

It's probably best to do this with MatterControl closed to ensure that the changes you make stick. You may also have to make these changes again if you update MatterControl. As it's open source, maybe it would be a good idea to submit a pull request for your particular printer, stating the issues that you've had.

I have to agree with @galexander1 that Marlin should not have a load of 'fixes' for slicers that generate inappropriate G-Code otherwise it'll just end up an unmaintainable mess.

thinkyhead, 2015-02-13:
Extending `M17` and `M18` is non-impactful, and that kind of control will be useful. It is also consistent with other commands that allow finer control, so it's a good thing to include in Marlin. And it just happens that it would supply a remedy for badly-behaved slicers – or as it is in this case, not the slicer but the Host Software.

@galexander1 @Natealus I appreciate this topic being continued because it suggests ideas which otherwise would never have come up, and the opportunity to think about how to accomplish a remedy while usefully enhancing Marlin. I don't like seeing users being discouraged from seeking solutions at the firmware level, where a host or slicer is merely showing the potential for hosts and slicers to misbehave. (We have special thermal protection for this reason.) On the other hand, you have to let the guardians do their job without giving them grief for being protective.

Natealus, 2015-02-13:
How did I cause any grief? I was discussing a topic that may not affect everyone but obviously there is interest in it.

@galexander1 If you don't feel it is a valid use of your time on the forum then you should look at other issues and not comment at all. You speak of respect? You've been very disrespectful to me Galexander1 and I've held my tongue until now.

galexander1, 2015-02-13:
Did I miss something?
What change does M17/M18 need?

thinkyhead, 2015-02-13:
@galexander1 I suppose that depends on who you ask.

shaggythesheep, 2015-02-13:
Sorry @thinkyhead, I was referring to a suggestion a while back about a custom M code to disable `M84`. I agree that M18 could be easily extended to allow finer control of which motors are disabled - I know this functionality is already there in M84, but when looking at the list of G-Codes and their descriptions Disable Steppers makes more sense to me than Stop Idle Hold. If broken out into a function and called from M84 and M18 it wouldn't have any real impact on the size of the firmware either.

@Natealus Have you taken the time to try out the solution I gave you yet?

galexander1, 2015-02-13:
M18 and M84 are already the same in Marlin.

Seriously, guys, wtf.

shaggythesheep, 2015-02-13:
Ha, so it is... I saw the 84 case and totally missed the 18!

Natealus, 2015-02-13:
@shaggythesheep I did look through and find the Seemecnc Rostock config.ini in there. Changed the ending g-code line but it still happens. Looking through the logs it seems pretty sure that the host is injecting it into the end after the g-code completes because it's absent from the generated g-code. So I'm pretty sure that if I had exported to SD that it would actually work as desired. Seems like something hardcoded into MatterControl.

shaggythesheep, 2015-02-15:
Yep, I think this is your issue:
https://github.com/MatterHackers/MatterControl/blob/27032dae48b665be596076df34e5ac65c788850d/PrinterCommunication/PrinterConnectionAndCommunication.cs#L2124

Might be worth raising an issue with them over on their GitHub. In the meantime you can fix your particular issues by commenting out this line in your local copy of Marlin:
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L2766

It'll mean you'll have to use M18 instead of M84 but at least you won't have to be ready to catch your extruder at the end of a job.

Natealus, 2015-02-17:
Wow yeah that does look like the spot perhaps I can just comment out that particular section in mattercontrol and recompile. I'm guessing it's probably something that I need Visual Studio for right?

shaggythesheep, 2015-02-17:
To be honest I have no idea - I've never really coded in Visual Studio, but from the looks of it that might work. Unless you have some experience with VS I wouldn't bother - for your case it's probably easier just to edit your local firmware.

Natealus, 2015-02-24:
Sorry been gone a few days. Yeah I just don't think the firmware is causing it though, seems like MatterControl's automatic response to a print being done or cancelled. One thing that definitely helped though is retentioning my belts and it has a much firmer grip so the effector doesn't fall nearly as much.

shaggythesheep, 2015-02-25:
Oh it's definitely MatterControl causing the issue - my suggestion was just to make Marlin ignore any M84 commands that MatterControl sends - this will 'fix' your issue until MatterControl are able to fix their bug.

icebluscorpion, 2016-11-13:
@Natealus that is a technical issue u must adjust the tension of the belt wheels and the tension of the belt it self to hold up its own weight even with the steppers disabled... so it will fix this issue... anny way don`t mess with marlin coding to fix mechanical issue because the M84 is to prevent overheating of the steppers

zakamik, 2017-01-27:
Hello,
I have the same issue - Marlin immediately releases steppers after printing file from SD card. No USB connection. no M84 command.
It is not entirely clear from the thread - is releasing steppers by firmware after SD print considered as expected behavior?
If not, should I raise a separate issue for that?

Blue-Marlin, 2017-01-27:
Configuration?adv.h
```
  #define SD_FINISHED_RELEASECOMMAND "M84 X Y Z E" // You might want to keep the z enabled so your bed stays in place.
```

zakamik, 2017-01-28:
Great, thanks. Hope this works.

jerryajay, 2017-05-15:
Check this work. They seem to have done it: http://dl.acm.org/citation.cfm?id=3037752

sunnyxiong2020, 2020-12-04:
update this value in the configuration_adv.h, I just changed mine from 120 to 1800
#define DEFAULT_STEPPER_DEACTIVE_TIME 1800