MarlinFirmware/Marlin #1491: Syntax Errors in  Marlin_main.cpp?

iman7, 2015-02-10:
Finally was time to upgrade my old Prusa i3 with RAMPS 1.4 to auto bed leveling. I've only used the pre-made configurations from MakerFarm and its my first time manually setting up Marlin. 

So changed the corresponding values in configuration h, enabled auto bed leveling, and swapped out some pins for the heater bed since the mosfet for that blew and I'm using the extra pins normally reserved for a second extruder. 

Any help in fixing this would be appreciated. 

```
Marlin_main.cpp:194: error: expected `}' before numeric constant
Marlin_main.cpp:194: error: expected ',' or ';' before numeric constant
Marlin_main.cpp:194: error: expected declaration before '}' token
```

The general area around the lines in question seem fine and are exactly the same a freshly extracted Marlin, which compiles fine. 

```
float homing_feedrate[] = HOMING_FEEDRATE;
bool axis_relative_modes[] = AXIS_RELATIVE_MODES;
int feedmultiply = 100; //100->1 200->2
int saved_feedmultiply;
int extrudemultiply = 100; //100->1 200->2
int extruder_multiply[EXTRUDERS] = { 100
  #if EXTRUDERS > 1
    , 100
    #if EXTRUDERS > 2
      , 100
	    #if EXTRUDERS > 3
      	, 100
	    #endif
    #endif
  #endif
};
```

Here's pastebins of my [configuration.h](http://pastebin.com/6586NxVR) and the [pins file](http://pastebin.com/edrzHay0) that I edited. 

Thanks.

brainscan, 2015-02-10:
I'm looking at this on my phone so my apologise if this isn't anything but I can't easily check it right now. 
In your configuration.h file you have spaces in the movement settings that may be causing a problem. 
You currently have (50*6    0,50*60,50,0). 

Sorry if this isn't the problem.

AnHardt, 2015-02-10:
Try 
BOARD_RAMPS_13_EFB in Configuration.h
and swap only 8 and 9 in the lines 67 and 106 in pins_RAMPS_13.h

thinkyhead, 2015-02-10:
+1 @brainscan gets the prize!

iman7, 2015-02-11:
@brainscan Well that fixed it thanks for the help!

closed 2015-02-11 by iman7

deva1974, 2020-06-13:
#include <IRremote.h>
const int RECV PIN=5
IRrecv irrecv(RECV_PIN);
decode_results results;
const int Pin One = 2
const int Pin two = 3
const int Pin three = 6
const int Pin four= 7

void setup(){
  irrecv.enableIRIn();
  irrecv.blink13(true);
  pinMode(Pin one, OUTPUT);
  pinMode(Pin two, OUTPUT);
  pinMode(Pin three, OUTPUT);
  pinMode(Pin four, OUTPUT);
}

void loop ()
{

  if (ir recev.decod(&result)){

    switch(results.value 1FEE01F){
          case : //Keypad button "0"
          digitalWrite(redPin, HIGH);
          delay(2000);
          digitalWrite(redPin, LOW);
          }

        switch(results.value 1FE50AF){
          case : //Keypad button "1"
          digitalWrite(greenPin, HIGH);
          delay(2000);
          digitalWrite(greenPin, LOW);
          }
          switch(results.value 1FEF807){
          case : //Keypad button "3"
          digitalWrite(redPin, HIGH);
          delay(2000);
          digitalWrite(redPin, LOW);
          }

        switch(results.value 1FE30CF){
          case : //Keypad button "4"
          digitalWrite(greenPin, HIGH);
          delay(2000);
          digitalWrite(greenPin, LOW);
          }

        irrecv.resume(); 
    }

  }
}

deva1974, 2020-06-13:
Hi..Anybody help me...while compiling the above sketch there is a error expected ')' before numeric constant.....
Regards
Dev

ellensp, 2020-06-13:
This is for Marlin firmware issue. NOT generic Arduino help!
See https://www.arduino.cc/reference/en/language/structure/control-structure/switchcase/