MarlinFirmware/Marlin #1492: LCD messages too short when using cyrillic

Nprod, 2015-02-11:
I've started adding bulgarian language support for the Hitachi HD44780 type character displays and have noticed that if the string is longer than about 9-10 characters, it gets cut off with plenty of room left on the screen. That makes it hard to add meaningful words to the menu without using a lot of abbreviations. In language_ru, quite frankly, it looks ugly since the "arrow" icons at the end of each menu entry don't line up with eachother and there is a lot of unused space on the right side.

Is this due to the 17 character limit for menu messages? Perhaps since cyrillic letters get converted by LiquidCrystalRus.cpp into a table address like "0x54", that causes the 17 character limit to get hit more easily?

thinkyhead, 2015-02-12:
@Nprod It's possible the code there needs attention, if it is only counting bytes and not characters which might be 2 bytes long. Ostensibly the language support is there for utf-8 but we seem to use single-byte character codes almost everywhere. I don't know what the situation is with Cyrillic character displays, but I assume they provide single-byte character versions of all the most common characters...

Nprod, 2015-02-12:
That could be the case - it's like the length of the menu's label string is being reported way shorter than it's supposed to be. It's not related to the type of LCD display used because when using a non-cyrillic display it only substitutes the missing symbols for japanese katakana characters while the length of the menu item remains the same. 

Something worth mentioning here is that language_ru.h is the only file where i've seen some messages followed by a number of white spaces and  "\x7E" (arrow right). Example: 
```
#define MSG_CARD_MENU                       "Меню карты      \x7E"
```

thinkyhead, 2015-02-13:
@Nprod Is it just the weird strings (arrows at the end, extra spaces, etc.) that are cut off? I posted a __strlen_utf8()__ function on another issue, I'll dig that up and you can try it on Cyrillic...

thinkyhead, 2015-02-13:
int my_strlen_utf8_c(char *s) {
       int i = 0, j = 0;
       while (s[i]) {
         if ((s[i] & 0xc0) != 0x80) j++;
         i++;
       }
       return j;
    }

Nprod, 2015-02-13:
@thinkyhead I don't think strlen is to blame here, because after looking into ultralcd_implementation_hitachi_HD44780.h i've noticed that not all the menu drawing functions use it. In some cases like line 635 you have a loop that's looking for a null character:
```
    while( ((c = pgm_read_byte(pstr)) != '\0') && (n>0) )
    {
        lcd.print(c);
        pstr++;
        n--;
    }
```
In any case, i'm attaching some photos to better illustrate the issue. In the first picture you have the messages on the left and the way they're being displayed on a cyrillic LCD on the right. The second picture is the same scenario, but using an LCD display that doesn't support cyrillic characters. The only thing that has changed are the symbols.

[image]
[image]

thinkyhead, 2015-02-15:
Huh, so if you were to add code to print out each byte that's being printed, do you think there would necessarily be a zero at those points where the string is being cut off?

Nprod, 2015-02-15:
@thinkyhead I've been messing with arduino stuff for only a week but that would be my guess, yes. Strings containing only latin characters don't get cut off, in the same language file. Can you tell me how to modify it to print out each byte?

thinkyhead, 2015-02-16:
@Nprod There are two possibilities actually. (1) Some characters are cues that the next byte means a different character (0xFF ?). (2) Some bytes which are zero are the second byte of a two-byte character code.

So, to see the bytes on the serial output (e.g., Pronterface), try inserting SERIAL_ECHO commands into loops in `ultralcd_implementation_hitachi_HD44780.h` where it's sending characters to the display... (Hopefully this works!)

    while( ((c = pgm_read_byte(pstr)) != '\0') && (n>0) )
    {
        SERIAL_ECHO_START;
        SERIAL_ECHOLN((int)c);
        lcd.print(c);
        pstr++;
        n--;
    }

thinkyhead, 2015-02-16:
This was brought up in #344 a couple of years ago. Will be good to see it fixed.

Nprod, 2015-02-16:
@thinkyhead Ok, i got a dump of all the characters and the first thing that grabs attention is that all cyrillic strings output their characters as negative integers. I've made the second menu item to be all-latin ("ABCDEFGHIJKLMNOPQ") and it's the only "positive" sequence in the lot.
```
echo:-48
echo:-99
echo:-48
echo:-80
echo:-47
echo:-121
echo:-48
echo:-80
echo:-48
echo:-69
echo:-48
echo:-75
echo:-48
echo:-67
echo:32
echo:-48
echo:-107
echo:65
echo:66
echo:67
echo:68
echo:69
echo:70
echo:71
echo:72
echo:73
echo:74
echo:75
echo:76
echo:77
echo:78
echo:79
echo:80
echo:81
echo:-48
echo:-99
echo:-48
echo:-80
echo:-47
echo:-127
echo:-47
echo:-126
echo:-47
echo:-128
echo:-48
echo:-66
echo:-48
echo:-71
echo:-48
echo:-70
echo:-48
echo:-48
echo:-99
echo:-47
echo:-113
echo:-48
echo:-68
echo:-48
echo:-80
echo:32
echo:-48
echo:-102
echo:-48
echo:-80
echo:-47
echo:-128
echo:-47
echo:-126
```
This must be related to issue #344 but some messages can be more than 8 characters, as shown in my photo.

AnHardt, 2015-02-16:
@Nprod 
I have a possible fix. Please test  #1508.

There is a small possibility i can provide a implementation for full graphic displays if you cold provide me with a link to the datasheet for your Russian character display. I need the pictures and the matching character codes to make a new conversation table into a dogm-font.

AnHardt, 2015-02-16:
[image]

Nprod, 2015-02-16:
@AnHardt Your fix worked beautifully, thanks! So it was a fault with strlen after all?

You can find the datasheet here, font table is on page 14 - http://store.comet.bg/download-file.php?id=466

Displays with cyrillic support are surprisingly hard to find, the most common are by Ampire and a few by Winstar and Orient Display. The font table is the same for all of them though. What is "test multiline" for by the way?
[image]

AnHardt, 2015-02-16:
@Nprod 
It was a combination of a not unicobe able strlen() and the part you discovered.
thest multiline was smugled in by https://github.com/MoonCactus/Marlin-JFR-features/commit/85e5aa4011cf65c0b09a4453bd1f3ac3752ba072#diff-34f7cdd618621ae7d8c113cf0b86a2f4R350
and will vanish soon - i hope.
Thanks for the link.

AnHardt, 2015-02-16:
And an other annotation.
The 17 chars (rule) in the head of the language files is more a rude rule of thumb. For 16 chars wide displays this obviously can not be correct.
It's more like: display width - select column - (back-sub menu column or ':') - may be value length - may be other appended strings - one UNUSED COLUMN on the 20 chars width display (what i will investigate in the near future)

Nprod, 2015-02-16:
Yeah i noticed that the 17 character limit is more of a guideline to make the menus look right. Ok, with this sorted, what about cases where a Hitachi 44780 type screen is used via PCF8574 extender? Then you need to use LiquidCrystal_I2C instead, but it doesn't support russian. I found a version that does here - https://github.com/treasure33/ARDUINO_LIB/tree/master/LiquidCrystal_I2C

What would be best - should this updated library be included with Marlin or should the user be expected to update it in their Arduino IDE? It's a very specific case that's useful only for electronics like Melzi and maybe Sanguinololu where the available I/O pins can be scarce.

thinkyhead, 2015-02-19:
Good job @AnHardt. That's the fix we were looking for.

@Nprod I'm not sure, but as I understand it the `LiquidCrystalRus.cpp` and `LiquidCrystalRus.h` are included to deal with the lack of support in __LiquidCrystal_I2C__.

Nprod, 2015-02-19:
@thinkyhead From what i understood, LiquidCrystalRus is based on the standard LiquidCrystal library from the Arduino IDE and it only applies when the LCD is connected using the standard parallel interface. If you declare that you're using an I2C interface with something like **LCD_I2C_TYPE_PCA8574** then it defaults to LiquidCrystal_I2C.h which doesn't support russian. I've linked to a modified version of it in my previous post which supposedly has that feature, but i haven't been able to test it yet. If it's not too different, it might be worth having it here instead.

thinkyhead, 2015-02-19:
Hmm, if it's good enough we should include it in the Marlin package. Do you think it supersedes  `LiquidCrystalRus.cpp` and `LiquidCrystalRus.h` ?

Nprod, 2015-02-19:
@thinkyhead Hmm no, it's a different use case - controllers like the Panelolu2, or one of these black I2C extender boards that attach to the back of the LCD module: http://img.alibaba.com/img/pb/224/728/686/686728224_079.jpg It's a popular choice because it reduces the LCD pin count from 8 to 4 and leaves room to hook up other things like an encoder, extra fans etc.

thinkyhead, 2015-02-19:
We should definitely make note of it in the Configuration comments for `LCD_I2C_TYPE_PCA8574` concerning support for Russian, that the other library is needed. We can include the alternative library too, we just have to name its root folder differently.

Nprod, 2015-02-19:
Yeah, we could rename it to LiquidCrystal_I2CRus and switch to it automatically when russian language is selected, the way it's currently done between LiquidCrystal and LiquidCrystalRus. But if it supercedes LiquidCrystal_I2C it may be better to just let the user know that they should update their library. In any case i have to test it first. As soon as my I2C expander arrives in the mail, i'll report back here.

AnHardt, 2015-02-19:
I did not analyse the *Rus.* -files to the end. But as far as i see they mainly deal wit the display hardware. The language specific part consists of a conversion table and a few lines of code to convert from Unicode to the displays charset. (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/LiquidCrystalRus.cpp#L16
https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/LiquidCrystalRus.cpp#L297)
I think this could be dealed with in the *_inplenentation_* files. 

The remaining hardware dependent things feel (for me) somehow as if integrated at the wrong place. Maybe as special cases in ultralcd_implementation_hitachi_HD44780.h, where the i2c things could be wrapped around it. But this is far beyond my current scope.

While locking at the code i think - there are not hardware differences but the charset.

Nprod, 2015-02-19:
@AnHardt Yes, the hardware is identical, only the font table in ROM is different. Cyrillic support is part of the original Hitachi HD44780 specification and it's up to the manufacturer to decide which one to flash. The developers of the standard arduino libraries (LiquidCrystal, LiquidCrystal_I2C) probably just didn't bother to support unicode because they never needed it. If you ask me the russian versions are better in every way since all they do is add that missing functionality. But who knows, maybe there's some hidden compatibility issues i'm missing...

AnHardt, 2015-02-19:
If the difference is only the charset i have a solution. Withe that one could use the normal i2c stuff. Please stand by.

AnHardt, 2015-02-20:
@Nprod 
Would you please be so kind to test https://github.com/AnHardt/Marlin/tree/rus-remove-extrafiles?

Nprod, 2015-02-20:
@AnHardt Hmm no, didn't work. It outputs the same type of broken text as what happens when you just add cyrillic text to language files that don't use LiquidCrystalRus. Perhaps theres something missing in the new lcd_print()?

Nprod, 2015-02-20:
@AnHardt I just got it running successfully with an I2C interface. The library i posted before doesn't work correctly, but i found a more recent one: 

http://blockduino.blogspot.com/2014/03/LiquidCristalRusI2C.html

This library works with both english and russian languages so it can be used as a universal drop-in replacement. The only changes that need to be made are to replace any occurences of **"LiquidCrystalCyr_I2C"** to **"LiquidCrystal_I2C"**, as i've done here - http://nik.benzinestudios.com/Razni/LiquidCrystal_I2C.zip

By the way i noticed that there is a noticeable 1-second lag between clicking the encoder button and the main menu appearing when using I2C compared to parallel. Is this a fault with my cheapo port expander or Marlin's implementation?

AnHardt, 2015-02-20:
@Nprod 
Updated the test-branch. https://github.com/AnHardt/Marlin/tree/rus-remove-extrafiles
 It's a good idea to use UNSIGNED chars when comparing with > 0x80! Sorry.
Would you please try again.

Sorry the I2C-stuff is outside my scope. I have no hardware to test.

Nprod, 2015-02-20:
@AnHardt Works great! Your implementation makes it so much cleaner now, and i can safely add a new cyrillic language without the need to switch to LiquidCrystalRus in ultralcd_implementation_hitachi_HD44780.h, nice!

I will test it using I2C now - if my hunch is correct, it should work out of the box without changing any more code.

Nprod, 2015-02-20:
@AnHardt Tested it with I2C - no problems. Consider your fix a complete success :)

thinkyhead, 2015-02-21:
Will this mean that LiquidCrystalRus can go away? :pray:

AnHardt, 2015-02-21:
In the tested branch it IS deleted. 
I think my pull request (#1508) is simple enough to be a fix for the original problem and can be applied now. 
Removing ' LiquidCrystalRus.*' is a bigger thing, i'd like to do later when 'codefreze' is over. As a 'side effect' this will allow to write all the language_*.h files to be written in Unicode. At least i will provide examples for 'ru' and 'de'.
The 12854 side of the problem will have to wait until the marlin symbols have been moved to an extra font.

thinkyhead, 2015-02-22:
@AnHardt Merged #1508.

boelle, 2015-03-10:
should this one still be open ?

AnHardt, 2015-03-10:
Yes. Still working on the big solution including Cyrillic on the full graphic displays, UTF8 for all languages and Japanese. Test are going on - tooooo many possible combinations.

boelle, 2015-03-10:
np then :-D

boelle, 2015-03-10:
remember to submit pull requests as soon you have a part done so we can implement in small steps

AnHardt, 2015-03-13:
@boelle 
Easier said than done. Things are pretty geared.

I have a new preview at: https://github.com/AnHardt/Marlin/tree/utf

All planned things are in place.
Cyrillic on the full graphic displays.
UTF8 for all when wanted.
Katakana on all displays. Katakana in UTF8.
'Kanji' on full graphic displays.
German in UTF8.
Finnish in UTF8.
Tree Hitachi fonts on full graphic displays for tests.
Separate font for Marlin symbols.
Full ISO fonts for Cyrillic, Katakana, European Languages in the right size.
Documentation.
Test_language.

Maybe I forgot to list same features here.

Remaining problems:
Cyrillic an especially katakana font have to be reviewed by someone who knows how this should look like.
Kana_UTF8 has to be reviewed by someone who knows what he does.
My English has to be checked.
A little merge-conflict in language.h has to be cleaned up.

AnHardt, 2015-03-13:
Some pictures:
[image]
Cyrillic on a graphic display.

[image]
Cyrillic on a graphic display simulating Japanese Hitachi.

[image]
Cyrillic on a Japanese Hitachi.

[image]
Kana on a japanese Hitachi.

[image]
'Kanji' on a graphic display.

[image]
Kana on simulated Hitachi font.

[image]
Kana_UTF8 on ISO font. Look at the ligatures.

[image]
Deutsch mit allen Umlauten. In KLEIN!

AnHardt, 2015-03-13:
And the highlight for the europeans:
[image]
Full ISO charset on graphic displays.

[image]
Separate Marlin symbols font.

Nprod, 2015-03-13:
Wow this looks amazing. Sadly i don't have a 128x64 screen to test the cyrillic font with... What about printing the whole alphabet and posting a screenshot?

AnHardt, 2015-03-13:
@Nprod 
Here we go. Upper halve of ISO10646_5
[image]
[image]

Nprod, 2015-03-14:
OK, here is an image with corrections that need to be made in red. The letters marked with an arrow next to them need to be shifted up with one pixel. The first row in your first image has letters that are not part of the hitachi implementation and are likely for mongolian cyrillic... I'm not 100% sure how they're supposed to be, but i've made a guess anyway. 

[image]

thinkyhead, 2015-03-14:
Ah, I see this is keeping active. We could start a new thread, or just change the topic of this one...

AnHardt, 2015-03-14:
@Nprod 
Thanks for you review.
The different symbols from yesterday come from ISO10646_5. The reference was http://www.utf8-zeichentabelle.de/unicode-utf8-table.pl?start=1024 up to U+045F.
Corrected version:
[image]
[image]

The simulated Hitachi ROM font looks like:
[image]
[image]

AnHardt, 2015-03-14:
Updated https://github.com/AnHardt/Marlin/tree/utf wit the corrected fonts.

Nprod, 2015-03-14:
Now it's much better, but you forgot to add one pixel in the letter Э (U+042D) - it looks exactly like the number 3. Refer to the pic i posted earlier as to how it should look like. Apart from that, it looks very good, the simulated hitachi ROM matches the code table from the datasheet too. Are there any other fonts that need to be tested?

AnHardt, 2015-03-14:
@Nprod Now i got it. More like a big mirrored C with a - in the middle. 
The kana fonts have still to be revised. I'll make pictures later.
The 'kanji' font is unchanged but we have legal problems - see #1513.

boelle, 2015-03-14:
have any contacted the author ?

AnHardt, 2015-03-14:
This is the font to simulate Hitachi Japanese. Reference is https://www.sparkfun.com/datasheets/LCD/HD44780.pdf Page 17
[image]
[image]

The Katakana ISO font. Based on http://www.utf8-zeichentabelle.de/unicode-utf8-table.pl?start=12448 up to U+30FF.
[image]
[image]

thinkyhead, 2015-03-15:
@boelle I did contact the author and ask about resolving licensing issues. I posted his reply on another topic. The translation I got was bad, so I could only get the basic gist. I believe he was asking for a link back to the source page and a credit in the source code. But, I may be reading it badly. Maybe it is worth it to hire a translation service on the internet. A decent translation would cost under $20 on the services I looked at.

On the one hand, the GPLv3 forbids creating a distribution that removes rights, including adding a commercial use restriction. And it also forbids Marlin from taking any conditional contribution. The font and the string translations are the problem. If someone were to "clean room" a Japanese font and also create original translations, the implementation itself is not under dispute. There is a Right Way to make a language contribution, and Kana – apart from the specific translations and font – must be as it is.

boelle, 2015-03-15:
any chance i could have a go at it with google translate? not perfect i know but slicing it in bits might do the trick

AnHardt, 2015-03-15:
Dear @boelle.
Would you please be so kind to cross-read #1513.

boelle, 2015-03-15:
i did... i wanted to know what the author wrote in his native language

boelle, 2015-03-15:
oh... there it was

Nprod, 2015-03-16:
Hey i just noticed one more tiny correction that needs to be made - the letter д (U+0434) should be shifted down one pixel in order to match the way it looks on a hitachi display.

[image]

AnHardt, 2015-03-16:
OK. That's why i asked an expert. :-)

AnHardt, 2015-04-08:
@Nprod 
If you are satisfied with the current solution you can close this.

Will we have a Bulgarian language file from you?

Nprod, 2015-04-08:
@AnHardt Yes, the cyrillic support seems fine - i can't see anything wrong with the 128x64 screenshots and the hitachi display here works. If i find any other problems, i'll open a new issue.

And yes i will make a Bulgarian language file soon, just haven't had the time to work on it recently. Should i post it here when its done? It shouldn't take long.

[3 more replies]

PR #1874 (2015-05-22): Introduction of Bulgarian - language_bg.h
Thank You @Nprod!

Please close  #1492
closed 2015-04-09 by Nprod