MarlinFirmware/Marlin #1493: pins_MEGATRONICS_3.h: servo pin missing

michielz, 2015-02-11:
While testing the development firmware my servo didn't move any more. It used to work fine when I used the firmware from the firmware from [ReprapWorld] (http://reprapworld.com/?software)

After adding the following code to the pins file the servo started working again.
```
 #ifdef NUM_SERVOS
#define SERVO0_PIN 46
#endif
```

thinkyhead, 2015-02-12:
I just had a peek at the data sheet. Does this look proper...?

    // Servo support
    #ifdef NUM_SERVOS
      #define SERVO0_PIN       46 //AUX3-6
      #if NUM_SERVOS > 1
        #define SERVO1_PIN     47 //AUX3-5
        #if NUM_SERVOS > 2
          #define SERVO2_PIN   48 //AUX3-4
          #if NUM_SERVOS > 3
            #define SERVO2_PIN 49 //AUX3-3
          #endif
        #endif
      #endif
    #endif

thinkyhead, 2015-02-13:
#1496

closed 2015-02-16 by thinkyhead