MarlinFirmware/Marlin #1494: Wrong acceleration-calculation for CoreXY

Wurstnase, 2015-02-11:
Hi,

just testing my 32bit-Edition. While I'm debugging some other issues I found this bug.

For X or Y-only-moves the acceleration_rate is correct. But for diagonal moves it isn't.
At the moment the staight moves you get

[acceleartion_rate](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L916)

For diagonal moves this is by the factor sqrt(2) bigger. But it should be smaller by this factor.

alexborro, 2015-02-11:
Hi @Wurstnase , I didn't get you. Could you please discuss a little more about it..

As far as I understand, all the speed and calculation rate are based on [inverse_second](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L767), which is based on [inverse_millimeters](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L764) and this one based on [block->milliliters](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L761).

The total milliliters length is based on head displacement not on motor displacement (A or B axis).

Let me know if anything looks wrong to you.

Cheers.

Alex.

Wurstnase, 2015-02-11:
That is right. But the acceleration should not based by the head-displacment.
Acceleration is needed for reduce the torque on the motors to reach the max speed.

~
Nico

alexborro, 2015-02-11:
Nico, depends on the point..

From my point of view, the acceleration should be applied to head
(extruder) movement.
I wanna control the extruder acceleration impact on print quality, not the
motor torque.

Of course the motor should withstand your max desired head speed and
acceleration. And it usually does.
In my bots (regular cartesian and CoreXY), a 48Kg torque motor can reach
head speeds of 200+ mm/s and 9000mm/s^2, which is way higher than I use for
printing.

That is actually a Marlin problem related to Deltabots support. The
acceleration is computed based on motors movement not the effector. This
brings weird behavior during printing (my Kossel Mini says me that  ;-) )
like too hard or too soft stops of the effector.

Cheers.

Alex.

Wurstnase, 2015-02-11:
Ok, your point should be than also the same problem.

Just some numbers:
acc = 1000
step_per_mm = 80
acceleration_rate = acceleration_st * 2^24 / 2M

with a straight move your acceleration_st is 1000*80. On a diagonal move this is sqrt(2) higher. So you can't control your acceleration anyway for your suggestion.

Wurstnase, 2015-02-12:
Btw.

```
598 block->steps_x = labs((target[X_AXIS]-position[X_AXIS]) + (target[Y_AXIS]-position[Y_AXIS])); 
599 block->steps_y = labs((target[X_AXIS]-position[X_AXIS]) - (target[Y_AXIS]-position[Y_AXIS])); 
```

should be

```
598 block->steps_x = labs((target[X_AXIS]-position[X_AXIS]) + (target[Y_AXIS]-position[Y_AXIS])*axis_steps_per_unit[X_AXIS]/axis_steps_per_unit[Y_AXIS]); 
599 block->steps_y = labs((target[X_AXIS]-position[X_AXIS])*axis_steps_per_unit[Y_AXIS]/axis_steps_per_unit[X_AXIS] - (target[Y_AXIS]-position[Y_AXIS])); 
```

untested.

Wurstnase, 2015-02-12:
@alexborro,

got you now. Just takes me the hole evening... 
The complete planner.cpp could need a rework. It's really a pain to read and understand.

alexborro, 2015-02-12:
Yes, planner.cpp and stepper.cpp are quite complicated. It took me many
whole days of study to understand them.

Cheers.

Alex.

Wurstnase, 2015-02-13:
A goal could be to clean up this files. Also the delta-kinematics are not in those files. They are in the main.cpp. Some days ago I take a look into the smoothie-firmware. They have for each printertype an extra kinematic file. ( [Smoothieware/armsolution](https://github.com/Smoothieware/Smoothieware/tree/edge/src/modules/robot/arm_solutions) )

thinkyhead, 2015-02-20:
:+1: on cleaning up the files. There is more of that coming via the coding standards topic.

closed 2015-02-20 by thinkyhead