MarlinFirmware/Marlin #1497: Extruder auto fan pins are not configured

rGlory, 2015-02-13:
Looks like code to initialize extruder auto fan pins is missing. I think it should be somewhere in tp_init() from temperature.cpp something like this

 #if defined(EXTRUDER_0_AUTO_FAN_PIN) && EXTRUDER_0_AUTO_FAN_PIN > -1
      SET_OUTPUT(EXTRUDER_0_AUTO_FAN_PIN);
  #endif

etc for each pin

MagoKimbra, 2015-02-15:
When called the void is also set pin as output.

void setExtruderAutoFanState(int pin, bool state)
{
  unsigned char newFanSpeed = (state != 0) ? EXTRUDER_AUTO_FAN_SPEED : 0;
  // this idiom allows both digital and PWM fan outputs (see M42 handling).
  pinMode(pin, OUTPUT); <-------------------------- THIS LINE
  digitalWrite(pin, newFanSpeed);
  analogWrite(pin, newFanSpeed);
}

thinkyhead, 2015-02-16:
@MagoKimbra Are you saying that the code is not missing ... the line you point out does the proper thing and this is not a bug?

rGlory, 2015-02-16:
@MagoKimbra probably it set the pin when temperature reaches threshold, but at startup that pin would be in input mode. At least it looks like so on my RAMPS, and when I added code above it was set to output at startup. If this is the case I believe it would be cleaner to initialize this pin at startup.

MagoKimbra, 2015-02-16:
No for me is not a bug. But we say that it is not a clean thing done so. It would be better to make the setup of the pin in the Void Setup.
Make setup of the pin every time you use it is a logical error.

grob6000, 2015-02-16:
While I agree it's logically excessive, and not very 'clean', it is convenient to set the pin state in the function call - eliminates a whole block of code to maintain in startup.

@rGlory have you observed any erroneous behaviour as a result of the pin mode being wrong (e.g. oscillation, noise, etc.)? If so, maybe we should just call the setExtruderAutoState function on startup to set the correct state initially (e.g. what happens when you startup with a hot extruder? Does the fan run as required?). If there's no real-world impact here, I'd just leave it...

rGlory, 2015-02-16:
@grob6000 if it is convenient to set the pin state in the function call, then other pins should be set the same way. I copy paste the code for initializing other pins and modified it. I think it better should be consistent rather than convenient. But of course it is up to you guys,

grob6000, 2015-02-17:
Ah fair enough, my bad! :)

boelle, 2015-03-10:
was this a bug or?

grob6000, 2015-03-10:
I'd guess it might have been, looks like the fault is untidy code rather than anything tangible. Possibly unknown pin state until extruder gets to autofan temp first time? Whether this affects fan behaviour or presents a problem would depend on hardware (possibly a real problem if the FET driver for the fan output has no gate pull-down resistor, a high-impedance at the gate could cause oscillation).

@rGlory is right that all the other pins get initialised in the start code, this should be no exception, but it's a tidy-up rather than a bug - nothing wrong with that, perhaps he's fixed it himself, and could submit a pull request???

boelle, 2015-03-10:
i was just confused if it was a bug or not... at least could use some feedback..

but yes in general the code needs a cleanup... but bug fixing has higher prio. than a clean up... so if it was indeed a bug as such the issue would have to be marked as such so it could get sorted before other issues

thinkyhead, 2015-03-11:
So, cleanup in which direction? Initialization of pins at startup, or ensure OUTPUT on each call?

boelle, 2015-03-12:
oh... was just thinking general cleanup.... nothing spec.

thinkyhead, 2015-03-13:
Keeping this one in mind as I start on some cleanup...

boelle, 2015-05-31:
any news on this one? cold we close it since there was no real issue?

thinkyhead, 2015-06-09:
#2271 eliminates the complaint. However, this is not a bug, just a feature. The pins being in INPUT state shouldn't affect behavior or performance.

closed 2015-06-09 by thinkyhead