MarlinFirmware/Marlin #1498: compiling error expected ')' before 'STR_THERMOMETER'

tanmanguy, 2015-02-14:
Arduino: 1.0.6 + Td: 1.20 (Windows NT (unknown)), Board: "Teensy++ 2.0"
ConfigurationStore.cpp:228: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:229: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:220: warning: unused variable 'storageSize'
ConfigurationStore.cpp: In function 'void Config_RetrieveSettings()':
ConfigurationStore.cpp:353: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:354: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp: In function 'void Config_ResetDefault()':
ConfigurationStore.cpp:463: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:420: warning: unused variable 'e'
ConfigurationStore.cpp: In function 'void Config_PrintSettings(bool)':
ConfigurationStore.cpp:474: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:477: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:478: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:479: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:480: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:498: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:501: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:502: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:503: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:504: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:509: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:512: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:513: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:514: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:515: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:519: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:522: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:523: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:528: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:531: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:532: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:533: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:534: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:535: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:536: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:541: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:544: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:545: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:546: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:571: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:574: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:575: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:576: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:625: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:628: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:633: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:649: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:656: warning: only initialized variables can be placed into program memory area
ConfigurationStore.cpp:661: warning: only initialized variables can be placed into program memory area
/ultralcd.h: At global scope:
/ultralcd.h:21: warning: 'blink' defined but not used
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=106 -DTEENSYDUINO=120 -felide-constructors -std=c++0x -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\cores\teensy -IC:\Program Files (x86)\Arduino\libraries\Wire -IC:\Users\taner\Documents\Arduino\libraries\LiquidTWI2 -IC:\Users\taner\Documents\Arduino\libraries\U8glib -IC:\Users\taner\Documents\Arduino\libraries\LiquidCrystal -IC:\Users\taner\Documents\Arduino\libraries\SPI C:\Users\taner\AppData\Local\Temp\build6707453120121765007.tmp\digipot_mcp4451.cpp -o C:\Users\taner\AppData\Local\Temp\build6707453120121765007.tmp\digipot_mcp4451.cpp.o 

In file included from /pins.h:78,
                 from /Marlin.h:23,
                 from /thermistortables.h:4,
                 from /Configuration.h:829,
                 from digipot_mcp4451.cpp:1:
/pins_PRINTRBOARD.h:60:21: warning: extra tokens at end of #ifdef directive
/pins_PRINTRBOARD.h:101:1: warning: "SDSS" redefined
/pins_PRINTRBOARD.h:80:1: warning: this is the location of the previous definition
In file included from /Marlin.h:23,
                 from /thermistortables.h:4,
                 from /Configuration.h:829,
                 from digipot_mcp4451.cpp:1:
/pins.h:126:1: warning: "_E1_PINS" redefined
/pins.h:121:1: warning: this is the location of the previous definition
In file included from /Configuration.h:829,
                 from digipot_mcp4451.cpp:1:
[... 659 lines cut ...]
ultralcd.cpp:961: error: expected `)' before 'STR_h3'
ultralcd.cpp:964: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:964: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:966: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:966: warning: only initialized variables can be placed into program memory area
ultralcd.cpp: In function 'void lcd_sdcard_menu()':
ultralcd.cpp:1038: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:1043: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:1046: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:1060: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:1062: warning: only initialized variables can be placed into program memory area
ultralcd.cpp: In function 'void menu_action_sdfile(const char*, char*)':
ultralcd.cpp:1186: warning: only initialized variables can be placed into program memory area
ultralcd.cpp:1190: warning: only initialized variables can be placed into program memory area
/ultralcd.h: At global scope:
/ultralcd.h:21: warning: 'blink' defined but not used
ultralcd.cpp:66: warning: 'void lcd_control_retract_menu()' declared 'static' but never defined
ultralcd.cpp:1120: warning: 'void menu_action_setting_edit_float43(const char*, float*, float, float)' defined but not used
ultralcd.cpp:1124: warning: 'void menu_action_setting_edit_long5(const char*, long unsigned int*, long unsigned int, long unsigned int)' defined but not used
ultralcd.cpp:1117: warning: 'void menu_action_setting_edit_callback_int3(const char*, int*, int, int, void (*)())' defined but not used
[... cut ...]
ultralcd.cpp:1119: warning: 'void menu_action_setting_edit_callback_float32(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:1121: warning: 'void menu_action_setting_edit_callback_float5(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:1122: warning: 'void menu_action_setting_edit_callback_float51(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:379: warning: 'void lcd_autostart_sd()' defined but not used

thinkyhead, 2015-02-16:
Have you defined one of these in __Configuration.h__: `DISPLAY_CHARSET_HD44780_JAPAN` or `DISPLAY_CHARSET_HD44780_WESTERN` or `DOGLCD` ?

haqnmaq, 2015-03-06:
I have this same problem. I have a Reprap Discount smart controller and a RUMBA. Everything works but the lcd. It remains blank. I get errors whenever DISPLAY_CHARSET_HD44780_JAPAN or DISPLAY_CHARSET_HD44780_WESTERN or DOGLCD are enabled, and even with the errors, if I upload it the screen is still blank. Ruled out hardware (tested older version of marlin), but the lcd works fine. I can disable the Reprap discount smart controller, and I don't get any compiling errors, but then I still cant use the lcd obviously.

Here are the errors that I get in the compiler.

ultralcd.cpp: In function 'void lcd_control_temperature_menu()':
ultralcd.cpp:793: error: expected `)' before 'STR_THERMOMETER'
ultralcd.cpp:793: error: expected `)' before 'STR_THERMOMETER'
ultralcd.cpp: In function 'void lcd_control_volumetric_menu()':
ultralcd.cpp:923: error: expected `)' before 'STR_h3'
ultralcd.cpp:923: error: expected `)' before 'STR_h3'

Tried troubleshooting marlin itself, but i'm not a programmer. Let me know if you need more information. I would really like to get my printer working again.

haqnmaq, 2015-03-06:
So I was able to fix it by disabling the #define REPRAP_DISCOUNT_SMART_CONTROLLER For some reason that fixed the lcd and now the lcd is working again!?! I still have that lcd controller, but for some reason commenting out the lcd controller worked! I'm confused...

haqnmaq, 2015-03-06:
Ok, well I spoke too soon. I restarted my 3d printer and the lcd is not working again.

thinkyhead, 2015-03-06:
I'll have a look at this later. Can you post your configuration files someplace, perhaps in a Gist?

thinkyhead, 2015-03-06:
Merged #1571 which _may_ fix this…?

boelle, 2015-03-10:
@tanmanguy and @haqnmaq ... was this fixed by the merge?

tanmanguy, 2015-03-10:
I have no way of testing this. I shorted out my 90usb1248... what ever it is on my printr board using ramps 1.4 now got a new chip tho. haven't got around to soldering on the replacement chip yet

thinkyhead, 2015-03-11:
Should be fixed. There is now an error thrown if users forget to set one of the HITACHI language options.

closed 2015-03-11 by thinkyhead