MarlinFirmware/Marlin #1499: Auto bed leveling lift Z-Probe before retraction

Kiessar, 2015-02-14:
The probe scratches over the print bed when retracting the servo. 

http://youtu.be/kWzHsQ8FqwY

The probe should lift e. G. Z_RAISE_BETWEEN_PROBINGS

I tried several times to fix it myself, but created other issues with my fixes.

thinkyhead, 2015-02-16:
If you include `E` does it work, versus if you don't include `E`? I think I see a potential bug in the code when that is omitted.

Kiessar, 2015-02-16:
What do you mean with including E?

Kiessar, 2015-02-16:
What I have seen so far is, that there is no code to delay the retraction or lift before retraction. 

The issue is only one in homing G28.
In G29 from my opinion you don't need  to retract at all between probing(configurable?  ). But only at the end of it. 
Is there any documentation about all the movement functions? Plan movement, and do blocking movement and so on? Where the real current position is stored? 
I could try to give a hand with small fixes like that. 
Usually I am programming java but did some c and c++. My skills are just to low for bigger changes. And the code looks pretty untidy already I don't want to make it worse.

brainscan, 2015-02-16:
This has all been done here http://3dprintboard.com/showthread.php?3105-Auto_Bed_Leveling-Enhanced-G29-command 
thanks to Roxy. There are instructions on how to implement the retraction for G28 and G29. I've been using her modified version for a while and it's great.

Kiessar, 2015-02-17:
OK got it. For G29 you are right, I will give it a try in the evening. But for homing G28 the problem persists.

brainscan, 2015-02-17:
Did you make the changes for G28? It is listed on the same page but slightly further down. After I changed to this I haven't had those problems anymore.

thinkyhead, 2015-02-19:
Proposing we integrate [these enhanced commands](http://3dprintboard.com/showthread.php?3105-Auto_Bed_Leveling-Enhanced-G29-command).

Kiessar, 2015-02-19:
I icluded the changes in my code.Works pretty well.

brainscan, 2015-02-19:
If you made all of her changes it's useful for manually levelling the bed too so the correction hardly does anything. If you send G29 n5 T V4 it outputs the topology of the bed so you can see where it's high and low. Be careful of capital letters as N and n are treated differently.

CptanPanic, 2015-02-19:
I would vote also to include those enhanced G29 commands from Roxy.

thinkyhead, 2015-03-03:
Check out #1556 for my adaptation of Roxy's enhanced `G29`. I kept the current parameter `P` for number of points instead of using `n`, `U`, and `u` as in Roxy's code. Our `E` parameter means the same thing (keep the probe down). So only the `V` and `T` parameters are totally new. I reduced some redundancy in Roxy's code as well, and applied some cleanup, optimization, and logical fixes.

japhillips87, 2015-03-06:
I tried to use the latest from development yesterday. It seems that either the changes to ABL are not working properly on my Prusa i3, or something is wrong with the hardware. After the ABL routine the print starts. The right side of the bed is level and correct but the left side is too far from the extruder. I'm not sure what is wrong. It didn't start happening until I updated my firmware.

thinkyhead, 2015-03-06:
Hi @japhillips87 – The G29 code is pretty new, and has been receiving some small tweaks today. Do you get any messages on the serial output when you use G29 from Pronterface?

japhillips87, 2015-03-06:
I haven't updated my firmware since sometime last year. The only thing that looked suspect to me was the matrix. It only used whole numbers. That may be intended but I'm not sure.It was something like:
    1.00, 0.00, 0.00
    0.00, 1.00, 0.00
    0.00, 0.00 1.00
The Z height was something like:
front left 4.9mm
front right 5.6mm
back right 5.4mm
back left 4.8mm

thinkyhead, 2015-03-07:
As soon as you can get to it, no rush. Try it with different parameters and see what it does. Use V4 for the most verbose output.

japhillips87, 2015-03-07:
Here is the output:

    SENT: G28
    RECEIVED: ok
    SENT: G29 E V4
    RECEIVED: G29 Enhanced Auto Bed Leveling Code V1.25:
    RECEIVED: Full support at: http://3dprintboard.com/forum.php
    RECEIVED: Bed x: 35.00 y: 20.00 z: 4.96
    RECEIVED: Bed x: 170.00 y: 20.00 z: 5.67
    RECEIVED: Bed x: 35.00 y: 180.00 z: 4.85
    RECEIVED: Bed x: 170.00 y: 180.00 z: 5.14
    RECEIVED: Eqn coefficients: a: 0.00 b: -0.00 d: 4.97
    RECEIVED: Mean of sampled points: 5.154487 
    RECEIVED:  
    RECEIVED: Bed Height Topography: 
    RECEIVED:  --0.01289 --0.30322
    RECEIVED:  +0.51126 --0.19515
    RECEIVED: 
    RECEIVED: planeNormal x: -0.00 y: 0.00 z: 1.00
    RECEIVED:  
    RECEIVED: 
    RECEIVED: Bed Level Correction Matrix:
    RECEIVED: 1.00 0.00 0.00 
    RECEIVED: 0.00 1.00 -0.00 
    RECEIVED: -0.00 0.00 1.00 
    RECEIVED: ok
    RECEIVED: echo:endstops hit:  Z:5.14
    endstops hit:  Z:5.14

thinkyhead, 2015-03-07:
Looks like it's getting numbers correct, if you will forgive the formatting. There've been some slight changes to G29 today, so you can try re-flashing. I will correct the display and also round those numbers so you don't get -0.00.

japhillips87, 2015-03-07:
Started another 6 hour print. I will re-flash in the morning and give an update. This time it almost got it perfectly level. The origin corner had a tad too much distance. I really appreciate all the hard work that has gone into this software. Let me know if there is anything else that you need from me.

thinkyhead, 2015-03-07:
I'm doing some cleanup of the bed leveling code, and more sanity-checking at compile time. I haven't found any obvious bugs, but I cleaned up some things and perhaps the sanity-checking will help.

japhillips87, 2015-03-09:
Just wanted to give an update. I re-flashed with `9dccd3a` and I still have a slight issue. The front left and back right corners are too low. I'm printing for some customers of mine and I'm almost using the entire bed for the prints. Could it be something to do with the points that I use for the grid? Here is the configuration that I use.

    #ifdef AUTO_BED_LEVELING_GRID
  
      // The edges of the rectangle in which to probe
      #define LEFT_PROBE_BED_POSITION 35
      #define RIGHT_PROBE_BED_POSITION 170
      #define FRONT_PROBE_BED_POSITION 20
      #define BACK_PROBE_BED_POSITION 170
  
      // Set the number of grid points per dimension
      // You probably don't need more than 3 (squared=9)
      #define AUTO_BED_LEVELING_GRID_POINTS 2
  
  
    #else  // !AUTO_BED_LEVELING_GRID
  
        // Arbitrary points to probe. A simple cross-product
        // is used to estimate the plane of the bed.
        #define ABL_PROBE_PT_1_X 33  // was 15
        #define ABL_PROBE_PT_1_Y 180
        #define ABL_PROBE_PT_2_X 33  // was 15
        #define ABL_PROBE_PT_2_Y 20
        #define ABL_PROBE_PT_3_X 170
        #define ABL_PROBE_PT_3_Y 20
  
    #endif // AUTO_BED_LEVELING_GRID
  
  
    // Offsets to the probe relative to the extruder tip (Hotend - Probe)
    // X and Y offsets must be integers
    #define X_PROBE_OFFSET_FROM_EXTRUDER 33     // -left  +right
    #define Y_PROBE_OFFSET_FROM_EXTRUDER 7      // -front +behind
    #define Z_PROBE_OFFSET_FROM_EXTRUDER -4.94  // -below (always!)

brainscan, 2015-03-09:
I know it takes slightly longer but I always have 5 probe points when printing over the entire bed. Might be worth trying.

Wurstnase, 2015-03-09:
Usually 3 is a good number. 5 will give not a big advantage for probing (just takes huge time) and 2 are not much enough for a good least squares.

thinkyhead, 2015-03-10:
@japhillips87 More points might give a better result, so check that. And try again with V4 so you get the full output. See if there are any anomalies with 3, 4, and even 5 points.

nophead, 2015-03-10:
If opposite corners are out then either the bed is a twisted plane, or more likely, your Y bars are not parallel. ABL will not correct for that, no matter how many points you use. All it will do is find the best compromise.

brainscan, 2015-03-10:
It's only an extra 10 seconds to probe the extra points and the improvement I had meant it was a no brainer. It was recommended to me by Roxy who made the enhanced code changes and it worked so I use 5 as a minimum now.

Wurstnase, 2015-03-10:
Is it possible that you probe doesn't go high enough after probing? I use an inductive sensor and this takes me ~5s per probe. If the probe don't release the hole process is done in seconds but the result is garbage.

With 3 grd_points you will have 9 points to probe. With 5 seconds each it takes ~45s. With 4 grid_points this takes 80s. With 5 you are at 125s and you should not have that good results as aspected. It's a least squared method. Only if your bed is realy uneven (where you have other issues) this could be a solution. But in most cases this makes no sense. The more accuracy is just theory and will go down in tolerances...

japhillips87, 2015-03-10:
My probe raises 6mm between probing. I'm using a mechanical endstop. @nophead I will make sure that my y bars are parallel. I will also go up to 3 grid_points. I won't be around my printer until tonight. I will try then.

boelle, 2015-03-10:
i see a merge... so are this one not "fixed" ?

japhillips87, 2015-03-11:
I just started 2 more prints. I went to 3 grid_points. On the first print, the front of the bed was too far and the back was too close. It was so far off that the front was not being pushed onto the glass plate at all and the back was pushed in so much that the filament started grinding. I started a second print and it was pretty close, but the front left and back right corners were again off. I'm starting to think that my mechanical endstop is not accurate enough. I use [this](http://www.makerfarm.com/index.php/misc-electronic-parts/mechanical-endstop-switch.html) endstop without the little metal prong. If this one is not accurate enough could someone recommend a better one?

Wurstnase, 2015-03-11:
Do you use "Z_PROBE_REPEATABILITY_TEST"?

japhillips87, 2015-03-12:
I do use it. I haven't run any tests with it though.

japhillips87, 2015-03-12:
I ran some tests. Here is the log from my Z_PROBE_REPEATABILITY_TEST:

    SENT: M48 X100 Y100 V4
    RECEIVED: M48 Z-Probe Repeatability test
    RECEIVED: Positioning probe for the test.
    RECEIVED: 1 of 10   z: 5.083833 mean: 5.083833   sigma: 0.000000
    RECEIVED: 2 of 10   z: 5.071881 mean: 5.077857   sigma: 0.005976
    RECEIVED: 3 of 10   z: 5.063913 mean: 5.073209   sigma: 0.008186
    RECEIVED: 4 of 10   z: 5.069142 mean: 5.072192   sigma: 0.007305
    RECEIVED: 5 of 10   z: 5.076114 mean: 5.072977   sigma: 0.006719
    RECEIVED: 6 of 10   z: 5.089809 mean: 5.075782   sigma: 0.008774
    RECEIVED: 7 of 10   z: 5.084331 mean: 5.077003   sigma: 0.008656
    RECEIVED: 8 of 10   z: 5.080348 mean: 5.077422   sigma: 0.008172
    RECEIVED: 9 of 10   z: 5.072380 mean: 5.076861   sigma: 0.007866
    RECEIVED: 10 of 10   z: 5.079849 mean: 5.077160   sigma: 0.007516
    RECEIVED: Mean: 5.077160
    RECEIVED: Standard Deviation: 0.007516

And here is the log from two back to back ABL sequences:

    SENT: G29 E V4
    RECEIVED: G29 Auto Bed Leveling
    RECEIVED: Bed X: 35.00 Y: 20.00 Z: 5.04
    RECEIVED: Bed X: 102.00 Y: 20.00 Z: 5.22
    RECEIVED: Bed X: 169.00 Y: 20.00 Z: 5.48
    RECEIVED: Bed X: 35.00 Y: 95.00 Z: 4.93
    RECEIVED: Bed X: 102.00 Y: 95.00 Z: 5.06
    RECEIVED: Bed X: 169.00 Y: 95.00 Z: 5.22
    RECEIVED: Bed X: 35.00 Y: 170.00 Z: 4.87
    RECEIVED: Bed X: 102.00 Y: 170.00 Z: 4.89
    RECEIVED: Bed X: 169.00 Y: 170.00 Z: 4.93
    RECEIVED: Eqn coefficients: a: 0.00 b: -0.00 d: 5.09
    RECEIVED: Mean of sampled points: 5.070692
    RECEIVED:  
    RECEIVED: Bed Height Topography: 
    RECEIVED:  -0.13725 -0.17859 -0.20548
    RECEIVED:  +0.14959 -0.00777 -0.14572
    RECEIVED:  +0.40706 +0.14910 -0.03093
    RECEIVED: 
    RECEIVED: planeNormal x: -0.00 y: 0.00 z: 1.00
    RECEIVED:  
    RECEIVED: 
    RECEIVED: Bed Level Correction Matrix:
    RECEIVED: 1.00 0.00 0.00 
    RECEIVED: 0.00 1.00 -0.00 
    RECEIVED: -0.00 0.00 1.00 
    RECEIVED: ok
    RECEIVED: echo:endstops hit:  Z:4.93
    endstops hit:  Z:4.93
    SENT: G29 E V4
    RECEIVED: G29 Auto Bed Leveling
    RECEIVED: Bed X: 35.00 Y: 20.00 Z: 5.03
    RECEIVED: Bed X: 102.00 Y: 20.00 Z: 5.24
    RECEIVED: Bed X: 169.00 Y: 20.00 Z: 5.50
    RECEIVED: Bed X: 35.00 Y: 95.00 Z: 4.91
    RECEIVED: Bed X: 102.00 Y: 95.00 Z: 5.04
    RECEIVED: Bed X: 169.00 Y: 95.00 Z: 5.21
    RECEIVED: Bed X: 35.00 Y: 170.00 Z: 4.85
    RECEIVED: Bed X: 102.00 Y: 170.00 Z: 4.88
    RECEIVED: Bed X: 169.00 Y: 170.00 Z: 4.90
    RECEIVED: Eqn coefficients: a: 0.00 b: -0.00 d: 5.09
    RECEIVED: Mean of sampled points: 5.063360
    RECEIVED:  
    RECEIVED: Bed Height Topography: 
    RECEIVED:  -0.16179 -0.17898 -0.21633
    RECEIVED:  +0.14896 -0.01912 -0.15482
    RECEIVED:  +0.44004 +0.17286 -0.03082
    RECEIVED: 
    RECEIVED: planeNormal x: -0.00 y: 0.00 z: 1.00
    RECEIVED:  
    RECEIVED: 
    RECEIVED: Bed Level Correction Matrix:
    RECEIVED: 1.00 0.00 0.00 
    RECEIVED: 0.00 1.00 -0.00 
    RECEIVED: -0.00 0.00 1.00 
    RECEIVED: ok
    RECEIVED: echo:endstops hit:  Z:4.90
    endstops hit:  Z:4.90

nophead, 2015-03-12:
The probe repeatability test looks pretty good but your two runs differ by 0.033mm in one point, which is not good. One interpretation of that is your bed position is not repeatable, another is your probe is only repeatable when it doesn't move.

The equation coefficients, plane normal and correction matrix all need to print more decimal points as they currently show no rotation at all.

You bed is not a flat plane as it slopes far more down the left side of the matrix than the right. This means it can't be levelled by a simple rotation, hence why BAL does not work for you. Either your bed is warped or your Y bars have different slopes.

japhillips87, 2015-03-12:
I just noticed something that might be responsible for all of my issues. My printer is a Prusa i3. It uses 8mm threaded rods for the z axis. It seems that the right side has some up and down play. I think my M5 nut is stripped. The left side is rock solid. Could this be what is plaguing me?

nophead, 2015-03-12:
Not sure, normally gravity prevents backlash on the Z axis.

japhillips87, 2015-03-13:
I'm just dumbfounded. I don't understand how anything could be wrong. The only thing I have changed about my setup is the firmware. I haven't moved the printer and nothing is loose. I use the same glass plate on my heated bed with the clips in the exact same place. Can a glass plate be warped? I have checked the y bars and they seem fine. I have also been printing daily before and after the upgrade. The version of the firmware that I had before the upgrade was a version from around June of 2014. It was before the migration to this repo. The auto bed leveling, up until the upgrade, worked flawlessly. After the upgrade, it is not working. I made sure that the bed was not moving at all, and held pressure down on the z axis to make sure there was no backlash. Nothing helped. @nophead when you say that the y bars may have different slopes, do you mean that one bar is not parallel to the ground while the other is? The only other thing I can think to do is go back to the super old firmware to see if that fixes it.

Wurstnase, 2015-03-13:
Afaik the only part which change in g29 is the sequence. Before patch it scans like a snake. Now like a typewriter.

Now:
```
>--------
>--------
>--------
```

before:
```
>--------
--------<
>--------
```

nophead, 2015-03-13:
@japhillips87,
   Yes I mean the Y bars are slightly different heights at each end. I think that would be pretty easy to get with an I3.

As the bed moves backwards and forwards it would rotate around the y axis and give the impression the bed was twisted when it was probed. BAL does not correct for a twisted bed, only an inclined flat bed.

It doesn't explain how an older version of the firmware worked though, so it would be interesting to try that again.

Also you could try levelling with a dial gauge and verify the probe readings. Also place a straight edge across the diagonals to see if the bed is flat or twisted. To fix it you would need to place a shim under one of the Y feet so both bars have the same slope. Ideally you want them perfectly orthogonal the X and Z but the I3 doesn't have anything to enforce that.

thinkyhead, 2015-03-13:
Probably worth comparing the old and new leveling code. There are subtle differences that may account for the new results being different from the old.

japhillips87, 2015-03-13:
I have good news for you, but bad news for me. Downgrading the firmware gave me the same results. It looks like the one Y bar is .73mm higher on one side and .12 lower on the other. The other bar is level. I will do my best to either shim it, or use my tools and make it level.

Roxy-3DPrintBoard, 2015-03-13:
@Wurstnase 

#1501 Afaik the only part which change in g29 is the sequence. Before patch it 
#scans like a snake. Now like a typewriter.

Just an FYI to help understand this.   The Zig-Zag of the original 'Accurate Bed Leveling' was slightly faster than the 'Typewriter' behavior.   But the problem is if you are doing a Topographical map it is much easier to pick the points you need out of a linear list.   Especially when you are trying to accommodate the origin being in any corner.    The 'Enhanced G29 Bed Leveling' code at 3D-PrintBoard did allow the Zig-Zag to happen so long as the Topographical map was not going to be computed and displayed.

In the big picture it doesn't matter.   The least squares fit algorithm doesn't care about the order of the points it is working on.   So the real question is:  Is it worth the extra complexity in the code to get that nice Zig-Zag path?

nicolas-rambaud, 2015-03-13:
To my opinion, the simpler the better, and the less cuputation the better.

Roxy-3DPrintBoard, 2015-03-13:
There is something obviously wrong with all these G29 commands that are being posted.   You can't have a non-level bed (as all of the topography maps indicate) and have a bed level correction matrix that is unity.   Something is wrong with the code that is generating these print outs!   The diagonal of the bed level matrix can't be all 1's with this topographical map!!!!

RECEIVED: Bed Height Topography: 
RECEIVED:  -0.16179 -0.17898 -0.21633
RECEIVED:  +0.14896 -0.01912 -0.15482
RECEIVED:  +0.44004 +0.17286 -0.03082
RECEIVED: 
RECEIVED: planeNormal x: -0.00 y: 0.00 z: 1.00
RECEIVED:  
RECEIVED: 
RECEIVED: Bed Level Correction Matrix:
RECEIVED: 1.00 0.00 0.00 
RECEIVED: 0.00 1.00 -0.00 
RECEIVED: -0.00 0.00 1.00

nophead, 2015-03-13:
As I posted earlier it needs more decimal places to be output to see the
effect.

Wurstnase, 2015-03-13:
Just took a look to the level-code today and saw the zig-zag part. But thanks for explanation.

thinkyhead, 2015-03-14:
Zigzag will happen as long as you don't include the `T` flag or set `V` > 2.

So is there still a bug in the code generating the topography report, or has this been patched recently in the PR that added more decimal places to the output?

closed 2015-03-14 by thinkyhead