MarlinFirmware/Marlin #1500: Issues pile'ing up.... whats going on?

boelle, 2015-02-14:
Been more away from the project and i can see there are more issues reported than solved... 

none of the milestone issues have been touched or assigned to anyone... 

So are people just going in circles? or ?

Sorry is sounding harsh but if those milestones issues are not solved i dont see the point in solving any issue at all... thou i'm not a coder i cant see the problem in picking one of them and work on it... 

locked to collabs only as i dont want to start a bigger stream on mails than need be

thinkyhead, 2015-02-15:
Been clearing a lot of issues. The pile would be a lot bigger without effort here. I'll have to look over the milestone issues and see what's holding them up. Perhaps between the two of us we can clear them.

boelle, 2015-02-15:
i cant code.... i just entered the project as an house elf so to speak... 

binned the issues that where over 2-3 years old and keept everything else... then i started to sort them in categories and put the oldest verified faults in mile stones

i know its not nice to complain if you cant contribute.... sure there has been done lot of work but i have only looked at the milestone issues

thinkyhead, 2015-02-15:
I haven't looked too closely at the milestone issues in recent days, but I will review them later today. I seem to recall some of them were non-issues, or otherwise tricky, but I will let you know their current status after I have a look.

thinkyhead, 2015-02-20:
Clearing up open issues, including this one. Thinking it's never a bad time for a "code freeze" and give the milestone issues another kick. There are some good PRs in the queue, so I feel it would be good to make a new tagged version soon, then some that are awaiting the tag can begin get merged as well. Some issues (and Development in-general) are just awaiting testing.

I like the Code Freeze idea (even putting a beta tag on when the freeze starts): No merging any new code while testing, only big-fixes, and when no bugs are evident from the testers group, give it a proper tag and move on to the next round. So many issues have been addressed lately, it's going to be a long CHANGELOG…

We can reconvene on the open milestone issues themselves.

closed 2015-02-20 by thinkyhead