MarlinFirmware/Marlin #1501: Some sort of configuration/upload UI

DanNixon, 2015-02-14:
Inspired by the conversations I have seen on many issues/PRs where discussion has always descended into "make Marlin very easy to use" and my general hatred for the Arduino IDE.

Some kind of UI that will parse the Configuration.h file and generate a nice table of configuration values with proper validation.
Of course this will likely need comments above each value in a certain syntax but that sounds like a reasonable compromise for good usability.

This would also allow you to compile and upload using something like ino which also gets round a lot of the "don't expect users to install Arduino libraries" issues as it has a more "conventional" library include mechanism.

This is something I would be happy to have a look at doing but can't commit a huge amount of time to it, if I do it then it will likely end up being written in Python and Qt.

thinkyhead, 2015-02-16:
Hi @DanNixon. Have you taken a look at my starting point? It's a configurator that does a lot of the parsing work you describe, and I have started adding extra comments to the Configuration files to fill in where the parser can't automatically determine things. I've been implementing it in pure Javascript, and you can see the current code at #1467. It reads the configs directly from GitHub so it can obtain updated or special versions, and is configurable itself, so it can be made to work with other GitHub repos.

If you decide to go the Qt+Python route so that we could have a desktop app that would also build+install, I have no doubt that would be a welcome to the Marlin project. The thing I'm building is testing at the limits of what can be made in-browser, and to convert settings into other forms - JSON, XML, etc. - and see how they shake out.

ntoff, 2015-02-16:
I would hate to have a bunch of messy config files with extra text in them just to make them parse-able. Wouldn't a better way be to have the user simply configure it how they like using the configurator and then save their settings which would store them in an ini file. Then, no matter what the configuration.h file ends up looking like in the future with everyone working on it, the configurator will still always just read the users .ini settings file and save a fresh, clean configuration.h with those settings loaded from the .ini.

Yes it means users won't be able to load their current configuration.h file but if you're a new user you won't have anything to load anyway.

DanNixon, 2015-02-16:
@thinkyhead I saw your work just after submitting this issue, I like what you have done but I think the ability to switch to something other than the Arduino IDE for compilation and uploading make a local solution more appealing (for me at least).

@ntoff The default Configuration.h is messy enough as it is, replacing all the current comments with something more machine readable wouldn't make it worse (if anything it would be an improvement as the format will be consistent).

galexander1, 2015-02-16:
@ntoff I disagree.  Marking up Configuration.h is desirable because that
is where developers add new features.  Also, adding an extra step between
config.ini and Configuration.h seems senseless to me.

DanNixon, 2015-02-16:
After thought what I propose is the following:

Each config option will have a Doxygen style comment block that will have the standard name and description fields as well as one of several validation fields that will be used by the UI to validate against:
- an integer or double range (```/validate-range 0-255``` or ```/validate-range 0.0-999.9```)
- a boolean value (```/validate-bool```) (which just uncomments the option is set to false in the UI
- an enum where there is a filed for each value (```/validate-enum ATX=1```)
- parameters without validation fields will not be editable from the UI

This is at least for the configuration part, I am open to comments/changes on this but am pretty much going to crack on with my ideas for the compilation and upload UI.

thinkyhead, 2015-02-19:
@DanNixon My point in building my configurator in Javascript was to make it able to run as-is without any other software apart from a modern browser, and to make sure no compilation is required to use the tool. You can open `index.html` in `file://` mode and it will run. It avoids browser security restrictions by using the Github API or reading from a remote folder (with appropriate allow headers added).

I personally would like a desktop-based configurator written in such a manner that it can be deployed on OS X, Windows, Linux, etc. would be nice also, but I think you should first build it as a standalone project, get it refined and working, etc.

If you examine `configurator.js` you may notice that most of the Configuration settings can be sussed-out just by looking at the default values in the file. I'm trying to annotate the configuration files _only where they are ambiguous_. My next step is to have it examine all the `#ifdef` blocks so that it can grey-out options that depend on other options.

I have floated the idea (and this is already implemented in some places) of sanity-checking input values using compiler directives, either within the Configuration files, or in a separate ConfigurationRules.h file. Once the legal values have been expressed in this manner, we can easily parse and use them to constrain the values in the configurator. We can even pull out the #error or #warning text and display it as the error message in the form, similar to the manner that I extracted the comments for the tooltips.

For now I am taking the most conservative approach, so as to require as little tweaking of the Configuration files as possible, and so that they don't in fact require a Doxygen-style header blocks on every setting, because this balloons the file with much that can be derived otherwise. That is not to say that I don't favor changing all the heading comments in these files to Doxygen style! That would likely be much easier to parse. However, I think it might be overkill given how much can be derived without them.

thinkyhead, 2015-02-19:
P.S. Ah yes, I almost forgot. Some of the other things I wanted to do in the browser, connected to this configurator...

- Graphically display the printer geometry. See: http://www.thinkyhead.com/_delta/
- Make the most important settings more up-front
- Get the listing from `example_configurations` and provide a popup list
- `Configuration_delta.h` and `Configuration_scara.h` files?

That last one. Currently for a SCARA or Delta you grab one of the `example_configurations` files, so these settings aren't in the default `Configuration.h` file at all. They will need to be, if the configurator will be using `Marlin/Configuration.h` as the base template.

DanNixon, 2015-02-23:
I did have the recommendation to just use a derivative of kconfig for this which would have handled all the original features I mentioned beautifully, but I see feature creep has started to set in.

I'm going to go ahead and do my work n the build process and see where I get with the configuration stuff, most of your additional features would be fairly simple to add to a Qt application also (I'm sorry to say but I'm pretty much against the web application deployment, mainly just because I want to add the build process into to this as well).

But the idea of sanity checking with the pre processor is definitely worth doing, even without the configuration utility.

thinkyhead, 2015-02-24:
Feel free to cannibalize the regexes and any other schemes I've employed in the javascript version. It's essentially an experimental development platform. Recently @josefprusa has also expressed some interest in adding a configurator to his website. Perhaps through some collaboration we can avoid some duplications of effort, or at least get on the same page about how best to annotate the existing Configurations, and/or to look at what a next-generation Marlin configuration scheme might look like.

boelle, 2015-03-10:
Q... can the configurator be placed on git and opened from here? or is the right way a web space with a domain name etc etc ?

thinkyhead, 2015-03-12:
@boelle Yes it can! https://rawgit.com/thinkyhead/Marlin/marlin_configurator/Marlin/configurator/index.html

boelle, 2015-03-12:
ok, just ordered a programmer so i can burn a bootloader on my tvrrug
electronics. any chance you know how to "upgrade" the hardware files i
placed on git a while ago so they work with ide 1.6 which is now the
latest. the files should be in a folder called OMC

on work now so very slow in replying

thinkyhead, 2015-03-13:
I'll look at that. I've has some struggles trying to update old hardware files in the past, might be worth another sleepless night.

boelle, 2015-03-13:
when i get my programmer and it works i can have that OMC hardware running same day.. 

also the thought behind upgrading everything to IDE 1.6 is that i assume it will be more easy in the future to upgrade the files if its needed

I have also made my own version of Ramps..  pin connection wise its the same but i have the single board up in 5 boards so i can change boards faster if a single thing blows... plus it made them suit the Mendel 90 better.. that one is a thing after summer as i want a few features added...

boelle, 2015-05-31:
will close this one... we do have a config ui on the marlinfirmware.org site,,,, we just need to add a link to it. @thinkyhead what was the link for the interface you made?

closed 2015-05-31 by boelle

thinkyhead, 2015-06-07:
@boelle Here it is! http://marlinfirmware.org/configurator/

boelle, 2015-06-07:
yep.... that one is under install on the wiki :-)