MarlinFirmware/Marlin #1502: Z_SAFE_HOMING overwrites X/Y min with center of bed

doomstrike, 2015-02-15:
In order to use z safe homing the X and Y axis must be homed. When I home X and Y it gets set to 0,0 (corner of my bed). Then when I home the Z axis the bed moves to the center of my print bed (90,90) and does it Z probe. Cool that works as advertised.

However, the problem is that the Z home has a defect in that it overwrites the X/Y home position to be the Z_SAFE_HOMING_[XY]_POINT be 0,0 again when it's at the center of the bed; so instead of the X/Y home begin at the corner of my bed with a correct value of 0,0, instead the X/Y home is at the center of my bed and it incorrectly sets the value to 0,0 again. This is resulting in run away X, Y, and Z movements.

Correct X/Y at 0,0 due to manual X and Y home.

X marks extruder position bottom left corner starting at 0,0
 -- | -- | --  | -- | --
 -- | -- | --  | -- | --
 -- | -- | --  | -- | --
 -- | -- | --  | -- | --
 X  | -- | --  | -- | --
 
Incorrect X/Y (0,0) due to z_safe_homing:
 -- | -- | --  | -- | --
 -- | -- | --  | -- | --
 -- | -- | X  | -- | --
 -- | -- | --  | -- | --
 -- | -- | --  | -- | --
  

The code I'm referencing:
```C
#define Z_SAFE_HOMING   // This feature is meant to avoid Z homing with probe outside the bed area.
                          // When defined, it will:
                          // - Allow Z homing only after X and Y homing AND stepper drivers still enabled
                          // - If stepper drivers timeout, it will need X and Y homing again before Z homing
                          // - Position the probe in a defined XY point before Z Homing when homing all axis (G28)
                          // - Block Z homing only when the probe is outside bed area.

  #ifdef Z_SAFE_HOMING

    #define Z_SAFE_HOMING_X_POINING_X_POINT (X_MAX_LENGTH/2)    // X point for Z homing when homing all axis (G28)
    #define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/2)    // Y point for Z homing when homing all axis (G28)

  #endif
``

brainscan, 2015-02-15:
So if you run G29 does it try and start probing the middle of the bed first? I use the safe z homing but have never had this problem.

doomstrike, 2015-02-15:
I use the grid probing system and its setup to be:
```C
    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 0
    #define RIGHT_PROBE_BED_POSITION 160
    #define BACK_PROBE_BED_POSITION 200
    #define FRONT_PROBE_BED_POSITION 50
```
Let me try again to make my steps more clear.

1) Enter G28
2) X, Y, and Z coordinates get assigned a value of 0. X0 Y0 Z0 at this point.
2) X homes to minimal X-end stop. 
3) Y homes to minimal Y-end stop.
4) X and Y move at the same time to position the Z-endstop probe to the center of the bed.
5) Z axis moves downwards until Z-endstop switch hits bed.

Now the problem is with step 4. Even though the X and Y moved the X,Y coordinates were not updated. In repetier-host it gives real-time X,Y,Z corrdinates. After step 4 the X,Y,Z coordinates are still at 0,0,0 even though the axis moves to the center of the bed plus the offset values for hotend - z probe.

At this point if I were to try and move the X or Y in a negative distance the software prevents it because it sees the axis at 0,0. Worse is if I give it a positive distance it will overrun its axis depending on the value I give it.

Okay so it may not be a bug with Merlin but rather with Repetier host. When I tried it with Pronter-face i observed different results.

vandarin, 2015-02-15:
Yes, this is because Repetier Host doesn't know where the extruder is. Add M114 after G29 to pull the coordinates back to RH.

doomstrike, 2015-02-15:
>Yes, this is because Repetier Host doesn't know where the extruder is. Add M114 after G29 to pull the coordinates back to RH.

Nice this did the trick. Although its after G28 and not G29 :)

vandarin, 2015-02-15:
I have it after G29 so that it pulls the new Z value back as well.

doomstrike, 2015-02-15:
Should I close or leave this ticket open? Seems like it would be a good idea to bullet proof this behavior in the firmware.

doomstrike, 2015-02-15:
>I have it after G29 so that it pulls the new Z value back as well.
Than it would need to be after both G28 and G29. If i only do it after G29 I have a runaway problem.

alexborro, 2015-02-15:
You can close it.  Actually you should open an issue for Repetier Host (and
wait forever).

Cheers.

Alex.

closed 2015-02-15 by doomstrike

doomstrike, 2015-02-15:
Ticket opened: http://forum.repetier.com/discussion/463/g28-and-g29-commands-need-to-issue-m114

Now to look into me using another program. Repatier can't home, Pronter runs out of memory for my 14-18 hour prints :(

alexborro, 2015-02-15:
Forget about hosts.. Get yourself a SD card reader with LCD. It will cost
you $20 to $40 bucks and it will be worth every penny.. No more hosts, no
more PCs, no more USB disconnections and so on.

Cheers.

Alex.

thinkyhead, 2015-02-16:
@alexborro I concur, SD printing is more teh happy.