MarlinFirmware/Marlin #1507: Z Raise after probing addition

doomstrike, 2015-02-16:
Current there are these these 3 options for probing:

```C

  #define Z_RAISE_BEFORE_HOMING 6       // (in mm) Raise Z before homing (G28) for Probe Clearance.
                                        // Be sure you have this distance over your Z_MAX_POS in case
  #define Z_RAISE_BEFORE_PROBING 10    //How much the extruder will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 3  //How much the extruder will be raised when traveling from between next probing points
```

There needs to be an option to raise Z after probing is done. I have the problem of my extruder hitting my bed clamps after probing is done when the print starts. By design my extruder leaves the bed area so that the Z-probe switch can contact the edges of the bed. Problem is the print starts at the height of the last contact causing my extruder to hit my clamps on the way to the center of the bed.

thinkyhead, 2015-02-16:
Yes, that is annoying. We'll look at that!

galexander1, 2015-02-16:
This probably is something Marlin should address, but in the mean time
you can get around it by adding something like G0 Z20 to your G-code
prologue after G29.

Kiessar, 2015-02-16:
Have the same problem. My extruder crashed into the clamps too. +1
An option center after probing or moving to home position would be helpful.

thinkyhead, 2015-03-21:
The feature has been added, but it seems to be misbehaving, retracting the servo immediately without waiting for the Z move to finish. We're still looking into it.

maverikou, 2015-03-21:
@SkyFlyer2 please test https://github.com/maverikou/Marlin/commit/a08e8f03a925b548491b48bfa4d18a6961838c67

I don't have access to a printer right now.

SkyFlyer2, 2015-03-21:
Ok,
I can check this only in monday, because printer on my work.

avluis, 2015-03-22:
I still need to learn how to cherry pick a patch from a different repo so I merged it all manually.
What can I say - it works as expected now.
G28 & G29 E raise the Z axis before retracting probe.

maverikou, 2015-03-22:
Cool. I'll clean it up and PR.

maverikou, 2015-03-22:
PR #1665 needs testing.

SkyFlyer2, 2015-03-22:
It seem to be working. I'll check  tomorrow.

Why change the order check points? This is incorrect settings or the code was changed?

Too much movement will not add exactly. Check, please.

Also, try to start printing to ensure that printer is not start print in air.

SkyFlyer2, 2015-03-23:
I checked the changes.

The probe lift now works, as a whole is normal.

There are new problems:

1) Why order measurement points changed? Now the probe moves diagonally when measuring between points.

2) The LED on the heated bed is blinks now, as a result of an overheat the control mosfet on the board Ramps 1.4.

What was changed?

In previous versions of the firmware did not have such problems.

thinkyhead, 2015-03-23:
@SkyFlyer2 The short answer is, a lot has changed in a short time, so the current Development branch HEAD has some extant bugs. We haven't tagged any releases lately, but if we can determine the most recent point before these bugs began to appear, we could tag that date's last commit as an Alpha or beta version, and encourage people to use that while we solve these issues.

avluis, 2015-03-23:
@SkyFlyer2
The order of measured points was in before @maverikou introduced his PR (https://github.com/MarlinFirmware/Marlin/issues/16660) so that would be unrelated.
I manually applied his patch as well so I may not have the latest Marlin just yet, but my pull was from March 21st towards the end of the day, so that may be a way to track down what has changed for the worse  (https://github.com/avluis/Marlin/commit/ab1914)

SkyFlyer2, 2015-03-24:
@thinkyhead

a lot has changed, a lot of bugs added.

When bed autoleveling is disabled, the servo makes retract before lift the probe.

For now, is not possible using this version Marlin.

thinkyhead, 2015-03-24:
@SkyFlyer2 Thank you for being part of the team testing the latest cutting edge code. It's good to have as much feedback as possible while trying to get these features worked out. With luck and your continued help, we should have them worked out pretty quickly.

avluis, 2015-03-24:
@thinkyhead @SkyFlyer2 :+1:

SkyFlyer2, 2015-03-24:
@avluis

Can you check that Marlin start printing in air after bed autoleveling?

avluis, 2015-03-24:
@SkyFlyer2 Starting a print right now, will report back.

SkyFlyer2, 2015-03-24:
Hmm, strange. The previous Marlin with autolevel prints on test bed, with the same settings on last Marlin it start print in air.

Z-offset=0
First layer=0.18
Other layers=0.24
nozzle=0.4

How I can attach the file here?

Here is slicer startup G-code:

-----
G28 ; home all axes
M106
G29 e P3
G1 Z5 F5000 ; lift nozzle
------

avluis, 2015-03-24:
@SkyFlyer2 
I like to have a z offset in slic3r so this may not work for you but this is what I use (I have a dual extruder setup) - z-offset: -1.52

G-Code:
M80 ; turn on printer
M117 Homing printer
G28 ; home
G90 ; use absolute coordinates
G92 E0 ; Reset extruder distance
M117 Preheating bed
M140 S[first_layer_bed_temperature] ; Set bed temp, no wait.
M117 Leveling bed...
G29 E ; level the bed
M190 S[first_layer_bed_temperature] ; Set bed temp, wait.
G92 E0 ; reset extruder distance
M117 Now printing...

SkyFlyer2, 2015-03-25:
@avluis

I do not see anything special in your code. But, what is Z-offset? If using autoleveling procedure, usually don't require any additional offset. Only offset from z-probe to nozzle. In my case is -16.31.

My "configuration.h" contains the same settings for both version of Marlin.

avluis, 2015-03-25:
I usually swap my nozzles and/or extruder so my Z-offset is usually set in slic3r and not on the firmware.

SkyFlyer2, 2015-03-25:
Any ideas, what to check?

When I manually add the lifting up code (do_blocking_move...) in the previous version Marlin (without lifting up probe before retraction), result was the same - printing start in the air.

thinkyhead, 2015-03-25:
@SkyFlyer2 If printing is starting in the air then it sounds like the Z coordinate is screwed up after the leveling procedure. But, @Sniffle may have figured it out – at least in his case it was related to configuration. https://github.com/MarlinFirmware/Marlin/issues/1682#issuecomment-85595876

Sniffle, 2015-03-25:
I ended up just using M851 to make my Z offset correct again.  Its not a proper fix but its a work around... For me it was ~2.3mm more negative to make it printable and it has remained at that value for 2 prints.  My original offset was 4.3

thinkyhead, 2015-03-26:
#1700 should take care of most issues, and I will merge that shortly. I still have one question concerning whether the engage/retract functions called for `Z_PROBE_ALLEN_KEY` should also be called for SERVO.

    #ifdef Z_PROBE_SLED
      dock_sled(false); // engage (un-dock) the probe
    #elif defined(Z_PROBE_ALLEN_KEY) //|| defined(SERVO_LEVELING)
      engage_z_probe();
    #endif

Should that `defined(SERVO_LEVELING)` be included? I'm hoping someone can test and answer.

avluis, 2015-03-26:
If my Z probe was allen key based I would be able to test it but I'm afraid I do not have that configuration.

thinkyhead, 2015-03-26:
@avluis Well, actually the change I'm proposing would only affect non-allen-key-based probes. If your probe is working with all the possible probe methods, then this code probably doesn't need to be changed. I don't have a probe at all, or I would test it myself.

maverikou, 2015-03-31:
@thinkyhead I don't think you need to engage/retract servo based probes in the allen key code path as servo based probes are triggered directly in the probing routine.

thinkyhead, 2015-03-31:
@maverikou How is the allen key engaged/disengaged – by a special movement or a switch?

thinkyhead, 2015-03-31:
#1755 should now have z raise working before, between, and after probing. Please test and see where it may need adjustment. Also raise before homing was fixed earlier tonight.

scotty1024, 2015-03-31:
@thinkyhead the Allen key is spring loaded with the Allen key L side up resting on a ramp mechanism. The L is moved against a belt which rotates the Allen key and the spring pushes it down the ramp deploying the Allen key as the probe with the L pushing on the micro switch. To stow the probe there is a screw mounted in an extrusion slot at a specific location. The bottom of the Allen key is pressed against the screw forcing it back up to the top of the ramp which causes the Allen key to rotate back on to the rest at the top of the ramp. Deployment energy is stored in the spring during this process.

SkyFlyer2, 2015-03-31:
Trying to test fixes.
After command G29 E V4 P3 it retract servo between probings. Is not correct.

If I after the command G28 slightly lowered down the extruder, this movement is not taken into account, and after the command G29 E B4 P3 Marlin tries to break the servo and a sample of the table.
It is normal?

thinkyhead, 2015-04-01:
@SkyFlyer2 After taking a poll, it was decided to reverse the meaning of  `E` in `G29` – for consistency with `M48` and to make leaving the probe down the default behavior. So if you want the probe to remain lowered, leave out `E`.

I'm not sure how well it's supported to move the extruder between `G28` and `G29`.

I'm still trying to solidify all the code that raises the Z axis, engages and disengages the probe, etc., and trying to make sure that at the end of the process, once there is a bed leveling plane, the firmware can derive the (corrected) position of the XYZ axes and continue on, having the current position and performing automatic compensation for all movement. So, for example, on a very tilted bed, moving Z alone will cause X and/or Y to also move to compensate.

In the probing routine I hope to get it solidly working like so…

- Use `G28` to establish the current XYZ position.
- At the end of `G28`, if the probe is engaged, lift Z according to settings, then retract the probe.
- Then, `G29` starts a bed probe…
- Before starting `G29`, if there is a servo probe, make sure Z is high enough before engaging.
- At each probe point, the probe is lowered, lifted, then lowered slowly again, recording a point.
- After probing, the Z axis is lifted based on the lift-between setting.
- If `E` is included, then the probe is also retracted.
- The carriage moves to the next point.
- If `E` was included, the probe is re-deployed. (Z position assumed already high enough.)
- And so on till all probe points are done.
- If `E` was not included, disengage the probe at the end.

SkyFlyer2, 2015-04-01:
@thinkyhead

please pay attention that the movement along the Z axis is not taken into account after executing the command G28. Before, in early commits, any such movement was taken into account

I can not imagine why movement along the Z axis cause to move the axis X and Y.
Z-axis is vertical.

Unfortunately, auto-leveling bed at the moment so inaccurate that I refused it - after manual alignment, the head is moving almost parallel to the bed. If you make auto-leveling procedure, the head in the end will not move parallel to the surface of the bed.

thinkyhead, 2015-04-01:
#1764 is the latest update to bed leveling code. Keep those helpful diagnostics coming!

@SkyFlyer2 The planner itself does the bed leveling compensation, so moving Z alone in the "abstract space" will in fact cause X or Y to change, but this would only noticeable with a very askew bed, tilted a full degree or more.

SkyFlyer2, 2015-04-01:
@thinkyhead

Trying to understand)
Let make the bed at a slight angle of about 5 degrees. (I did this slope for the test - marlin can successfully print with auto-leveling). Next, perform the G29 command. Further, if I want to raise or lower a bit the Z axis, it will move the bed along the axes along the X/Y ? Why?
Can't understand why you need such a move. This never happened before.

nophead, 2015-04-01:
The model is tilted to match the slope of the bed, so any move along a
single axis of the model will, in general, move all three machine axes.

For an explanation of how it is supposed to work and the correct maths, see
http://hydraraptor.blogspot.co.uk/2011/04/auto-bed-leveling.html. I don't
think any version of Marlin has got this right yet. The maths has always
had bugs.

thinkyhead, 2015-04-01:
@boelle I think the "On Hold During Bugfix" label should be removed, because this feature already exists, but should be part of the current bug-fix round.

boelle, 2015-04-02:
but if its there should it not be marked as a bug then?

thinkyhead, 2015-04-02:
@boelle It was added some time after this issue was posted. This was originally a feature request, so the label has stuck, but now it's turned into one of many threads on auto bed leveling.

thinkyhead, 2015-05-09:
Maybe it makes sense to rename `Z_RAISE_AFTER_PROBING` to `Z_RAISE_FOR_STOWING` for the more general use of making clearance under the Z probe so the servo can move it without hitting the bed. Then `Z_RAISE_FOR_STOWING` would apply also in-between probes when the servo is being stowed and re-deployed. Once the firmware knows how much clearance is needed for the Z probe to home, probe, deploy, and stow, it can choose which clearance value to use in each case.

boelle, 2015-06-29:
so this should be turned in to a bug if i understand correct?

Sniffle, 2015-06-29:
Ill verify this, this week and let you know.

closed 2015-11-06 by AnHardt