MarlinFirmware/Marlin #1512: Ultra LCD compilation errors

MathersMax, 2015-02-18:
When uncommenting #define ULTRA_LCD, i get the following :

ultralcd.cpp: In function ‘void lcd_goto_menu(void (*)(), uint32_t, bool)’:
ultralcd.cpp:195: error: ‘encoderPosition’ was not declared in this scope
ultralcd.cpp:196: error: ‘lcd_quick_feedback’ was not declared in this scope

Using arduino 1.0.6 and ubuntu.
Nothing on the internet
Any idea?

Regards

Maxime from France

thinkyhead, 2015-02-19:
Unless I am mistaken, that's not enough. You have to specify the model that you are using, below ULTRA_LCD. The `Configuration.h` file comments might not be clear on this point.

Nprod, 2015-02-19:
Yes, #ULTRA_LCD is not enough, you also need to define #ULTIPANEL and #NEWPANEL and set the height and width. Quickest way would be to just uncomment #REPRAP_DISCOUNT_SMART_CONTROLLER because it enables everything you need for a 20x4 display.

MathersMax, 2015-02-19:
Hi ! 
Thanks for the quick answer!
When uncommenting #REPRAP_DISCOUNT_SMART_CONTROLLER I got this:

ultralcd.cpp:73: error: ‘void lcd_quick_feedback()’ used but never defined

Same when manualy uncomment #ULTIPANEL and #NEWPANEL ...

I'm looking for a solution by myself but maybe you already know this problem?

MathersMax, 2015-02-19:
I found the solution : 

(Redownload merlin an try again)

I think arduino v 1.0.6 change .ino to .pde and this might had cause the error.

Issue closed. Thanks

closed 2015-02-19 by MathersMax