MarlinFirmware/Marlin #1523: Compilation error when EXTRUDERS 2

clefranc, 2015-02-21:
Just downloaded the latest development version, default configurations:

EXTRUDERS 1 & PIDTEMP = compilation OK
EXTRUDERS 2 & PIDTEMP = compilation error
EXTRUDERS 2 & no PIDTEMP = compilation OK

Error detail:

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:721:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:721:1: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 141 [ ivtmp.170 ])) [4 MEM[base: _101, offset: 0B]+0 S4 A8])) temperature.cpp:505 99 {*movsf}
     (expr_list:REG_INC (reg:HI 141 [ ivtmp.170 ])
        (nil)))
temperature.cpp:721: confused by earlier errors, bailing out
Error compiling.

thinkyhead, 2015-02-22:
@clefranc Just fixed some typos in temperature.cpp that might have been causing that. Try the very latest Development head.

clefranc, 2015-02-22:
@thinkyhead Nope

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:719:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:719:1: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 147 [ ivtmp.209 ])) [4 MEM[base: _113, offset: 0B]+0 S4 A8])) temperature.cpp:503 99 {*movsf}
     (expr_list:REG_INC (reg:HI 147 [ ivtmp.209 ])
        (nil)))
temperature.cpp:719: confused by earlier errors, bailing out
Error compiling.

thinkyhead, 2015-02-22:
Hmm, any other settings involved? If I do `PIDTEMP` and `EXTRUDERS 2` alone I don't get the error. I'll poke around...

clefranc, 2015-02-22:
Downloaded the whole firmware, just change EXTRUDERS from 1 to 2, get compile error. Removing PIDTEMP gives no error.

Arduino 1.5.8 / Windows 7 x64

thinkyhead, 2015-02-22:
Internet forums suggest there may be funky characters messing with the compiler. Try replacing the end of manage_heater with freshly-pasted code and see if it fixes it.

    //code for controlling the extruder rate based on the width sensor
    #ifdef FILAMENT_SENSOR
    if (filament_sensor) {
      meas_shift_index=delay_index1-meas_delay_cm;
      if (meas_shift_index < 0) meas_shift_index += MAX_MEASUREMENT_DELAY + 1;  // loop around buffer if needed
      // get the delayed info and add 100 to reconstitute to a percent of
      // the nominal filament diameter then square it to get an area
      meas_shift_index = constrain(meas_shift_index, 0, MAX_MEASUREMENT_DELAY);
      volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM] = pow((measurement_delay[meas_shift_index] + 100.0) / 100.0, 2);
      if (volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM] < 0.01)
        volumetric_multiplier[FILAMENT_SENSOR_EXTRUDER_NUM] = 0.01;
    }
    #endif

clefranc, 2015-02-22:
Tried the code above: error.
Copied entire .cpp from github: error.

Will downgrade the IDE now...

clefranc, 2015-02-22:
Arduino 1.6.0, still compile error!

thinkyhead, 2015-02-22:
Hmm, what if you do some tweaking to the text, such as de-tabbing, and changing the line-endings to Windows, then to Unix, to get them all consistent... Does that make any difference? I suppose there could be something like a missing close-parenthesis, but it's just such a weird error...

clefranc, 2015-02-22:
In temperature.cpp, there is no erro if I comment line 530 & 533.

thinkyhead, 2015-02-22:
How about [this one](https://github.com/thinkyhead/Marlin/blob/cleanup_temperature/Marlin/temperature.cpp)?

clefranc, 2015-02-22:
Tried your branch on my computer and another computer: same error
Tried the official development branch on my computer and another computer: same error

Just saw the new issue https://github.com/MarlinFirmware/Marlin/issues/1529

I uncomment the Thermal Runaway Protection, compilation error gone...

Arandir, 2015-02-24:
Arduino 1.6.0, 1 extruder: no error
Arduino 1.6.0, >1 extruder: error
Arduino 1.0.6, 1 extruder: no error
Arduino 1.0.6, >1 extruder: no error

Possibly problem is in the compiler Arduino

thinkyhead, 2015-02-24:
@Arandir That is my conclusion, as well. It's too bad that 1.6.0 is displaying this behavior. Maybe if we share this with @Arduino they can find and fix the compiler issue, but I am sure it's a tricky bug to find. In the meantime, try rearranging the code, running filters on it to remove whitespace, extra line-endings, and invisible characters.... Anything to coerce the compiler into choking differently or (ideally) not choking.

nophead, 2015-02-24:
Changing whitespace, line endings, etc will have no effect. This is back end compiler bug where it is trying to fit instructions and registers to the semantic representation of the code, long after lexical and syntax analysis. The only thing that will work around it is code changes that mean it needs less registers.

emartinez167, 2015-02-24:
I have been doing some searches and this bug is not exclusive to Arduino,
but rather to the toolset behind it. Try googling it!

thinkyhead, 2015-02-24:
@nophead Sometimes these errors really are spurious, but I am certainly willing to look closely at the scope where this error is arising and count registers. I'm dubious, because I actually do not get these errors. Moreover I have yet to find a reference via Google where this _wasn't_ spurious, but caused by too many registers. Also, I can't see why the implicated code region _would_ require more registers than other, more complex code regions.

@emartinez167 I thought, as I was writing, that this is really a gcc problem, and really the Arduino geeks probably don't want to go there!

nophead, 2015-02-25:
@thinkyhead,
  Yes they are always spurious because they are a compiler bug in the back end that is specific to the AVR. What I mean is it is at the last part of the compiler where it is fitting instructions and allocating registers. Changes to white space, etc, won't affect it because the representation the compiler is working in on at that stage is register transfer logic. White space doesn't even get the the AST.

See http://en.wikibooks.org/wiki/GNU_C_Compiler_Internals/GNU_C_Compiler_Architecture_3_4#Main_Representations

Things that will affect it is the complexity of the code after it is has been converted to RTL and optimised and the number of pointers it needs to keep live at the same time.

emartinez167, 2015-02-25:
Are there any compiler flags we can set to change the optimization, or is
that going to break even more stuff?

thinkyhead, 2015-02-26:
@nophead @emartinez167 An interesting problem to solve, anyway… and weird that I don't get the compile bug with 1.0.5 or 1.5.2. (About to try 1.6.0.)

Well, I will mess with it now, as I'm re-flashing my RAMPS, and see what happens…

nophead, 2015-02-26:
Are you using the same version of gcc as the OP? A big problem is when you run Arduino on Windows you use the version bundled with it but on Linux you use the version of AVR installed system wide.

thinkyhead, 2015-02-27:
@nophead I'm using OS X and on this platform every version of Arduino comes with its own build of gcc (and other tools) inside the application bundle. I presume that Arduino is using the gcc that it comes bundled with.

nophead, 2015-02-27:
Well as this is a compiler bug and you don't get it with the same source
code the most likely explanation is you are using a different version of
the compiler or different command line flags.

thinkyhead, 2015-02-27:
@nophead I have confirmed that Arduino is building using the toolchain in its bundle, as it should. I won't paste the build log here, but in verbose mode it's quite clear.

Using the settings `EXTRUDERS == 2 && PIDTEMP` I get no error when building with Arduino 1.5.2, Arduino 1.0.5, or UECIDE 0.8.7z36, but I do get an error consistently with Arduino 1.6.0.

The only trick I have found to get rid of the error is to comment out lines 529-535 in temperature.cpp. Make of that what you will! I haven't found any way to tweak those lines (so far) that eliminates the error. Perhaps someone else can mess with them and see if anything makes a difference.

chron0, 2015-03-02:
Had the same issue and "worked" on temperature.cpp with dos2unix and tabs to spaces -> spaces to tabs. Afterwards it compiled, can you check if that solves your issues too? https://github.com/apollo-ng/Marlin/commit/fb75a9272da33ea0a9f2793a1fc89b69428d103c

Dimonira, 2015-03-08:
I did so. File temperature.cpp.
1. Uncomment this lines:
// static float pid_input[EXTRUDERS];
// static float pid_output[EXTRUDERS];
and edit their (delete [EXTRUDERS]):
static float pid_input;
static float pid_output;
2. Comment out or delete line(s) in manage_heater() function:
float pid_input, pid_output;
That's all.

chron0, 2015-03-08:
It's already been fixed by @thinkyhead in a heavy rework of temperature.cpp which is already merged and tested in pr-felix, that's why it was reverted :)

thinkyhead, 2015-03-09:
@Dimonira Are you getting the same compiler issue with the latest Development code? Or, are you getting a different error?

Dimonira, 2015-03-09:
@thinkyhead 
Yes, with the latest Development code and EXTRUDERS=2 (nothing else is changed) and Arduino: 1.6.0 I get this same error:

Arduino: 1.6.0 (Windows 8), Плата"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:726:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:726:1: error: this is the insn:
(insn 103 102 105 4 (set (reg/v:SF 105 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 140 [ ivtmp.169 ])) [4 MEM[base: _98, offset: 0B]+0 S4 A8])) temperature.cpp:543 99 {*movsf}
     (expr_list:REG_INC (reg:HI 140 [ ivtmp.169 ])
        (nil)))
temperature.cpp:726: confused by earlier errors, bailing out

It is noteworthy that in the latest version (download right now), my version of the "patch" does not help.

Dimonira, 2015-03-09:
After these changes Arduino 1.6.0 was compiled without error. I moved some of the code from the for-loop in the separate function get_pid_output(int e) (variable pid_input replaced its value current_temperature[e]):

float get_pid_output(int e)
{
  float pid_output;
  #ifdef PIDTEMP
    #ifndef PID_OPENLOOP
      pid_error[e] = target_temperature[e] - current_temperature[e];
      if (pid_error[e] > PID_FUNCTIONAL_RANGE) {
        pid_output = BANG_MAX;
        pid_reset[e] = true;
      }
      else if (pid_error[e] < -PID_FUNCTIONAL_RANGE || target_temperature[e] == 0) {
        pid_output = 0;
        pid_reset[e] = true;
      }
      else {
        if (pid_reset[e] == true) {
          temp_iState[e] = 0.0;
          pid_reset[e] = false;
        }
        pTerm[e] = PID_PARAM(Kp,e) * pid_error[e];
        temp_iState[e] += pid_error[e];
        temp_iState[e] = constrain(temp_iState[e], temp_iState_min[e], temp_iState_max[e]);
        iTerm[e] = PID_PARAM(Ki,e) * temp_iState[e];

        //K1 defined in Configuration.h in the PID settings
        #define K2 (1.0-K1)
        dTerm[e] = (PID_PARAM(Kd,e) * (current_temperature[e] - temp_dState[e])) * K2 + (K1 * dTerm[e]);
        pid_output = pTerm[e] + iTerm[e] - dTerm[e];
        if (pid_output > PID_MAX) {
          if (pid_error[e] > 0) temp_iState[e] -= pid_error[e]; // conditional un-integration
          pid_output = PID_MAX;
        }
        else if (pid_output < 0) {
          if (pid_error[e] < 0) temp_iState[e] -= pid_error[e]; // conditional un-integration
          pid_output = 0;
        }
      }
      temp_dState[e] = current_temperature[e];
    #else
      pid_output = constrain(target_temperature[e], 0, PID_MAX);
    #endif //PID_OPENLOOP

    #ifdef PID_DEBUG
      SERIAL_ECHO_START;
      SERIAL_ECHO(MSG_PID_DEBUG);
      SERIAL_ECHO(e);
      SERIAL_ECHO(MSG_PID_DEBUG_INPUT);
      SERIAL_ECHO(current_temperature[e]);
      SERIAL_ECHO(MSG_PID_DEBUG_OUTPUT);
      SERIAL_ECHO(pid_output);
      SERIAL_ECHO(MSG_PID_DEBUG_PTERM);
      SERIAL_ECHO(pTerm[e]);
      SERIAL_ECHO(MSG_PID_DEBUG_ITERM);
      SERIAL_ECHO(iTerm[e]);
      SERIAL_ECHO(MSG_PID_DEBUG_DTERM);
      SERIAL_ECHOLN(dTerm[e]);
    #endif //PID_DEBUG

  #else /* PID off */
    pid_output = 0;
    if (current_temperature[e] < target_temperature[e]) pid_output = PID_MAX;
  #endif
  return pid_output;
}

void manage_heater() {

  if (!temp_meas_ready) return;

  updateTemperaturesFromRawValues();

  #ifdef HEATER_0_USES_MAX6675
    float ct = current_temperature[0];
    if (ct > min(HEATER_0_MAXTEMP, 1023)) max_temp_error(0);
    if (ct < max(HEATER_0_MINTEMP, 0.01)) min_temp_error(0);
  #endif //HEATER_0_USES_MAX6675

  unsigned long ms = millis();

  // Loop through all extruders
  for (int e = 0; e < EXTRUDERS; e++) {

    #if defined (THERMAL_RUNAWAY_PROTECTION_PERIOD) && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
      thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], target_temperature[e], e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
    #endif

    float pid_output = get_pid_output(e);    //****************************************************

    // Check if temperature is within the correct range
    soft_pwm[e] = current_temperature[e] > minttemp[e] && current_temperature[e] < maxttemp[e] ? (int)pid_output >> 1 : 0;

//...original code...

Dimonira, 2015-03-09:
Sory, how to separate code from text?

thinkyhead, 2015-03-09:
Are these useful changes, apart from working around that error? I don't want to change behavior if it can be avoided, unless there's a real benefit. The things is, other changes can also eliminate this compiler error, because it isn't entirely a result of the code. Such changes as changing the order of lines, making statics to replace autos, and even replacing spaces with tabs, are all known to eliminate the error message.

Dimonira, 2015-03-09:
@thinkyhead 
My changes don't change the behavior.
Marlin coding style is very bad: a huge number of preprocessor directives, a large number of global variables, too much complex code in one function, C++ isn't used, etc.
There is no reason to worry. All this code is better to rewrite :) Sory.

thinkyhead, 2015-03-10:
@Dimonira Marlin's architecture has some method behind it. It's not all ad-hoc madness. There are good reasons why it uses compiler directives to limit the code compilation, and for using globals also. At this point I don't see that Marlin for AVR stands to gain much from encapsulating its procedures in C++ classes, because it doesn't have any evident inheritance chain, and with very limited resources malloc/free and new/delete are pretty much verboten. You could take the existing code and wrap it up in classes, but it would pretty much look the same. Instead of globals you'd just have statics, etc.

The existing code is being cleaned up, redundancies are being reduced, and a separate codebase is being developed for ARM, which has more luxury to be fully OOP with the additional resources available on the faster boards.

boelle, 2015-03-10:
has this passed the mark where we can call it verified?

thinkyhead, 2015-03-11:
@boelle Well, it's still kind of spurious. The errors come and go as changes are made to the code. The workaround by @Dimonira happens to also be a code cleanup, so I might merge it. But I do expect to see this kind of error pop up again occasionally.

boelle, 2015-03-12:
if it does... well i'm tempted to close this one... then again....

clefranc, 2015-03-13:
I don't know about others, but every single PR I've tried to compile since we reported this "potential bug" failed with the same error (only the line number changes from time to time).

@thinkyhead @boelle The @Dimonira code is the only one compiling without error.

thinkyhead, 2015-03-14:
#1603 moves the code for pid outputs into separate functions to shorten manage_heater, and hopefully this stops the linker from choking. Move discussion to that PR.

closed 2015-03-14 by thinkyhead