MarlinFirmware/Marlin #1529: Compilation error when EXTRUDERS > 1 and Thermal Runaway commented out

emartinez167, 2015-02-22:
Hello all. I am running the latest rev off this repository, with the following configuration:
Azteeg X3 Pro on RigidBot Big frame, and HPX2-MAX Dual extruder.

I tried disabling the thermal runaway protection as I was doing some tests to try to drive the hot ends to almost 270 Deg C, but kept getting runaway errors. When recompiling, this error came up:

Arduino: 1.6.0 (Mac OS X), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:719:1: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:719:1: error: this is the insn:
(insn 98 97 100 4 (set (reg/v:SF 103 [ pid_input ])
        (mem:SF (post_inc:HI (reg:HI 147 [ ivtmp.212 ])) [4 MEM[base: _113, offset: 0B]+0 S4 A8])) temperature.cpp:503 99 {*movsf}
     (expr_list:REG_INC (reg:HI 147 [ ivtmp.212 ])
        (nil)))
temperature.cpp:719: confused by earlier errors, bailing out
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Is this issue documented?

Regards,

Ernesto.

thinkyhead, 2015-02-22:
Weird compiler errors like this are tricky to track down. Sometimes they relate to weird characters in the source code, while other times it's just the compiler getting confused by some esoteric aspect of the code. Try rearranging code, moving functions around, to see if the error moves around, or if it consistently appears in the same lines.

Dimonira, 2015-03-07:
I did so. File temperature.cpp.
1. Uncomment this lines:
  // static float pid_input[EXTRUDERS];
  // static float pid_output[EXTRUDERS];
and edit their:
  static float pid_input;
  static float pid_output;
2. Comment out or delete line(s) in manage_heater() function:
   float pid_input, pid_output;
That's all.

thinkyhead, 2015-03-09:
@Dimonira See my comment on #1523

boelle, 2015-03-17:
Are the issue still present with latest code and latest non-beta IDE (1.6) ?

https://github.com/MarlinFirmware/Marlin/archive/Development.zip

emartinez167, 2015-03-17:
I will try it out tonight when I get back home and report back to you.

Regards,

Ernesto

emartinez167, 2015-03-17:
I can confirm I can compile on IDE 1.6.1 with EXTRUDERS = 2 and thermal runaway protection commented out.

This can be closed.

Regards,

Ernesto

closed 2015-03-18 by boelle

SydneyMakers, 2015-11-26:
Note for people heading here from Diamond group using their supplied marlin version. Following the steps from Dimonira will fix your issue

Baspro, 2015-12-28:
This solution works for me too.. Thans