MarlinFirmware/Marlin #1530: Last added (odd) functions to LCD Menus.

alexborro, 2015-02-22:
I have just uploaded the latest version and there are a lot of new things included in the LCD menus (REPRAP_DISCOUNT_SMART_CONTROLLER, don't know others).. And I have no idea what they are meant for.. 

In the main menu there is a "test multiline" ??? What the hell is this??

In the Prepare menu there are:
"Check Level" ??? What Level?? 
"Calibrate Bed" ??? Ok, this one is more descriptive, but HOW TO USE it??
"Set home offsets"?? Ok, good description, but again, HOW TO USE it??
"Finalize filament"!!! Oh my gosh, something from Mortal Kombat???

In Preheat menus:
"Preheat ABS 1" Ok, I have one extruder defined.
"Preheat ABS 2" Why, I have no other than extruder 1 defined.

And so one.
BTW, I have no additional feature set in configuration files other than defaults.
I propose:

1) Create a documentation for LCD functions;
2) Keep it up to date when adding new things;
3) When applicable, keep menu functions inside feature compilation flag.

And I'm really curious about "test multiline" function in main menu.

Cheers.

Alex.

alexborro, 2015-02-22:
My mistake on "Preheat ABS 2". The second thermistor was set to -1.. Actually I don't know why thermistor 1 and 2 are -1 by default.. Almost all user has thermistor, not thermocouple.. And most have just one extruder.

alexborro, 2015-02-22:
After looking the ultralcd.cpp, I found:

    // JFR for RMud delta printer
    MENU_ITEM(gcode, "Calibrate bed", PSTR("M702\nG28\nG1 X-77.94 Y-45 Z36 F8000\nG4 S3\nM701 P0\nG1 X77.94 Y-45 Z36\nG4 S3\nM701 P1\nG1 X0 Y90 Z36\nG4 S3\nM701 P2\nM700\nG1 X0 Y0 Z100 F8000"));
    MENU_ITEM(gcode, "Check level", PSTR("G28\nG1 X0 Y0 Z1 F4000\nG1 X-77.94 Y-45 Z1\nG1 X77.94 Y-45\nG1 X0 Y90\nG1 X-77.94 Y-45\nG4 S2\nG1 X-77.94 Y-45 Z0.3 F2000\nG1 X-77.94 Y-45\nG1 X77.94 Y-45\nG1 X0 Y90\nG1 X-77.94 Y-45\nG1 X0 Y0 Z0"));
    MENU_ITEM(gcode, "Retract filament", PSTR("M302\nM82\nG92 E0\nG1 F4000 E-800"));
    MENU_ITEM(gcode, "Insert filament", PSTR("M302\nM82\nG92 E0\nG1 F4000 E60"));
    MENU_ITEM(gcode, "Finalize filament", PSTR("G1 F4000 E790"));

Really?? Something particular to a printer model set to default for ALL users??? What hell is " JFR for RMud delta printer" ???

thinkyhead, 2015-02-22:
@alexborro I mistook those for legitimate new commands, but they are evidently part of some test code that was left behind with the recent multi-line gcode feature. I have a PR pending already patching up the "test multiline" and will remove these others also.

alexborro, 2015-02-22:
Thanks @thinkyhead , I think I downloaded the dev branch before you merge the PR.

Anyway, I think we should try to document the functions. They are nice, but only if the user could use it. 
And the functions I mentioned above is nice as well (Retract, Insert and so one) but it should be user configurable. I use sd files to perform such functions and it would be nice having menus for that.

Cheers.

Alex.

thinkyhead, 2015-02-22:
I tend to prefer using g-code for insert/retract, because the code can use small increments and then it can be interrupted. A single loooonnnggg retract or insert can't be interrupted from the LCD controller.

thinkyhead, 2015-02-22:
"Set Home Offsets" I should document, and it should not be available before homing. Basically, you home the robot, then you adjust its position with the controller (X, Y, Z) until it has a good home position (generally, with the Z at the right height). You then choose "Set Home Offsets" and the current position is applied as the `homing_offset[]` values. It's a quick-and-dirty way to fix up homing when a Z endstop is set too low and it's better not to move it.

avluis, 2015-02-23:
Any way to have these LCD menus show up according to the already set up config file, e.g:
If I have auto bed leveling enabled, should the LCD show options to Home and Auto level if I wish to do so via the LCD, for example - or are these menu items not supposed to be in at all?

alexborro, 2015-02-23:
I think if you have a feature disabled, you should remove the LCD function
menu.

thinkyhead, 2015-02-24:
@avluis It hasn't been previously suggested to remove menu items that are "obsoleted" by an enabled feature – for example, if you have Autolevel enabled, removing other leveling options. In some cases this may make sense.

closed 2015-02-24 by thinkyhead