MarlinFirmware/Marlin #1531: Review configuration files default

alexborro, 2015-02-22:
Related to issue #1530 , I don't know why some configurations are set to "rare" options.

Temperature sensors 1 and 2 are set by default to "-1". How many reprappers over there use thermocouple?? I think majority use thermistor.. and just one extruder.

MOTHERBOARD is set to ULTIMAKER.. I think it should be set to "RAMPS" or "Unknown".. Some days ago I built a machine and forgot to set the Motherboard and of course the machine behavior was really weird.. my fault, but if the default was "nothing", it will break compilation and makes me set it accordingly. 

DEFAULT_AXIS_STEPS_PER_UNIT set to 78.7402 .. Again for Ultimaker.. This is a really unique value.. I think the most common is 80 steps/mm.. 

And so one..

Cheers.

Alex.

thinkyhead, 2015-02-22:
@alexborro The Ultimaker defaults are mostly an historical artifact, from contributors with special affinities for Ultimakers. I agree the best default now is some kind of RAMPS or an UNKNOWN setting that will throw an #error. Common defaults also make sense. Helpful documentation about how to get the steps-per-unit values would also be helpful. In my own config, I keep a block around that looks like this:

    // default settings

    #define XY_BELT_PITCH 2
    #define XY_GEAR_TEETH 20

    #define Z_ROD_PITCH 0.8

    // NEMA17 with 16 microsteps
    #define STEPS_NEMA17 200
    #define MICROS_NEMA17 16

    // PG with 50:1 ratio, 16 microsteps
    #define STEPS_PG50 48
    #define MICROS_PG50 16

    // A typical Wade's Extruder. Gregstruder: 51/11 or 43/10
    #define LG_GEAR_WADE 39
    #define SM_GEAR_WADE 11

    // Diameter of the drive wheel or hobbed bolt
    #define DRIVE_DIAM_WADE 7
    #define DRIVE_DIAM_MK7 10.56

    // calculated step unit values
    #define M_PI 3.14159265
    #define STEPS_PER_TURN_NEMA17 (STEPS_NEMA17 * MICROS_NEMA17)
    #define STEPS_PER_TURN_PG50 (STEPS_PG50 * MICROS_PG50)
    #define GEAR_RATIO_WADE (LG_GEAR_WADE/SM_GEAR_WADE)
    #define GEAR_RATIO_PG50 (50/1)
    #define DRIVE_CIRC_WADE (DRIVE_DIAM_WADE * M_PI)
    #define DRIVE_CIRC_MK7 (DRIVE_DIAM_MK7 * M_PI)

    #define STEPS_PER_MM_XY (STEPS_PER_TURN_NEMA17 / (XY_BELT_PITCH * XY_GEAR_TEETH))
    #define STEPS_PER_MM_Z  (STEPS_PER_TURN_NEMA17 / Z_ROD_PITCH)

    // Some extruder steps-per-mm examples:

    // Classic Wade
    #define STEPS_PER_MM_WADE_39_11 (STEPS_PER_TURN_NEMA17 * GEAR_RATIO_WADE / DRIVE_CIRC_WADE) // = 515.91048

    // MK7 Direct Drive with 50:1 PG motor
    #define STEPS_PER_MM_MK7_50_1 (STEPS_PER_TURN_PG50 * GEAR_RATIO_PG50 / DRIVE_CIRC_MK7) // = 1157.49147

    // A Wade that I bought, 48:10 (?) with 7mm diameter hobbed bolt
    #define STEPS_PER_MM_MYWADE 694.49429712

    #define DEFAULT_AXIS_STEPS_PER_UNIT   {STEPS_PER_MM_XY, STEPS_PER_MM_XY, STEPS_PER_MM_Z, STEPS_PER_MM_MK7_50_1}  // Default for Prusa i3 with GT2 belts

alexborro, 2015-02-22:
I think we should review the config files and default set the most used
values.

thinkyhead, 2015-02-22:
Probably something like a vanilla Mendel i2 or i3.

alexborro, 2015-02-22:
Ok, let's start our guesses.

For cartesians:
DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,500}
This is the most common in my country (GT2 belts, 20t pulley, 1.8º motors
and 0.8mm pitch screw for Z axis).

DEFAULT_MAX_FEEDRATE          {250, 150, 3.5, 25}
This is the values I usually get on Prusas i3.

DEFAULT_MAX_ACCELERATION      {3000,3000,100,10000}
I think the original one (9000 mm/s^2) is too aggressive with no major
benefits.

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1
Again, most common config I see around my friends.

thinkyhead, 2015-02-24:
Looks good, almost exactly my own settings. My default E Steps for an i3 with a Wade extruder and NEMA 17 with 16x micro-stepping is around 695. For my 2engineers PG 50:1 motor (16x) with MK7 drive gear, steps are around 1258. We might list a few common values for XY, Z, and E in the comments section.

As for Max Acceleration, that value has always been a little confusing to me, at least the E value. Whether 10,000 or 9,000 isn't it basically saying "no acceleration"? Or, is there a nuance here I'm missing? How does it interact with the low Jerk setting?

The Temp Sensors are right on. Exactly what most of us have. Thermocouples are far more rare.

thinkyhead, 2015-02-26:
#define DISABLE_MAX_ENDSTOPS

thinkyhead, 2015-02-26:
Might elaborate on the endstop switch values... Do I have this backwards? Can it be reversed due to the electronics?

    const bool X_MIN_ENDSTOP_INVERTING = true; // NC Switch = false, NO switch = true

nophead, 2015-02-26:
The default should be DISABLE_MAX_ENDSTOPS and X_MIN_ENDSTOP_INVERTING =
false as that is the most common usage.

It could be reversed electronically but I don't know of any boards that do
that. It is simply for people that wire their switches the wrong way I
think.

thinkyhead, 2015-02-27:
@nophead Well I mainly want to determine if it's correct that __false__ will always correspond to Normally Closed (NC) and __true__ will always correspond to Normally Open (NO) since the comment there is pretty vague, and AFAIK these are the only two classes of endstop. I understand that electronics could use a closed switch to "pull the signal low" but yeah it must be simpler and cheaper to let the switches be open=low, closed=high...

ZetaPhoenix, 2015-02-27:
NC to ground is normally recommended as is keeps both sides at ground
potential during printing.  I've seen local user who had prints randomly
stop mid-way through prints when they had switches connected as NO. At our
meeting we swapped them around and uploaded the inverted code and it worked
perfectly.

-Jon

thinkyhead, 2015-02-27:
@ZetaPhoenix I'm a big fan of the NC switch, myself, because Induced current doesn't affect them. But can you define "swapped them around"? Do you mean that the endstop plug was just flipped over, or that the NO switches were replaced with NC switches? If the same switches were used, but in a different "mode" then I guess we can leave the comments as they are.

ZetaPhoenix, 2015-02-27:
We swapped contacts on the mechanical switches and used the NC contact from
the NO contact and inverted the logic in marlin to handle the change.

-Jon

thinkyhead, 2015-02-27:
@ZetaPhoenix Great, thanks!

nophead, 2015-02-27:
Some people use optos instead of switches. In that case NC is the most
common but you can get optos with inverted outputs.

NC is preferred because it fails safe if it gets disconnected. It is more
immune to noise but not totally as current can be induced in the loop of
wire giving a voltage at the receiving end despite the switch end being
shorted. It is more immune to capacitively coupled noise but less immune to
inductively coupled noise.

ntoff, 2015-02-28:
>Helpful documentation about how to get the steps-per-unit values would also be nice. 

Put a link to this in the documentation http://www.prusaprinters.org/calculator/

Maybe if we all upload our configs to pastebin or something but here's what I use on my mendel90

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_BED 1

#define HEATER_0_MAXTEMP 260 *because my hot end is peek*

#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,2560,83}  // Mendel90 1/16th microstepping - GT2 belts w/20 tooth pulley, M8x1.25 Z rods,  direct drive extruder 1:1 gearing
#define DEFAULT_MAX_FEEDRATE          {500, 300, 4, 25}    
#define DEFAULT_MAX_ACCELERATION      {5000,5000,25,10000}    

#define DEFAULT_ACCELERATION          2000    

#define DEFAULT_XYJERK                20.0

boelle, 2015-03-10:
This is again one where i think why not have a web config tool that gives you a config file based on what machine you pick and settings etc

thinkyhead, 2015-03-11:
@boelle We can pick up that info from the example_configurations (loading those configs into the web tool) or I can add a bunch of smaller files that only contain the customized settings and merge them into the base config, producing full configuration files for download. The customized settings can be in the form of #defines or in JSON, or just typical INI style...

boelle, 2015-03-12:
:-D me ready to test in about 4 weeks or less.... my bootloader programmer is shot :-/

thinkyhead, 2015-03-13:
I think a recent merge made the default MOTHERBOARD value BOARD_RAMPS_13, which is not defined. I'll make that tweak, plus some others, to fix up default configs soon.

boelle, 2015-05-31:
can this one be closed? did we not agree that ramps should be the default... and if not remebering wrong we also agreed that we would drop having default config unless they are providede by makers of hardware?

Wackerbarth, 2015-05-31:
Two comments on this:

1) "they are provided by makers of hardware" -- Requiring a Custodian for each configuration is reasonable. However, that need not, and for orphan hardware cannot be the maker. The project should assist the community in supporting any hardware configurations for which there is community support.

2) I suggest that there be NO default configuration. The first step in preparing to build the firmware would be to run a configurator or copy an example configuration into the "active position". (As a future enhancement, perhaps this mechanism can be refined into a menu selection, or other mechanism)
However, in the present methodology, there is a bias that makes ALL of the configuration files too complex and complicates maintaining the non-default configurations.

The issue of which cases to test in the Travis CI tests is not at issue. In fact, testing the current "default"
becomes cleaner because it no becomes just the same as any other.

dantman, 2015-06-04:
Commenting on @Wackerbarth's 2). Defaulting to 'nothing' for things that aren't 98% ubiquitous does sound like a good idea.

**Especially** if we can make the Arduino compiler output display errors saying what setting (eg: MOTHERBOARD) needs configuration (even better, where to find it; but maybe that's tricky) or explaining why Marlin can't work reasonably without it.

thinkyhead, 2015-06-10:
As a developer and tinkerer generally I'm a fan of being able to compile Marlin "out of the box" even if it's not for my specific printer. Makes it a lot easier to develop.

Roxy-3DPrintBoard, 2015-06-10:
Agreed.   It should compile and run on the default configuration straight from the download.

closed 2015-09-25 by AnHardt