MarlinFirmware/Marlin #1533: None of the callback menu functions are executed

clefranc, 2015-02-23:
None of the callback menu functions are executed.

This affects:
lcd_control_temperature_menu https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L874
lcd_control_motion_menu https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L997
lcd_control_volumetric_menu https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/ultralcd.cpp#L1033

Originally discovered with https://github.com/MarlinFirmware/Marlin/issues/1532

thinkyhead, 2015-02-24:
Haha! Well things are beginning to make more sense. One question might help me clear it up... the LCD_CLICKED macro.... It should be callable more than once in a loop, but does reading the clicked bit from the controller also clear it??

thinkyhead, 2015-02-24:
This issue probably still exists with these updates, but it's worth a try...
https://github.com/MarlinFirmware/Marlin/pull/1526/files

clefranc, 2015-02-24:
I'll finally can go forward with the multiple filament PR, this nasty behavior was unexpected.

clefranc, 2015-02-24:
The LCD_CLICKED in the BOOL callback function is called only once.

thinkyhead, 2015-02-24:
This rotary encoder code implies that the button bit is persistent - we should be able to use LCD_CLICKED more than once. And in fact, I have used update menu code that calls it twice (for callback menu items) without any issue. So I'm looking closer at the logic, and at the most recent commits to see which might be involved. So far, nothing sticks out!

thinkyhead, 2015-02-24:
Well, BOOL is the only one of the callback functions that isn't defined via the menu_edit_type macro. So, perhaps this is the source of the issue. Perhaps try this, replace the `menu_edit_callback_ ## name` macro with these lines and see if it works...

    void menu_edit_callback_ ## _name () { \
        if ((int32_t)encoderPosition < 0) encoderPosition = 0; \
        if ((int32_t)encoderPosition > maxEditValue) encoderPosition = maxEditValue; \
        if (lcdDrawUpdate) \
            lcd_implementation_drawedit(editLabel, _strFunc(((_type)((int32_t)encoderPosition + minEditValue)) / scale)); \
        if (LCD_CLICKED) (*callbackFunc)(); \
    } \

thinkyhead, 2015-02-24:
Ah, I see you are ahead of me!

closed 2015-02-24 by thinkyhead