MarlinFirmware/Marlin #1538: Adding Steps for second Extruder

ibraheem5000, 2015-02-24:
Hi 

In Marlin, steps/mm, max speed, and acceleration for each extruder are the same. In other words, Marlin assumes both extruders are identical.

But in my case I have 2 extruders with different stepper motor specs...
First extruder have 200 steps per rev. (1.8 deg/step), and second extruder have 400 steps per rev. (0.9 deg/step).

Now I need to make the first extruder steps half the second one!

thinkyhead, 2015-02-24:
@ibraheem5000 It was bound to happen. Although Marlin now supports up to 4 extruders, it does assume they all have identical stepper motors. I'm not sure how deep this limitation goes, but it seems like the kind of thing we should soon address. The data structures around axes are all 4 items long (for XYZE), while the data structures specifically for extruders have as many members as there are extruders - So, E0 E1 E2 E3. As I understand it, extruders are handled in G-Code like the other axes, but even with multiple extruders, E is still a single axis. The G-Code just switches to the next extruder and keeps moving "forward". Although all E axes share the same position-in-mm value, they do not have to use the same steps-per-mm value in the planner. But this is where it may get tricky.

So I'm thinking, initially we can add a special case so that the planner code can draw from a set of steps-per-mm values instead of always using the single steps-per-mm value. We can name it `EXTRUDER_STEPS_PER_MM` and if it exists, then the planner will use its values instead...

    // Enable this if your extruders use different steps per mm
    #define EXTRUDER_STEPS_PER_MM { 695.43, 1251.21, 0, 0 }

MagoKimbra, 2015-02-24:
Thinkyhead If you want see my marlin, i have step, feedrate, accelleration and retraction feedrate for 4 extruder....

thinkyhead, 2015-02-24:
@MagoKimbra Yes I would be very interested to see the changes you made. This is just the sort of configuration flexibility that Marlin should support.

whosawhatsis, 2015-02-24:
While we're at it, I'd love to have the extruders be individually addressable with their own axis letters, so that you can move more than one of them at a time. I'm thinking assign the letters A-D to the extruders, then E is redirected to whichever one is currently the active extruder.

nophead, 2015-02-24:
A-D isn't enough. There needs to be provision for CMYK and white for full
colour mixing. A-E could work.

whosawhatsis, 2015-02-24:
You'd have to skip E, because that's already reserved to redirect to the active extruder. Ideally, the letters would be configurable, so with CMYKW color, you could actually use those letters.

whosawhatsis, 2015-02-24:
Though I'll add that for the special case of color mixing, it might be better to set the ratios of the various colors and then use E as normal. I think Reprappro already has an implementation that's along these lines.

thinkyhead, 2015-02-24:
Well, I presume the letters assigned to the extruders will actually correspond to indices, and they will just need defines that follow that convention, for the sake of configuration. Or do you propose we show these letter names in the interface, and have them be usable in GCode?

whosawhatsis, 2015-02-24:
I'm suggesting making them usable in gcode.

boelle, 2015-05-31:
closing... the idea is mentioned here: https://github.com/MarlinFirmware/Marlin/issues/1342

closed 2015-05-31 by boelle