MarlinFirmware/Marlin #1542: G28 w/ bed leveling probe only homing in corner.

iman7, 2015-02-25:
Makerfarm Prusa i3 
RAMPS 1.4
ReprapDiscount LCD controller
Marlin 1.0.2

When I finish a print from the SD card homing breaks in the sense that the LCD will display the right XY coordinates for the center of the bed but the hotend will stay at XY home and deploy the probe for z homing.

If I press the reset button on RAMPS soon after the print finishes it seems to fix the issue. However If I don't press it and leave it be for a while the method becomes ineffective and I have to plug into repetier-host  and home there which for some reason fixes it. 

Im assuming something weird is going on with Marlin because I've double checked the connections on the LCD controller and the motors and they both seem fine.

thinkyhead, 2015-02-25:
@iman7 Homing with the Z probe is _relatively_ new code, so I wouldn't be surprised if it still has some kinks. So your i3 has no Z endstop, but does have a Z probe, or does it have both?

iman7, 2015-02-25:
@thinkyhead I don't have a Z endstop, used it for the probe. As for the Marlin version. I downloaded the development branch as a zip file in the Feb 9, sorry Im not sure what you mean by the tagged 1.0.2 version.

thinkyhead, 2015-02-26:
@iman7 Welcome to the Github labyrinth... everything just 3 clicks away!

My best suggestion is to try the latest Development version, which has definitely had some bug-fixes since Feb 9, and see if it fixes the problem. If it doesn't, then we are at a good point to start investigating. We definitely want to fix all the bugs we can right now, looking forward to a 1.0.3 release.

boelle, 2015-03-10:
but since you have Ramps why not have a Z endstop and probe? endstop to Z min and probe to Z max ?

iman7, 2015-03-11:
@boelle I don't have access to those black connectors used on endstops locally. I don't feel like having to deal with customs for such a small part though. 

@thinkyhead 

Sorry for the delay. It's exam season where I am and had to put down tinkering for a while.
In that two week span the servo died on me and I went out to get a replacement, the replacement is much worse in the sense that it always homes only in the corner (Using the Feb 9 build). Unplugging the servo still causes it to home in the corner unlike previously where the head would move to the middle even without the servo. 

I also tried Marlin downloaded from git on Mar 11. Z-motors wont move when running G28 after the probe is swung into position but will move fine if manually ordered to. Unplugging the servo still causes the problem of homing in the corner and the lack of movement. 

Its either RAMPS or Marlin now, or could it be the wire inbetween the servo and RAMPS?

[Link to Mar 11 config.h](http://pastebin.com/0W4gBVgM)

thinkyhead, 2015-03-13:
Try enabling `DISABLE_MAX_ENDSTOPS` for a start. Ar.your X and Y endstops the same type? Check your endstop inverting values. Just a couple things I thought of peeking at your config.

iman7, 2015-03-14:
That fixed it! I had to basically turn my printer around by swapping motor directions and the location of the endstop pins on the board but that fixed everything!

Thanks! Kinda funny that everything prints facing away from me though.

closed 2015-03-14 by boelle