MarlinFirmware/Marlin #1543: Missing "}" in Marlin_main.cpp in line 3893

tigertooth4, 2015-02-26:
Hi guys,

I noticed there's back bracket "}" missing in Marlin_main.cpp, line 3893, in the end of the super lengthy function  process_commands(). 

I'm using Xcode, the indentation is incorrect without it (the function list dropdown menu would not show functions behind process_commands().)

Although it seems the compilation is OK without it, it may cause some potential errors. With it, the compilation will pass.

tigertooth4, 2015-02-26:
Sorry! I was wrong. Never mind!

closed 2015-02-26 by tigertooth4

thinkyhead, 2015-02-27:
Thanks for the notice, even if it was mistaken. Braces are an easy one to miss, especially if the indentation isn't consistent. We're working on cleaning up the code a bit so it's more obvious where blocks start and end, and so code-folding will properly work in "naive" editors like TextMate and Sublime.