MarlinFirmware/Marlin #1547: Auto Bed Levelling with grid / 3 points mix up

woodsmoke, 2015-03-02:
Core XY printer, Marlin downloaded 01/03/15. Arduino 1.05

Trying to compile with 3 points method. Turned off grid method. Getting errors on compile:

Marlin_main.cpp: In function 'void process_commands()':
Marlin_main.cpp:1737: error: 'LEFT_PROBE_BED_POSITION' was not declared in this scope
..RIGHT/FRONT/BACK_BED_POSITIONS all not declared.

Marlin Config text below.

#define ENABLE_AUTO_BED_LEVELING
#define Z_PROBE_REPEATABILITY_TEST  

#ifdef ENABLE_AUTO_BED_LEVELING

  //#define AUTO_BED_LEVELING_GRID

  #ifdef AUTO_BED_LEVELING_GRID

    // set the rectangle in which to probe
    #define LEFT_PROBE_BED_POSITION 30
    #define RIGHT_PROBE_BED_POSITION 150
    #define BACK_PROBE_BED_POSITION 150
    #define FRONT_PROBE_BED_POSITION 70

    #define AUTO_BED_LEVELING_GRID_POINTS 2

  #else  // not AUTO_BED_LEVELING_GRID

      #define ABL_PROBE_PT_1_X 30
      #define ABL_PROBE_PT_1_Y 70
      #define ABL_PROBE_PT_2_X 30
      #define ABL_PROBE_PT_2_Y 150
      #define ABL_PROBE_PT_3_X 150
      #define ABL_PROBE_PT_3_Y 150

  #endif // AUTO_BED_LEVELING_GRID

thinkyhead, 2015-03-02:
Should now be resolved in the referenced PR #1548 and the Development branch.

closed 2015-03-02 by thinkyhead

woodsmoke, 2015-03-02:
Bug resolve while one sleeps. Thank you.

PR #1548 (2015-03-02): Fix undefined *_PROBE_BED_POSITION error
- Resolves #1547
- Can now compile with `AUTO_BED_LEVELING_GRID` disabled and `ENABLE_AUTO_BED_LEVELING` enabled