MarlinFirmware/Marlin #1550: Z_RAISE_BEFORE_HOMING / G28 / G28XY

woodsmoke, 2015-03-02:
Core XY printer, Marlin development downloaded 02/03/15. Arduino 1.05

#define Z_RAISE_BEFORE_HOMING 4 

Start Printer, G28 (or) G28XY. Neither command raises the Z before homing.

See #1549 for Marlin config.

thinkyhead, 2015-03-02:
Try adding these lines immediately after line 1638 (`#if defined (Z_RAISE_BEFORE_HOMING) && (Z_RAISE_BEFORE_HOMING > 0)`):

    current_position[Z_AXIS] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);

thinkyhead, 2015-03-02:
Hmm, well never mind! That change won't matter if you have `Z_SAFE_HOMING` enabled.

thinkyhead, 2015-03-02:
So... You should print out some messages to the serial out at the point where it might be failing, and see what's going on. Insert some statements like the following between [lines 1648-1660](https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Marlin_main.cpp#L1648-L1660) and then try `G28` to see how it behaves....

      SERIAL_ECHO_START;
      SERIAL_ECHOPAIR("z dest:", destination[Z_AXIS]);
      SERIAL_ECHOLN("");

thinkyhead, 2015-03-02:
From what I can see, this routine has never been programmed to raise the Z before homing with `G28 XY`. Only homing command that include Z (`G28`, `G28 Z`, `G28 XZ`, `G28 YZ`, and `G28 XYZ`) are programmed to invoke `Z_RAISE_BEFORE_HOMING`, and only `G28` and `G28 XYZ` are programmed to invoke it without first having homed X and Y to a known position (setting the `axis_known_position` flags).

woodsmoke, 2015-03-02:
I'll begin my adventures in coding forthwith

boelle, 2015-03-10:
is this related to #1553 ?

woodsmoke, 2015-03-24:
It works now

closed 2015-03-24 by woodsmoke