MarlinFirmware/Marlin #1553: Z_SAFE_HOMING X/Y_POINT

woodsmoke, 2015-03-02:
Core XY printer, Marlin development downloaded 02/03/15. Arduino 1.05, Config: #1549 

#define Z_SAFE_HOMING_X_POINT (X_MAX_LENGTH/2)    
#define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/2) 

Start Printer, G28 XY, G28 Z    //Gives probe outside bed area error
Start Printer, G28    //Homes XY, Moves to (X_MAX_LENGTH/2),(Y_MAX_LENGTH/2) and Homes Z correctly. 

However:

#define Z_SAFE_HOMING_X_POINT (X_MAX_LENGTH/6)   
#define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH/77)    //Effectively arbitrary coordinates 

Start Printer, G28    //Homes XY, then Z, with no move to Safe Homing Point.

And:

#define Z_SAFE_HOMING_X_POINT 2   
#define Z_SAFE_HOMING_Y_POINT 30  //Same arbitrary coordinates but as integers

Start Printer, G28    //Homes XY, then Z, with no move to Safe Homing Point.

alexborro, 2015-03-02:
Are your probe able to reach such small coordinates like X=2 ??

woodsmoke, 2015-03-02:
In my case, Marlin creates negative XY coordinates for safe homing (HOMING_POINT - PROBE). The coordinates are outside MIN to MAX positions range. 

Perhaps Marlin should generate an error, rather than simply defaulting to Z HOME at 0,0?

woodsmoke, 2015-03-02:
It's the first time I've looked at the Marlin Source code, Apologies

woodsmoke, 2015-03-02:
#if ((Z_SAFE_HOMING_X_POINT - X_PROBE_OFFSET_FROM_EXTRUDER) < 0 )
	     #error "The Safe Z Homing Coordinates are outside the X axis range of travel" 
	  #endif
    #if ((Z_SAFE_HOMING_Y_POINT - Y_PROBE_OFFSET_FROM_EXTRUDER) < 0 )
	     #error "The Safe Z Homing Coordinates are outside the Y axis range of travel" 
	  #endif
	#else

Inserted around about line #1648 perhaps. It's an unlikely mistake to make really.

thinkyhead, 2015-03-03:
Sanity checking these ranges is a good idea. The sanity-checking for config features lives in `Configuration_adv.h` so that's where these checks should probably go. We also have `L`, `R`, `B`, and `F` arguments to `G29` and we need to add sanity checking for those as well.

thinkyhead, 2015-03-11:
#1556 and later commits added sanity checking. See if this issue is still happening with the latest code.

closed 2015-05-09 by thinkyhead