MarlinFirmware/Marlin #1557: active_extruder in plan_buffer_line function

clefranc, 2015-03-03:
I'm a beginner here, not the best coder, but looking at the "plan_buffer_line" function (planner.cpp),

https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/planner.cpp#L533

there is some use of "active_extruder" in it. I see the function is called with "active_extruder", but inside the function should it not use its argument name "extruder" instead?

It perhaps has no impact on the result of the code, just that I find it odd.

thinkyhead, 2015-03-04:
@clefranc `active_extruder` is a global extern exposed in `Marlin.h`. This global is used in `plan_buffer_line`, and the `extruder` argument to plan_buffer_line is also used (for example, line 658). There might be some instances where these values can differ, so I'm hesitant to make any changes here just yet. We should look more closely at these vars, however.

boelle, 2015-03-18:
can this one be  closed or?

thinkyhead, 2015-03-19:
planner.cpp will be getting more attention soon.

closed 2015-03-19 by thinkyhead