MarlinFirmware/Marlin #1559: Strange behaviour when trying to use single motor for two extruders (HPX2-MAX)

emartinez167, 2015-03-04:
Trying to use a single step motor to drive two extruders, I tried using E0 pins assigning to E1, but when executing, X axis moves (Running on Azteeg X3 Pro)

boelle, 2015-03-10:
did you solve this one? and where did you get the firmware from?

emartinez167, 2015-03-10:
Haven't been able to fix it. I am running the latest firmware from this repository. Trying to work out what is wrong with the guys from Dglass3D so please close it for now.

Regards,

Ernesto

boelle, 2015-03-10:
looking at that extruder....

is the servo not used to decide what extruder is active?

emartinez167, 2015-03-10:
That is correct, but since they are both driven by a single step motor the
proposed solution from them was to assign the same pins for E0 extruder to
E1. That does not work as for some reason the motor foes not turn and
instead the X axis moves!!! Any idea what might be happening there?

boelle, 2015-03-11:
their idea is wrong....

what i would do when going for the other colour is to retract say maybe 2-3mm

then activate the servo to change colour... and then feed the other colour 2-3mm

from the videos it seems that when the servo change colour it also disables the extruder not in use... and it lifts the extruder a bit at the same time so it will not hit stuff already printed.

and as a start sequence before printing i would extrude out plastic from both extruders and make some Fat lines on the bed to prime and make them ready...

boelle, 2015-03-11:
https://www.youtube.com/watch?v=yJ7D1I59mFo&t=17

magic starts there

scotty1024, 2015-03-11:
Actually their idea is correct but you don't understand what is going on. In the video you aren't just seeing an HPX2 at work. The hot end lifting action isn't being supplied by the HPX2 but rather by special hot ends. The lifting action isn't to avoid hitting but rather to snap off the end of the filament to prevent dragging a bit of that material onwards as printing begins with the other hot end. The lifting action also works using a non-HPX2 extruder. And in fact absolutely prevents stringing for a single hot end printer making for cleaner prints with less retraction which speeds up prints eg they provide benefits to non-multiple extruder printing.

https://www.kickstarter.com/projects/dglass3d/autolift-precision-3d-printer-hot-ends

emartinez167, 2015-03-11:
Scotty is right. What is not happening is that when the command to change
extruders hits Marlin, it should still use the same motor, but it moves the
X axis motor!!!

boelle, 2015-03-11:
but i can see so much that the servo changes extruder and one stepper
drives filament....

i can also see that when the servo changes the extruder one extruder gear
disengages and the other engages...

hence you dont need to do the trick they tell you to make it work....

to make it very simple to understand:

1) retract extruder  2-3mm
2) make servo move enough to fully disengage active extruder gear.. and
fully engage the other
3) prime extruder 2-3mm

emartinez167, 2015-03-11:
I am not making myself clear. Since when you compile Marlin you specify 2
extruders, the command to change extruders also changes the motor pins to
those configured for extruder 2 as Marlin expects one motor per extruder.
Am I right?

boelle, 2015-03-11:
i'm not making myself clear either...

for this extruder you only need to define 1 extruder... and use servo to change colour...

so in gcode its prints..
then it stops
retracts
moves the servo
prime
print continues...

as i see this thing there are

2 hotends
1 stepper
1 servo

servo controls the gears that decides what hotend is in use...
lift and drop of hotend is controlled by retract and prime

so unless i'm very wrong you dont need 2 extruder steppers... just 1

boelle, 2015-03-11:
autolift as i expected "The AutoLift system automatically lifts the nozzle during filament retraction.."

so nothing magical there...

emartinez167, 2015-03-11:
I see so it is ok to declare 1 extruder in Marlin and 2 in Cura? The other
problem then lies in the code to swap extruders. It states in Marlin_main
that if EXTRUDERS > 1 then extruder swapping is enabled... How do we go
around that?

boelle, 2015-03-11:
i guess marlin expects multiple extruders to be multiple... ie 1 stepper for each extruder... 

this extruder is a special case...

but i take back what i have said partly.... there is something wrong here

if 2 steppers are using the same pin it should only drive the same stepper... that i will give you

but your extruder is still special in the swapping process... to swap it needs first to retract to lift the active nozzle... then it has to move the servo... then prime to lower the now active nozzle...

not a coder here... but an idea:

a define for your kind of extruder... so when defined it will use the same stepper for extruder 0 and 1

also if defined it will tell marlin to use servo 1 to do swap...

but we need some sort of code to tell it to swap... and then a routine to do the swap itself

emartinez167, 2015-03-11:
It already does the swapping by adding extra code and calling a new
function that moves the servo to the other position. I can send you the
files that are changed as a result of this extruder, perhaps that should
give you a better idea.

boelle, 2015-03-11:
post them to paste bin http://pastebin.com/

the coders might want to have a look

but would it not be better to directly define this extruder and just send a M code to do the swap? say M799 to do a swap

emartinez167, 2015-03-11:
I will leave the decision to you as I do not thoroughly understand how
Marlin's internals work. I have also asked Brian at Dglass3D to take a
look. (Cc'd)

Brian, just trying to figure out how to make the HPX2-MAX with the Marlin
team. Perhaps you an shed some light here on how you made it work in
earlier versions?

scotty1024, 2015-03-11:
Bo,

What is going on here is the OEM didn't upstream their patches to make their proprietary extruder work. 

The OEM patched their way to 4 extruders last Spring (2014) before Marlin was upgraded to support the 4 extruder E3D Kraken over the last couple of months (2015).

My guess is the Kraken effort modified something the OEM patches relied upon.

I suggest this issue be renamed "Modify Marlin to support DGlass3D HPX extruders" and I will develop, test and produce a PR.

I've been looking for something to sink my development teeth into on Marlin.

I backed both DGlass3D Kickstarter projects. I have an HPX2-MAX with two of their miniaturized all metal hot ends and I have two of their Autolift hot ends. They've all been on the shelf waiting for me to go produce Marlin firmware to use on a Trinity Labs Aluminatus on a RRD Rumba.

The time has come. Any help Brian wants to offer is welcome. If he wants to own this I'm happy to step back to just testing.

emartinez167, 2015-03-11:
Excellent! Would you like me to send you my source? It already has the code from Dglass3D, just not working properly on the motor, but servo works ok.

Regards,

Ernesto

boelle, 2015-03-11:
yep, post the important files to paste bin http://pastebin.com/

i'm not one of the coders so sending me direct will not help much

scotty1024, 2015-03-11:
Ernesto, absolutely. Im guessing you would be available to help test? What hardware are you using?

emartinez167, 2015-03-11:
Certainly willing to test it! I hope this pastebin link works. Let me know.

http://pastebin.com/Xh11RRe0 <http://pastebin.com/Xh11RRe0>

I am running a RigidBot Big frame with an Azteeg X3 Pro board, HPX2-MAX dual extruder and latest Dev Marlin (I try to refresh daily).

Regards,

Ernesto.

thinkyhead, 2015-03-13:
If this is the switching extruder I'm thinking of, it uses a single motor and it needs to reverse which direction is "forward" when switching inputs also, because it engages the filament on the other side of a single drive wheel.

emartinez167, 2015-03-13:
That is correct. You have to reverse direction when extruding on the second
nozzle.

thinkyhead, 2015-03-13:
This is enough of a special case to warrant its own feature, rather than trying to shunt it into any existing paradigm. Perhaps switching extruders generally...

boelle, 2015-03-13:
i agree, thats why i suggested some sort of define thats says what type of
extruder

start with a few defaults one which could even contain steps pr mm etc...
just a thought thou

Den fredag den 13. marts 2015 skrev Scott Lahteine <notifications@github.com

boelle, 2015-03-13:
hmm before i forget... thou this is not related byt some sort of auto conf
routine that could help dial in or fine tune all params for a given
machine???

Den fredag den 13. marts 2015 skrev Bo Herrmannsen <bo.herrmannsen@gmail.com

boelle, 2015-05-31:
will close this one as we have agreed that hardware support must come from the makers of the hardware in form of a PR

closed 2015-05-31 by boelle