MarlinFirmware/Marlin #1562: Motors not working after trying Repetier and coming back to Marlin

eelmasllari, 2015-03-05:
Hello! 
My Prusa i3 worked fine with Marlin v1. I updated to Marlin v2 (OK), then I uploaded Repetier for a test but decided I preferred Marlin, so I reverted to Marlin v2 again. The motors now don't move; they get signals, they are active and "locked" into position, but only hum and vibrate lightly and don't move. Same thing happens when I downgrade to Marlin v1, the exact code that used to work fine before. When I upload Repetier again, everything works. I upload Marlin, it stops working. (With v2 no motor works, with v1 X and Y don't work but Z does)

I tried erasing the EEPROM completely, but it didn't change anything. Tried increasing current to the motors, again no change. Moved the Z motors to the X/Y pins, nothing.

Any idea what may be the cause or how I can fix this?

This is my Repetier config

    #ifndef CONFIGURATION_H
    #define CONFIGURATION_H
    
    /**************** READ FIRST ************************
    
       This configuration file was created with the configuration tool. For that
       reason, it does not contain the same informations as the original Configuration.h file.
       It misses the comments and unused parts. Open this file file in the config tool
       to see and change the data. You can also upload it to newer/older versions. The system
       will silently add new options, so compilation continues to work.
    
       This file is optimized for version 0.92
       generator: http://www.repetier.com/firmware/v092/
    
       If you are in doubt which named functions use which pins on your board, please check the
       pins.h for the used name->pin assignments and your board documentation to verify it is
       as you expect.
    
    */
    
    #define NUM_EXTRUDER 1
    #define MOTHERBOARD 33
    
    #include "pins.h"
    
    // ################## EDIT THESE SETTINGS MANUALLY ################
    // ################ END MANUAL SETTINGS ##########################
    
    #define FAN_BOARD_PIN -1
    
    //#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
    // Uncomment the following line if you are using arduino compatible firmware made for Arduino version earlier then 1.0
    // If it is incompatible you will get compiler errors about write functions not beeing compatible!
    //#define COMPAT_PRE1
    #define MIXING_EXTRUDER 0
    
    #define DRIVE_SYSTEM 0
    #define XAXIS_STEPS_PER_MM 80
    #define YAXIS_STEPS_PER_MM 80
    #define ZAXIS_STEPS_PER_MM 4000
    #define EXTRUDER_FAN_COOL_TEMP 50
    #define PDM_FOR_EXTRUDER 0
    #define PDM_FOR_COOLER 0
    #define DECOUPLING_TEST_MAX_HOLD_VARIANCE 20
    #define DECOUPLING_TEST_MIN_TEMP_RISE 1
    #define RETRACT_ON_PAUSE 2
    #define PAUSE_START_COMMANDS ""
    #define PAUSE_END_COMMANDS ""
    #define EXT0_X_OFFSET 0
    #define EXT0_Y_OFFSET 0
    #define EXT0_STEPS_PER_MM 721.19
    #define EXT0_TEMPSENSOR_TYPE 1
    #define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
    #define EXT0_HEATER_PIN HEATER_0_PIN
    #define EXT0_STEP_PIN ORIG_E0_STEP_PIN
    #define EXT0_DIR_PIN ORIG_E0_DIR_PIN
    #define EXT0_INVERSE 1
    #define EXT0_ENABLE_PIN E0_ENABLE_PIN
    #define EXT0_ENABLE_ON 0
    #define EXT0_MAX_FEEDRATE 50
    #define EXT0_MAX_START_FEEDRATE 20
[... 1133 lines cut ...]
    #endif
    
    #ifdef DOGLCD
    # ifndef DEFAULT_LCD_CONTRAST
    #  define DEFAULT_LCD_CONTRAST 32
    # endif
    #endif
    
    #define SOFT_PWM_SCALE 0
    
    
    #define FILAMENT_SENSOR_EXTRUDER_NUM	0  //The number of the extruder that has the filament sensor (0,1,2)
    #define MEASUREMENT_DELAY_CM			14  //measurement delay in cm.  This is the distance from filament sensor to middle of barrel
    
    #define DEFAULT_NOMINAL_FILAMENT_DIA  3.0  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation
    #define MEASURED_UPPER_LIMIT          3.30  //upper limit factor used for sensor reading validation in mm
    #define MEASURED_LOWER_LIMIT          1.90  //lower limit factor for sensor reading validation in mm
    #define MAX_MEASUREMENT_DELAY			20  //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)
    
    #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially 
    
    #include "Configuration_adv.h"
    #include "thermistortables.h"
    
    #endif //__CONFIGURATION_H

thinkyhead, 2015-03-05:
What does Marlin report for `M501`?

eelmasllari, 2015-03-05:
SENDING:M501
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E721.19
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z1.00 E45.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2000 Y2000 Z200 E2000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S180.00 T60.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P149.63 I11.42 D490.33

eelmasllari, 2015-03-05:
Well, it's working now. I don't know which of the two steps below did it, but this is what solved it:
1) updated Arduino IDE to latest and compiled/uploaded again
2) in Printrun (Mac Pronterface), which I'm using for testing, increased the speed from 60mm/min to 100mm/min. I had set it that low so I would have time in case I had put the directions inverted, but apparently the motors didn't like that speed. Fun thing is, when I set it back to 60, they work now.

Thanks!

closed 2015-03-05 by eelmasllari