MarlinFirmware/Marlin #1564: Specific Configuration files: difficult to maintain.

alexborro, 2015-03-05:
Hi folks,

I'm adding a new define to configuration.h and thinking about I need to change all the specific configuration files, otherwise the compilation will break.

That is not funny..

What if we include a "main" configuration file in the beginning of the specific configuration files so they just redefine the already defined labels?? I mean, If I add something new to the "main" config file I do not need to adapt all the specific configs. It is suppose to have a responsible for each specific config file to keep them up to date.

thinkyhead, 2015-03-05:
We've talked about the idea of just using overrides before, perhaps as a standard thing. Always included before `Configuration.h` and `Configuration_adv.h`. But it requires most if not all configuration defines to be wrapped in #ifndef. So, another option is to include a blob file that does the overrides, which can be loaded into my fabulous web-based configurator, and it spits out the updated configuration files as a zip.

alexborro, 2015-03-05:
You can define the same label twice.. The preprocessor will issue a Warning (among millions we already have) and assume the last defined value.

Wackerbarth, 2015-03-06:
Rather than including the overrides BEFORE the default definition, they can be included AFTER the defaults. As alex notes, the preprocessor will "do the right thing". If you don't want the warnings, you can always #undef something immediately before you redefine it.

IMHO, the current configuration is really poor because values are not shared. Rather than duplicating the definitions for each different configuration, a scheme that allows the definitions to be inherited, with overrides, is much easier to maintain.

alexborro, 2015-03-06:
That's my point @Wackerbarth .
Today's model is too much complicated to maintain.. If I add one #define to main config file, I need to replicate it to dozen other specific files.. 

I will do this mod soon.

Cheers.

Alex.

Wackerbarth, 2015-03-06:
__@alexborro__: I hope that you didn't think that I was questioning your approach. My intent was to support the idea. I was trying to point out that the __#ifndef__ structure isn't necessary and that it is not difficult to avoid the warnings.
The main difficulty is in getting coders to add things in a structured way. On far too many projects, I see where someone has forked the configuration and added their feature at the top level rather than maintaining a structure that can be shared with other configurations.

alexborro, 2015-03-06:
Perfect Richard, I've got your point. I'm gonna do exactly what you said..
Load the defaults and then redefine the specific labels.

Cheers.

Alex.

thinkyhead, 2015-03-06:
@alexborro The problem is that any override concept has to be applied in front of the configuration files (currently) because they use #ifdef to handle some things. In order to do overrides following the configuration files, everything that would be altered would have to be included.

alexborro, 2015-03-06:
@thinkyhead , let me try an example.

--------------------- configuration.h ----------------------
#define A = 10
#define B = 20
#ifndef C
#define C = 30
#endif
#include "config_specific.h"

----------------------config_specific.h--------------------
#define B = 60
#define C = 10

Result:

A = 10 (from configuration.h)
B = 60 (from config_specific.h)
C = 10 (from config_specific.h)

If you have any #ifndef in the config_specific.h just to avoid redefining, it must be removed.

cheers.

Alex.

thinkyhead, 2015-03-06:
Also would require the defaults in the configurations to be neutral so there's very little to "undo." Also it would require that sanity checks be moved elsewhere besides the end of Configuration_adv.h.

Wackerbarth, 2015-03-06:
But I suggest that the "sanity checks" be in a separate file from the specification of the configuration anyway. That "blob" could be included from the main routine.

alexborro, 2015-03-06:
@thinkyhead  That's the point we were discussing in other topic: making the configuration.h as neutral as possible.

And create a new Error_Check.h to be included after all configuration files.

thinkyhead, 2015-03-06:
@alexborro We'll have to see how close we can get to a neutral configuration. I believe it's customary to call files that test for invalid conditions `Asserts.h` or `Asserts.cpp`.

ntoff, 2015-03-09:
>But we want things to generally work for users "out of the box" and hopefully they don't need a configurator 

This will never happen unless you pre-install the firmware for them. There are too many different hardware configurations to worry about trying to make things work for everyone by default.

I've seen someone ask for help on why nothing moves, turns out they hadn't even changed the control board from default values.

Just because it looks simple to us, doesn't mean it will be for everyone else.

thinkyhead, 2015-03-09:
@ntoff Huh, well we could have the most essential settings set to some bad values that we test for, meaning users haven't changed them. Throw compiler errors telling users to set them...

boelle, 2015-03-10:
just a wild thought here.... but if the web config tool was developed some more could it not be made to build the config files ? ie user picks machine etc and it gives out the config file?

and if the settings does not match we should make so the tool allows uploads of config files so we can include the settings next time someone picks the same machine...

again just a wild thought... prop total bonkers

thinkyhead, 2015-03-20:
#1651 is a start...

boelle, 2015-05-31:
thin we recently agreed to strike all those special config files unless they are providede by someone and also maintained by that person... so as marlin dev we only have to think of the main config files-.

and closing on that same reason

Wackerbarth, 2015-05-31:
I disagree with closing this issue.  Just because you expect someone to maintain a configuration is no excuse to make their life more difficult by having a mechanism that is greatly biased toward some particular configuration. Unless you wish to see Marlin used by only a particular subset of the overall community, you need to maintain a structure that permits everyone to maintain compatibility with your configuring mechinisms.

thinkyhead, 2015-06-09:
> make their life more difficult by having a mechanism that is greatly biased toward some particular configuration

Can you provide an example where this is the case?

In order to overcome the basic limitations of Embedded C++ and make the `#define` options actually override-able, particularly by a Makefile or other external method we should probably merge #2169 first.

Wackerbarth, 2015-06-09:
Well, for starters, in the present scheme, EVERYONE is expected to have a file full of all of the possible options, often with parameter values that apply only to a different system.

Also, currently, there is not mechanism to build a configuration "like ... except ...."

I agree that #2169 is a step in the right direction, but I consider it only a start to addressing the underlying situation.

But #2269 should go in first. Then #2169, and from there continued incremental improvements.
Remember that this entire subject is about infrastructure, the mechanism to select the lines of code that will actually reflect in the firmware image.

The more organized we can be in the way the options are presented, the easier it will be for the users to understand how they should approach customization.

 We don't have to "finish" this process to see some improvement. Every little bit helps.

closed 2016-05-04 by jbrazio