MarlinFirmware/Marlin #1566: Documentation - Is there a Changelog?

thiagopeixoto16, 2015-03-05:
Do we have a changelog? 
If not, is there any plans to make one? 

It will be great for the users to evaluate an version upgrade.

Thanks!

boelle, 2015-03-05:
there are a kind of... on each release there are a link where all the commits can be read...

but its mostly still bugfixing... new features are last in line

thiagopeixoto16, 2015-03-05:
Okay sir, thanks!

thinkyhead, 2015-03-06:
The subject has come up recently. It takes some work to make a concise changelog. But one of the easier ways to gauge changes is to compare the Configuration files between releases. These reveal any new features and sometimes fixes. Bug-fixes and other enhancements take more work to track down in the issues list and commit logs. It also takes some extra work to maintain a Known Bugs list for tagged releases. The thing is, Marlin is always changing (improving!) and it's just hard to keep up with so much fluctuation.

boelle, 2015-03-10:
i agree... but will close this one so we dont end up with an mount everest list of issues... but once we start to get to a a state we can call stable we should start to make a change log of some sorts

closed 2015-03-10 by boelle