MarlinFirmware/Marlin #1573: Smart controller issue with latest Marlin. #1498

haqnmaq, 2015-03-06:
So someone else had this problem. It was #1498 https://github.com/MarlinFirmware/Marlin/issues/1498#issuecomment-77497895 , but they closed it without any solutions or possible solutions. I posted some comments in that thread that will hopefully narrow down the problem. Basically the Reprap Discount smart controller does not work anymore with Marlin on my RUMBA. The backlight lights up, but doesnt show any characters. I have more details in the other thread, but I believe this is a bug, based on the fact that I am not the only one experiencing the problem. I tried trouble shooting the problem, but to no avail. Let me know if you need more info.

alexborro, 2015-03-06:
I have a RUMBA + RRD Smart Controller working fine on the latest Marlin.

Cheers.

Alex.

haqnmaq, 2015-03-06:
Care to share your config file? Particularly pertaining to the lcd. I don't get any errors when compiling, until I comment out the smart controller. Once I comment that out, I get the compiling errors and faulty lcd problem. Also, how recently did you update? I don't know how often they update Marlin, but my build of Marlin is only 2 days old.

haqnmaq, 2015-03-06:
Here is someone else with the same problem. http://forums.reprap.org/read.php?146,445885 Unfortunately, they didn't provide any solutions.

alexborro, 2015-03-06:
@haqnmaq , what *exactly* LCD do you have??
The Graphic LCD is one type, the Smart Controller is other. 
Share some pictures and I will try to help you.

Cheers.

Alex.

thinkyhead, 2015-03-06:
Check against the latest code which just got a potential solution (language defines conflict?). I'm skeptical that it fixes it, but I can check it out later if the problem persists.

haqnmaq, 2015-03-06:
@alexborro It is the Smart Controller, not the larger Graphic lcd. @thinkyhead I will check that tonight and let you know. Thanks for the help btw.

haqnmaq, 2015-03-06:
Ok, so I was able to narrow down the problem with a brand new version of Marlin. So by just activating #define REPRAP_DISCOUNT_SMART_CONTROLLER, I don't get any compiling errors. But when I also activate Auto Bed Leveling, I get these compiling errors below. I am not near my printer right now, so I don't yet know if these errors will cause the problems with the lcd, but considering that I got similar errors yesterday, I don't think that the lcd will work.

SdFile.cpp.o:(.rodata._ZTV6SdFile+0x8): warning: internal error: out of range error
LiquidCrystal\LiquidCrystal.cpp.o:(.rodata._ZTV13LiquidCrystal+0x4): warning: internal error: out of range error
LiquidCrystal\LiquidCrystal.cpp.o:(.rodata._ZTV13LiquidCrystal+0x6): warning: internal error: out of range error
LiquidCrystal\LiquidCrystal.cpp.o:(.rodata._ZTV13LiquidCrystal+0x8): warning: internal error: out of range error
Wire\Wire.cpp.o:C:\Program Files (x86)\arduino-0023\libraries\Wire/Wire.cpp:256: warning: internal error: out of range error

haqnmaq, 2015-03-06:
So it seems to have something to do with the Auto Bed Leveling. It's only when the Auto Bed Leveling and the Smart Controller are activated when I get the compiling errors.

nophead, 2015-03-06:
Perhaps the code is too big for your chip.

haqnmaq, 2015-03-06:
Binary sketch size: 83140 bytes (of a 258048 byte maximum)

Doesn't seem like it.

thinkyhead, 2015-03-06:
@haqnmaq I just tested with the latest code. Starting with the default configuration, I simply enabled `REPRAP_DISCOUNT_SMART_CONTROLLER` with the JAPAN character set option. No compile errors. I then enabled `ENABLE_AUTO_BED_LEVELING` and got no errors. The ugly error you're getting is a linker error, and unless it's a "missing symbol" error, it's unlikely to be code-related.

You will get errors if you fail to define either `DISPLAY_CHARSET_HD44780_JAPAN` or `DISPLAY_CHARSET_HD44780_WESTERN`. I am adding an error check now to point this out.

haqnmaq, 2015-03-06:
Ok, let me go through it again real quick.

haqnmaq, 2015-03-06:
Huh. I Just enabled the Smart controller and autoleveling, with and without DISPLAY_CHARSET_HD44780_JAPAN or DISPLAY_CHARSET_HD44780_WESTERN enabled and I still get the error on both Arduino 23 and Arduino 1.0.0

thinkyhead, 2015-03-06:
I'm using 1.5.2 at the moment.

haqnmaq, 2015-03-06:
Ok, let me try that version.

thinkyhead, 2015-03-06:
Also successfully compiled with 1.0.5.

haqnmaq, 2015-03-06:
It compiled without errors in 1.5.2! Same exact file that I tested in 23 and 1.0.0. I used to remember they reccomended using arduino 23 for Marlin. Has that since changed? Thanks again for your help, I shouldnt have any other problems now, but I will let you know in a few.

thinkyhead, 2015-03-06:
Arduino 1.0.1 added the __--relax__ compiler option, which works around this problem.

haqnmaq, 2015-03-06:
Awesome! It works! My printer is up and running again. Thanks for your help! Sorry for the runaround looking for errors.

thinkyhead, 2015-03-06:
Not a problem. It was a good chance to make a better error message when the controller ROM isn't selected, and now we now the code is getting too big for old-school Arduino.

closed 2015-03-06 by thinkyhead