MarlinFirmware/Marlin #1574: G29 - New homing behaviour?

avluis, 2015-03-06:
Just tried newest push on dev branch after merging on my fork for a Prusa i3v 12" with ABL.
Did the G29 command get modified/improved to home to center once done (before it stayed at the last corner where it proved) with the Auto Bed Leveling routine?

In an somewhat related note, I like to extrude a bit of filament (call it a anchor) before starting a print (usually along with a brim and/or skirt) and notice the new G29 behavior - should I modify my slic3r start G-code to move to a corner and do this or will this behavior change (not final, bug, etc)?

Wurstnase, 2015-03-06:
Why should the printer move anywhere after a G29?
Use a G1 to move your printer in any edge you want.

thinkyhead, 2015-03-06:
G29 did just get an update, so it's possible there may be a typo or bad logic. I'd like to hear more info about how it's behaving. Check out the code comments in Marlin_main.cpp in front of the gcode_G29 function for the latest flags. Also watch the serial output when using G29 to see if it reports anything unusual.

avluis, 2015-03-06:
Got it, trying another print in a few to verify.

thinkyhead, 2015-03-06:
Aha, good that is seems to be working correctly after all. I ought to put in a compile-time check for bad default values so that will be caught early. I should also include the probing limits in the serial error output.

thinkyhead, 2015-03-07:
Bed size has been used as the hard limit on movement up to now. If we keep it that way, then you would have to increase the bed size in the firmware to allow the probe to go over the whole bed area. On the other hand, the total freedom of movement of the XY axes could be specified separately, and this would allow full motion. The downside is it would mean a pretty major change to a core configuration option. My current solution of limiting the probe to places where the nozzle is still within the bed may be a minor bummer, but it's consistent with the settings, which reasonably should forbid going past the set limits.

avluis, 2015-03-07:
Nah - it makes sense to limit to the bed area - keeps things from crashing into each other.
So while it may not be ideal for many other reasons, there really aren't many ways to get around this limitation - after all, a 3D printer's bed has a dimensional limit and most extruder assemblies will be moving about the bed anyways.
In the end, it is nothing major once documentation is updated (provided) and takes into account the appropriate methods to setting probe limits.

nophead, 2015-03-07:
Note this is not limiting to the bed size as that is not defined in the
firmware but is in the host. It is limiting to the axis limits, which by
definition is as far as the firmware is allowed to move the axes. Some
machines can travel past the bed and others can't.

thinkyhead, 2015-03-07:
@nophead Ah yes, well I am using `MAX_X_POS`, etc., as the limit for nozzle position, so it sounds like these added sanity-checks are doing the right thing. I actually set my printer's axis limits to the same as my bed size (it can't go much further anyway) so I forgot that it wasn't actually the bed size, and it could differ from the bed size set in the slicer.

nophead, 2015-03-07:
If it can go further then you should probably make your axis limits bigger as it is advantageous if you have a probe.

ntoff, 2015-03-09:
>Some machines can travel past the bed and others can't.

Yeah my k8200 used to home to where the nozzle was about 3mm below the bed and about 5mm outside of the print area in X and Y. I had about 220mm of travel in all directions, but the actual print bed itself was still only 200x200.

thinkyhead, 2015-03-09:
Can someone confirm that it works okay to set `X_MIN_POS` to a negative value? Of course, only bots with a MAX endstop will be able to test this...

vandarin, 2015-03-09:
Yes, MIN_X_POS is fine being negative. I have 280mm of travel from -17 to +264.

thinkyhead, 2015-03-09:
@vandarin Excellent. I guess that's the proper solution for users experiencing issues because of the new sanity-checking. Maybe I can update the error messages to point this out.

alexborro, 2015-03-10:
I have negative X values even homing to X-Min. My X range is -4 to 325.

Working perfect.

Cheers.

Alex.

alexborro, 2015-03-10:
Scott, let's put all preprocessor sanity check in a separate file.. To make
config files clean.

Cheers.

Alex.

thinkyhead, 2015-03-10:
@alexborro Gladly! I will roll a PR soon.

boelle, 2015-03-17:
:-D shold this one stay open until the PR is merged?

thinkyhead, 2015-03-19:
I have the sanity check issue in mind, meanwhile the original issue is all set.

closed 2015-03-19 by thinkyhead