MarlinFirmware/Marlin #1580: Sanity check and bed leveling

mStrangers, 2015-03-07:
I have problem with the sanity check for bed leveling point.
i use prusa i3 with servos mounted 23mm right from extruded , the x axe can move below the 0 for bed leveling (set at 15mm and end stop set to max travel) for left probing but there is an error in sanity check ,can you add an option for disable sanity check for bed leveling ?

thinkyhead, 2015-03-09:
It might not need it. What is the X_MIN_POS you have set in Configuration.h?

boelle, 2015-03-10:
is this still relevant or can we close it?

mStrangers, 2015-03-11:
Sorry for the delay, lot off work.
the X_MIN_POS is set to 0.
i haven't tested with -8 (i've modified the testing code in main instead).
i'll test with X_MIN_POS -8 when my work give me the time.

mStrangers, 2015-03-11:
i think is the same as #1574 set x_min_pos to negative value do the job.

closed 2015-03-11 by mStrangers