MarlinFirmware/Marlin #1588: Arduino IDE 1.6.0... who has the brains to update all in ArduinoAddons ?

boelle, 2015-03-09:
as per subject i wonder if it was not an ide to update everything so we only have one addon folder ?

gege2b, 2015-03-09:
I've correctly uploaded Marlin to a sanguino using the 1.6.0 IDE by just copying the 1.5.x folder from ArduinoAddons (after having corrected the typo)

as mentioned here #1535 , the only thing I had to change was the line "sanguino.upload.protocol=stk500" 
to
"sanguino.upload.protocol=stk500v1"

Anyway, it just worked on my old sanguino so it needs more testing

boelle, 2015-03-09:
i was thinking of everything :-D

but i have to sort an issue with my 644 based board... i cant burn a
bootloader ...

drogge, 2015-04-17:
Hey gege2b, where exactly did you copy the ArduinoAddons/Arduino_1.5.x to in the arduino.1.6.0 tree?

Drew

gege2b, 2015-04-19:
Hi,
I've copied all the files  in "Hardware" directory, following the same tree structure as in the repo 

./arduino1.6/hardware/marlin/avr/*

drogge, 2015-04-19:
Thanks for the info gege2b. I'll give it a try.

gege2b, 2015-04-20:
If you have a sanguino, dont forget to read https://github.com/MarlinFirmware/Marlin/pull/1535#issuecomment-75564108

I've lost a few hairs because of this ^_^

Wackerbarth, 2015-05-31:
This one will be back .... There are issues with some particular processors and, currently, NONE of them are being tested.

boelle, 2015-05-31:
i just think it was a brainfart that i created the post in the first place...

Wackerbarth, 2015-05-31:
Actually, I suggest that we enter into a deliberate effort to support the IDE 1.6 and deprecate ALL prior versions.  This won't happen overnight, but should be a development goal and, in particular, any changes that work against that goal should be avoided.

boelle, 2015-05-31:
i can see the advantages.... but for myself i dont know where to start.

Wackerbarth, 2015-05-31:
Surely that is not a criterion for the existence of an Issue.

PS, I do have an idea where to start, and I am, in fact, working on it in my fork of Marlin.

boelle, 2015-05-31:
i know its just moving files arround.... i could try it but do not gurantee the outcome :-P

fmalpartida, 2015-05-31:
Next weekend I hope to get the SAV MkI moving to 1.6.x. That should start taking care of the AT90USB cores or at least a subset.

Wackerbarth, 2015-05-31:
@fmalpartida -- I'm working on it for my KosselPro.
I have it compiling for the AT90USB1286, but using a very old Arduino Core.
Right now, I'm trying to get Travis builds working.

Please assist me in my github fork so that we don't duplicate effort.

boelle, 2015-05-31:
me got all the former boards in the addon folder converted so that they should worj on ide 1.6.4,,, but i want sombody to test it first... 

i dont have a fork but work on a copy of the marlin firmware on my desktop.... 

should it just commit and we can revert?

fmalpartida, 2015-05-31:
@Wackerbarth got you. Next weekend I'll bug you to wink into your repo and get it going.

Most likely you are in the process of getting the boot loader going?

Wackerbarth, 2015-05-31:
I "punted" on the boot loader.  The avr109 protocol handles only 64K and that is too small for Marlin.
Instead, I am using a HID version based on the LUFA code.  It (almost) works except for some issues with restarting after flashing.  I either have to power it off or press the reset multiple times.

The main issue that I currently face is that the Serial I/O from the Arduino 1.6 core doesn't play well with the way Marlin has hacked Serial into it's interrupt service loop.

fmalpartida, 2015-05-31:
I use the teensy environment and the cdc bootloader.

boelle, 2015-05-31:
anyway i just added addon for ide 1.6.4.... i think i got all board that could be found for older versions 

please give some feedback if i did it right?

Wackerbarth, 2015-05-31:
@boelle --
By "don't have a fork", do you mean that you are using the MarlinFirmware repository to push pull requests for any code that you change? I think that it would be more appropriate for you to "publish" your changes in the form of pull requests so that they can be reviewed and receive comments before being included in the Development branch.

@fmalpartida -- My repository is git@github.com:Wackerbarth/Marlin.git
Pull requests are welcome.  I'd like to get things working and then clean up the commits before submitting a pull request to the main repository.

Wackerbarth, 2015-05-31:
@boelle -- Although I like to see the -DHAS_AUTOMATIC_VERSIONING in platform.local.txt,
there is some more code that goes along with it.

Also, rather than Arduino_1.6.4+/Marlin/avr/... , It works better for these files to be in
Arduino_1.6.4+/hardware/Marlin/avr/... to be consistent with the 1.5.x version.

Also, there are more pieces missing.

Please revert that commit and let's clean it up through the PR process.

AnHardt, 2015-05-31:
@boelle 
Indeed this looks a bit unfinished.
My request would be to replace U8GLIB v1.11 by v1.17, what is tested for a while now.

boelle, 2015-06-01:
yes, i do changes directly.... since this one is so simple (just a new folder) it should be ok

boelle, 2015-06-01:
oki... i will just delete the folder

Wackerbarth, 2015-06-01:
@boelle --
I don't like your "solution" (removing the folder from the active source)
Unfortunately, this still leaves all of that information in the .git cache because you can "recover" the state where you had all of these files added. That causes EVERYONE to have to carry the overhead associated with an unnecessarily bloated repository.

You really should go back to the point before you added your submission and redo the subsequent commits. YES< this will cause you to do a force commit (and that might temporarily inconvenience a few), but it is the only way to avoid penalizing EVERYONE for eternity.

boelle, 2015-06-01:
i did a revert instead

Wackerbarth, 2015-06-01:
Sorry, but as the attached graphic shows, your "revert" just adds an additional commit which reverses the previous one. It does not remove the offending one (or the repository space that it occupies)

Commit 2724baa is still present.

As I said before, the only way to REMOVE your commit is to go back to 11e3f33 (or 764e617 which is earlier in its history) and then merge the contents of pull request #2209
[image]

boelle, 2015-06-01:
i used the web interface to do it....

Wackerbarth, 2015-06-01:
Well, that doesn't do what you need.  Use git directly.

Wackerbarth, 2015-06-01:
In your local repository add my github repository git@github.com:Wackerbarth/Marlin.git as a remote.
then set your Development branch to mine (119970b56a646d1807a2fe57e64ab9a4da45cd70) and force push it to the MarlinFirmware repository

thinkyhead, 2015-06-10:
Looks like the 1.6.x ArduinoAddons update is happening, so we can close this and focus on the PR.

closed 2015-06-10 by thinkyhead