MarlinFirmware/Marlin #1589: Servo retraction at last point of Auto Bed Level

Ursus63, 2015-03-10:
How can i set to move up Z BEFORE retract servo at last point of ABL Grid when i use G29 E? I have an optical endstop on servo and when servo retracts after last point of grid it scratches bed.

brainscan, 2015-03-10:
Are you using the latest version of marlin? I think it should do this as standard now but I haven't checked.

Ursus63, 2015-03-10:
Downloaded Marlin today. And it don't do this. Now i just disable retraction in firmware at all and add to g-code G1 Z15 and M402 after G29 E and its work. But i think its better to do it in firmware.

brainscan, 2015-03-10:
Sorry I thought this was added with the recent update to the bed levelling. It should be the same as the one I've been using which does raise z before retracting. The code can be found at http://3dprintboard.com/showthread.php?3105-Auto_Bed_Leveling-Enhanced-G29-command 
Hope this helps.

boelle, 2015-03-10:
i have at least now added both potential bug label and needs more feedback... if 1-2 more can verify this i will upgrade it to verified bug and include it in a milestone so it can be fixed...

to me its seems pretty important that you raise Z just a bit before you swing up probe arm

of course depends much on what kind of probe... some can raise the probe without raising Z...

Ursus63, 2015-03-11:
It will be good to rize probe before retract in G28 and G30 too. G28 i'm modified as described at the link above, but i think it must be in firmware by default. And G30 i think too.

SkyFlyer2, 2015-03-11:
The best way is to add the code lifting the Z-axis before any retract servo

Please add the code to function  - static void retract_z_probe() 
I think it might look something like this:

do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS] + Z_RAISE_BETWEEN_PROBINGS);

I  damaged already one servo without lifting Z-axis before servo retraction.

SkyFlyer2, 2015-03-11:
I tried add code to function static void retract_z_probe() - this is not correct, because it always lift Z-axis before retract, even if not required. Need to find another place in code.

alexborro, 2015-03-11:
I have added a new feature yesterday. Check this out in configuration.h:

  #define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10"
//These commands will be executed in the end of G29 routine. Useful to
retract a deployable probe.

Cheers.

Alex.

SkyFlyer2, 2015-03-11:
A nice solution. I will check it.

But this is not solve same problem for commands G28, G30 and M48.  No any changes  in their code, but the same issue also present there.

SkyFlyer2, 2015-03-12:
Hmm. I've checked this changes. No useful changes. The servo retracting BEFORE lift Z-axis, as result it scratches the bed and make damage to servo.

thinkyhead, 2015-03-13:
This could be done as an enhancement to G29 or as a compile-time option, or both. I'll see where new code may help.

avluis, 2015-03-17:
@thinkyhead Can you think of implementing something similar at the end of a G28 as well - a lift of even 1mm will save many micro switches ;)

maverikou, 2015-03-18:
G29 does this:

do_blocking_move_to(current_position[X_AXIS], current_position[Y_AXIS], Z_RAISE_AFTER_PROBING);

However that same code is not in G28. Is that the bug you're seeing?

maverikou, 2015-03-20:
This is a duplicate of #1507

thinkyhead, 2015-03-21:
Let's move the discussion over to #1507 then....

closed 2015-03-21 by thinkyhead