MarlinFirmware/Marlin #1591: Marlin does not compile with MAX6675

Grogyan, 2015-03-11:
RAMPS 1.4
Arduino IDE 1.60
Development Branch

temperature.cpp: In function 'void manage_heater()':
temperature.cpp:529:59: error: 'max_temp_error' was not declared in this scope
     if (ct > min(HEATER_0_MAXTEMP, 1023)) max_temp_error(0);
                                                           ^
temperature.cpp:530:59: error: 'min_temp_error' was not declared in this scope
     if (ct < max(HEATER_0_MINTEMP, 0.01)) min_temp_error(0);
                                                           ^

Commit 9c9726d had some cleanup done of the code, which possibly broke code compilation for the MAX6675 

I did not have the time to check the code prior to this commit which endevours to squash #1227

boelle, 2015-03-11:
i have marked it verfied and added it to a milestone

Grogyan, 2015-03-11:
Shouldn't this be added to Milestone 3, which has the SPI issue earmarked for fixing?

boelle, 2015-03-11:
dooh yes of course.... i guess there is a common issue that once fixed will solve them all

AnHardt, 2015-03-11:
At least compiles again with pull request #1595.

boelle, 2015-03-11:
but not with current code?

AnHardt, 2015-03-11:
@boelle 
As the title of this issue says.

thinkyhead, 2015-03-11:
I'm going to re-roll #1595 to keep those methods private, just move them above `manage_heater`.

Grogyan, 2015-03-12:
Missing changes from #1595 
Does not compile without these changes

thinkyhead, 2015-03-13:
@Grogyan Try again…? That code should work. Watch out for the Arduino IDE. It doesn't always notice code changes without a reload.

closed 2015-03-13 by thinkyhead

Grogyan, 2015-03-13:
Compiles now.
All good.