MarlinFirmware/Marlin #1592: web configurator

boelle, 2015-03-11:
as suggested i'm creating this issue so it has its own list of discussion

@MagoKimbra created one and @thinkyhead  created one...

either way i think a tool as this should have its own web space

thinkyhead, 2015-03-12:
:meat_on_bone: This actually works, but it's not affiliated with Github: https://rawgit.com/thinkyhead/Marlin/marlin_configurator/Marlin/configurator/index.html

boelle, 2015-03-12:
i wonder if marlin should have its own domain name?

daid, 2015-03-12:
You'll have to look around (or ask around) on the Ultimaker forums. But there is an online configurator which compiles as well as generating the config file. So that's very easy for people to use.

boelle, 2015-03-12:
the one scott made? or somerhing we dont know?

scotty1024, 2015-03-12:
$9000 USD for Marlin.org? Have they been smoking crack?

boelle, 2015-03-12:
hehe some make a living of buying domains and sell them at insane prices

what about .net or .nl?

scotty1024, 2015-03-12:
Bo,

That isn't what is happening here. The domain is not registered. They are asking for $9000 because they think they can get it because it is desirable. Essentially it is more like  raising the price of gasoline after a hurricane tears up everything.

boelle, 2015-03-12:
ok... let me try from little denmark

boelle, 2015-03-12:
here i get it listed as taken

Den torsdag den 12. marts 2015 skrev Bo Herrmannsen <
bo.herrmannsen@gmail.com>:

boelle, 2015-03-12:
> <javascript:_e(%7B%7D,'cvml','bo.herrmannsen@gmail.com');>>:

Den torsdag den 12. marts 2015 skrev Bo Herrmannsen <
bo.herrmannsen@gmail.com>:

Drakelive, 2015-03-12:
Hi Bo

marlinfirmware.org is available.

Drake

boelle, 2015-03-12:
might be worth considering...

what are peoples opinions ?

Drakelive, 2015-03-12:
A domain allows us to concentrate in one place Information, Wiki, Guides,
Documentations, RSS Feed, Forum and a complete Web Configurator.
Banner, Donations and Gadgets may also allow us to finance the project.

My opinion,  this is a necessary step and should be ready with the release
of version Stable

Drake

scotty1024, 2015-03-12:
Domain paid for, droplet created at DigitalOcean. Waiting for things to stabilize and gel a bit.

Spinning up an Ubuntu image.

boelle, 2015-03-12:
i agree...

so donations... who should take care of those and in what order should
donataions be spent?

scotty1024, 2015-03-12:
The domain is on its own NS account so it can be handed off. The droplet can be backed up and transferred. For now the domain and droplet are paid for.

Those that want to be admin for Wiki et al can be configured in with admin for the various services later today.

thinkyhead, 2015-03-13:
Nice, thanks everyone! So do we get to install an AVR build system on this Ubuntu box??

scotty1024, 2015-03-13:
Absolutely. 

The droplet is easily enhanced as needed to give it more resources. Their top tier is currently 20 CPUs, 64GB RAM, 640 GB SSD, and 9TB of traffic for $640 USD a month. Currently it’s sized to 1 CPU, 1GB RAM, 30 GB SSD, 2 TB of traffic. I will monitor resource usage and adjust.

The domain is hooked to the droplet.
The droplet is LAMP’d.
The droplet is MediaWiki’d as you can see when you go to http://www.marlinfirmware.org <http://www.marlinfirmware.org/>

The Wiki can be moved but I don’t have a home page so the wiki is the landing page for now.

Let me know what you need for the web configurator e.g. account, packages….

Let me know what other goodies to light up.

I’m going to load the avr tool chain and head for bed.

boelle, 2015-03-13:
also some sort of donation link for those of us that want to chip in on the
cost

scotty1024, 2015-03-14:
I’ve used one of those silly free web site templating tools to whip up an idea of a landing page.

The Wiki remains the default page but you can slip around it and see the landing page here: http://www.marlinfirmware.org/index.html <http://www.marlinfirmware.org/index.html>

The avr tool chain is installed.

If you are ready to move forward with the build system I’m ready to work with you to begin wiggling it into the web site and droplet.

scotty1024, 2015-03-14:
Hi Scott,

Forgot you already had code up I could copy over to start the conversation.

If you email me a password I will setup an account that will let you get in and start welding and sanding.

http://www.marlinfirmware.org/configurator/index.html <http://www.marlinfirmware.org/configurator/index.html>

scotty1024@me.com <mailto:scotty1024@me.com>

thinkyhead, 2015-03-14:
Thanks Scotty, I will keep plugging away at the configurator to get the first phase (Advanced Interface) complete, and then get into the more public-facing side soon after that. I will contact you pretty soon about the dev account.

closed 2015-05-31 by boelle