MarlinFirmware/Marlin #1593: The button "stop" on the Reprap Discount Full Graphics Smart Controller is not working.

SkyFlyer2, 2015-03-11:
This button worked in previous versions of the firmware Marlin (v.1.0.0 RC2).
Now when i click on it nothing happens. I used this button if something goes wrong and needs immediate stop.
For example, sometimes Z-probe fail and if not stop the program, as result will damage extruder, heated bed, servo and more.

thinkyhead, 2015-03-12:
@SkyFlyer2 Which electronics board are you using with that controller? Is it a RAMPS? It's possible the pin-assignment got altered someplace as we've been cleaning up pins. The pin that would be affected is (I believe) the BTN_ENC pin.

SkyFlyer2, 2015-03-12:
Yes, it is ramps 1.4 + arduino 2560 R3.

Where and what i must change to make it work again?

thinkyhead, 2015-03-13:
@SkyFlyer2 I'll have a look at this and get back to you.

Kiessar, 2015-03-14:
I also noticed this problem, my stop button didn't worked last time. 

Ramps 1.4
Arduino 2560

thinkyhead, 2015-03-14:
Maybe if you change line 1416 in ultralcd.cpp to this…?

    if ((millis() > blocking_enc) && (READ(BTN_ENC) == 0)) newbutton |= EN_C;

thinkyhead, 2015-03-14:
Can you determine the Arduino pin number the button is using?

SkyFlyer2, 2015-03-15:
How to do it?

thinkyhead, 2015-03-15:
@SkyFlyer2 Good question! Well, I have compared an earlier version of Marlin with the current version, really just to see how `BTN_ENC` might differ, and I couldn't see any difference. So, one way you can check whether these values differ is to insert some code into the `setup()` function in `Marlin_main.cpp` to print out the value of the pin. For example:

    SERIAL_ECHO_START;
    SERIAL_ECHOPAIR("BTN_ENC = ", BTN_ENC);

Try this with an earlier version of Marlin where the button works, and then try it with the current Development branch, in which the button doesn't work. If the value of `BTN_ENC` is different, then we will know that this is the problem. If it is the same, then there must be some other problem.

AnHardt, 2015-03-16:
@SkyFlyer2 
Digged a bit in the Schematics:
http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
http://reprap.org/wiki/RepRapDiscount_Smart_Controller

The full graphic controller has the kill-pin on D41 and
the normal smart controller on D29

In Configuratin.h 
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER automatically defines
REPRAP_DISCOUNT_SMART_CONTROLLER.
See https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/Configuration.h#L640

So https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/pins_RAMPS_13.h#L82
is ether wrong for the one or the other.

Try 29 here (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/pins_RAMPS_13.h#L83).

Has to be differentiated better in the pins file. Will lock for the other pins later.

AnHardt, 2015-03-16:
@SkyFlyer2 
The beeper is identical. The other stuff works - i know.

thinkyhead, 2015-03-16:
Ah, `KILL_PIN` … of course …

SkyFlyer2, 2015-03-16:
I tried change 41 to 29 (https://github.com/MarlinFirmware/Marlin/blob/Development/Marlin/pins_RAMPS_13.h#L83). 

Has no effect.

With this:

[code]
SERIAL_ECHO_START;
SERIAL_ECHOPAIR("BTN_ENC = ", BTN_ENC)
[/quote]

Gets error:

[code]
In file included from Marlin_main.cpp:30:0:
Marlin_main.cpp: In function 'void setup()':
Marlin.h:87:74: error: call of overloaded 'serial_echopair_P(const char*, int)' is ambiguous
 #define SERIAL_ECHOPAIR(name,value) (serial_echopair_P(PSTR(name),(value)))
[/code]

thinkyhead, 2015-03-16:
Huh, I guess pin 29 isn't the one you need.... As for SERIAL_ECHOPAIR, maybe the first string parameter should be wrapped in PSTR().

AnHardt, 2015-03-16:
@thinkyhead 
No, its not 29. Was to tied yesterday. Both controllers have KILL_PIN on D41.
Hm, and at lest i get the kill-message on the console. RAMPS1.4, Full graphic controller.
[image]
Sorry, no printer at the test-boards.

AnHardt, 2015-03-16:
@SkyFlyer2 & @Kiessar 
All is locking well for me.
For how long do you press the button? There is a loooong debouncer loop to avoid accidental kills.

SkyFlyer2, 2015-03-16:
BTN_ENC is not a variable, this just a define. There is no suitable function for it.

BTN_ENC = 35 in both version of firmware.

Quickly (less than one second).
On the latest version of the firmware, nothing happens, even if the button is held for a long time.

I don't get any messages when pressing the button. Only in early Marlin, where the button works normally.

SkyFlyer2, 2015-03-16:
I solved this.

In file "pins_RAMPS_13.h" 

#if defined(REPRAP_DISCOUNT_SMART_CONTROLLER) || defined(G3D_PANEL)
  #define KILL_PIN           29

The value must be 41.

" #define KILL_PIN           41"

also, "KILL_DELAY=10000". This value is too high for an immediate response. Must be reduced to 1000 (or 500).

Now, it is works good.

Please, correct this in code.

AnHardt, 2015-03-16:
Tested KILL_DELAY with 1000 and there is still a chance to to press the button not long enough.
750 matches my taste.

There is rarely used piece of code beneath - dealing with a HOME-button where the loop is equally long. Pin is only defined in pins_SAV_MK1.h an there the standard is -1. Shall we adjust this to?

SkyFlyer2, 2015-03-16:
750 will be enough, i think.

HOME-button will useful, if it will not have a long delay, so, i think, you can also change delay for it.

boelle, 2015-03-17:
So can we close this one as solved?

AnHardt, 2015-03-17:
I now what to do. But PR is not ready. Hunting other bugs.

SkyFlyer2, 2015-03-17:
Please, don't forget make changes according to this issue.

Then you can close this issue. It's solved.

AnHardt, 2015-03-18:
PR #1631 is ready and online.

AnHardt, 2015-03-18:
Hunted bug was a died analog pin on one off my megas. :-(

avluis, 2015-03-18:
A dead pin is never fun, glad to see that the RepRapDiscount Full Graphic Display's kill button will now work properly after the merge is complete.

boelle, 2015-03-18:
@thinkyhead  you think its ok to merge that PR so we can close this one?

thinkyhead, 2015-03-19:
Done!

closed 2015-03-19 by thinkyhead

jeremykoerber, 2019-01-14:
Hi all, I know this issue is old and closed but it was the most relevant issue I could find regarding my issue. I have an MKS Gen L board and the RepRapDiscount Full Graphic Smart LCD Controller and the kill button isn't working. Running the latest bugfix-2.0.x code, and the controller and display work as expected otherwise. I tried adding `#define KILL_PIN 41` to `pins_MKS_GEN_L.h` but it seems to have no effect. Not sure that's actually the correct pin. The controller is connected via 2 ribbon cables to EXP1 and EXP2 on the board. Anyone know the correct pin to use, and if anything else might need to be added to the firmware to make this work? I confirmed `KILL_DELAY` is 750. Thanks.

kAdonis, 2019-01-15:
`#define KILL_PIN 41`
is still correct for the MKS Gen L and it is defined (in the included Ramps.h)
You have probably faulty hardware, I dont think its firmware related

jeremykoerber, 2019-01-15:
I have 2 different MKS GEN L boards, and they both do the same thing. The controller is fresh out of the box. Not saying it couldn't be hardware, but it seems unlikely. As far as `pins_RAMPS.h` goes, I only see a `#define` block for `REPRAP_DISCOUNT_SMART_CONTROLLER`, but there's no definition for `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` which is what I have enabled. Thoughts?

kAdonis, 2019-01-15:
It is unlikely that the Problem is in the MKS Gen L, I would suspect the ribbon cables or a bad solder at the LCD,
If REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is defined, REPRAP_DISCOUNT_SMART_CONTROLLER is defined internally (in Conditionals_LCD.h )