MarlinFirmware/Marlin #1594: irc channel

boelle, 2015-03-11:
is anyone actually using the channel? i drop in quickly from time to time but there hardly seems to be anyone using it...

i'm considering an unregister from my freenode registration.. others can of course register it

my grounds for this is that it seems to be prefered to discuss things in the issues instead...

thinkyhead, 2015-03-12:
I would use it if there was something to discuss in real-time. But there may be other channels for that. G+ etc. That being said, I have lately grown more fond of IRC, but it works best when two or more are working on something and it helps to compare things in real-time.

boelle, 2015-03-12:
oki doki, the channel stays, but i'm still looking for some suitble log
software

closed 2015-03-14 by thinkyhead