MarlinFirmware/Marlin #1596: Motherboard name mismatch between Makefile and Arduino IDE

ivan-galvez, 2015-03-11:
Tested with Witbox printer, but might affect others.

If Marlin is compiled using Arduino IDE,  the $MOTHERBOARD name is only defined in Configuration.h, which in Witbox's case would be:
#define MOTHERBOARD BOARD_WITBOX

However, using the Makefile, MOTHERBOARD is set by:
CTUNING += -DMOTHERBOARD=${HARDWARE_MOTHERBOARD}

which in Witbox's case would be number 33, which corresponds to BOARD_RAMPS_13_EFB in boards.h and then translated to MENDEL when printed on the LCD.

The root cause of the issue is the fact that Witbox is sharing the

thinkyhead, 2015-03-13:
Witbox is 41, not 33. That should work from the Makefile.

boelle, 2015-03-13:
@ivan-galvez did 41 work?

ivan-galvez, 2015-03-16:
Precisely, the issue here is the mixture between printers and boards. In the Makefile there is no number 41 defined, so 33 was being used.

boelle, 2015-03-16:
So 41 worked?

ivan-galvez, 2015-03-16:
Well, if I just copy-paste the the block for 33 as a 41 it would work, but the thing here is that we are mixing the definition of boards with printers and that the Makefile and the Arduino IDE are configuring the project in a different way.

boelle, 2015-03-16:
upgraded to verified bug and put on milestone 5... the others are filled up

Lukelectro, 2015-04-12:
Does this fix work for this bug? https://github.com/bq/Marlin/commit/9baf402462e10a1dc13791a5a1cc7cca26fe714c

thinkyhead, 2015-04-13:
I've made the change recommended by @ivan-galvez and @Lukelectro in #1898. It almost makes sense to do this in all the Configuration.h files, even in some of them adding `#ifdef MOTHERBOARD` `#undef MOTHERBOARD` – always overriding the Makefile.

closed 2015-04-26 by thinkyhead