MarlinFirmware/Marlin #1599: Serial communication with Marlin

Samickz, 2015-03-12:
Hi,
I'm trying to open a serial communication with the Marlin firmware to start a print from linux..
There is a sequence to follow before sending the gcode file of the stl file or i can open the communication and send the raw gcode file on the usb and the printer will print it?
I mean.. i have to send some other gcodes before the gcode of the model to tell the printer that the print is starting or no?

nophead, 2015-03-12:
Normally gcode from a slicer has everything to do the print. The host adds line numbers and checksums before sending it and waits for OK before sending the next line. It also handles resend requests from the firmware. The printer would print without these additions but any communication error would not be handled.

Samickz, 2015-03-12:
Do you know where i can find these added lines and checksums?

nophead, 2015-03-12:
They are described here: http://reprap.org/wiki/G-code#N_and_.2A

Samickz, 2015-03-12:
Thanks :) going to try these next time

boelle, 2015-03-13:
Open a new issue if you need more help.. closing this one as it seems to have suited its purpose

closed 2015-03-13 by boelle