MarlinFirmware/Marlin #1600: JKN Advance algorithm implementation

alexborro, 2015-03-13:
I think we need to fix the JKN Advance implementation.
I'm following the sailfish forum and folks over there says it has a huge impact on print quality.

@bkubicek , are you able to help us on that??

Cheers.

Alex.

dcnewman, 2015-04-12:
FWIW, https://groups.google.com/d/msg/ultimaker/QMFVqEBarbQ/TYYYCbj6hwcJ

bkubicek, 2015-04-12:
any help that you want.
I also would like to point out that machinekit driven printers have a neat
way of using my advance idea.
http://blog.machinekit.io/2014/05/velocity-driven-extrusion.html

Bernhard

boelle, 2015-05-31:
any news on this one`?

boelle, 2015-06-29:
will close this one... we have not heard any news in 29 days and no word from alex since he posted tihs

closed 2015-06-29 by boelle

Nandox7, 2015-07-02:
I was curious to try this in Marlin.
Is there any place to get details on this algorithm? The details I can find searching the web are quite old, so I suspect by know Sailfish must be using something a bit more tweaked.

Best i could find was this, but doesn't seem to be much activity going on there: https://github.com/jetty840/Sailfish-G3Firmware

dcnewman, 2015-07-02:
https://github.com/jetty840/Sailfish-MightyBoardFirmware/

Nandox7, 2015-07-02:
Thanks!
Was any work ever done to try and add this to Marlin?

dcnewman, 2015-07-02:
Not that I'm aware of.

thinkyhead, 2015-07-03:
It's an interesting topic. MachineKit is also cited as having a good advance implementation at [Bas's Velocity Extrusion page](http://basdebruijn.com/tag/velocity-extrusion/). And there are some good posts and links on the topic at https://github.com/Smoothieware/Smoothieware/issues/645 also.

What is the name (or type) of the current `ADVANCE` algorithm feature in Marlin (as it's not "JKN")? It seems to be attempting to calculate based only on pressure, but no velocity factor.

Nandox7, 2015-07-03:
Could it be the Nozzle Pressure control (Repetier) or Advance Pressure Relax (Sailfish)?
Both seem to

thinkyhead, 2015-07-03:
Here are some highlights of Sailfish's implementation…

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/1852182effceadf9ded8c59c4ff4da0e1c94c64e/firmware/src/MightyBoard/Motherboard/Steppers.cc

- advanceK and advanceK2 defined and used in plan_init()

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/1852182effceadf9ded8c59c4ff4da0e1c94c64e/firmware/src/MightyBoard/Motherboard/StepperAccel.cc

- Define enum and vars needed for JKN
- Define st_extruder_interrupt()
- Setup the st_extruder_interrupt rate in st_init()
- In st_deprime_enable(bool) set lastAdvanceDeprime[e] = 0 if LEAD_DE_PRIME is being used
- Handle deprime behavior in setup_next_block()
- Set advance_state = ADVANCE_STATE_ACCEL in setup_next_block()
- Also test e_steps before calling stepperAxisSetHardwareEnabledToMatch(axesEnabled) in st_interrupt()
- Handle advance for accelerated blocks in the steps loop in st_interrupt()
- Handle deceleration phase in st_interrupt(), updating advance_pressure_relax_accumulator and advance_state
- Handle nominal phase in st_interrupt(), updating advance_state

https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/1852182effceadf9ded8c59c4ff4da0e1c94c64e/firmware/src/MightyBoard/Motherboard/StepperAccelPlanner.cc

- Define final_speed_step_rate
- Do extra processing in calculate_trapezoid_for_block(), set extra fields in current block instance
- Set k, k2 in plan_init()
- Test whether to use advance_lead in plan_buffer_line(), set true or false